webcamoid/webcamoid #577: Setting Output Directory Path on Windows using Search/Select Folder adds extra slash

Anutrix, 2022-12-27:
# Reporting a problem?
Yes.

## Summary
Setting Output Path on Windows using Search/Select Folder adds extra slash.

## Current Behavior
Setting Output Path on Windows using Search/Select Folder adds extra slash.
E.g, Selecting `D:/Webcamoid` *using the search button* stores path as `/D:/Webcamoid`.
Thus, Capture button doesn't do anything when path starts with `/`.
Happens for both video and picture output directories.

## Expected Behavior
Store correct path. And capture buttons should work.

## Steps to Reproduce
1.  Open Preferences.
2.  Click Search button for Video Directory path.
3.  Select any folder using the Windows Explorer selection UI.
4.  Notice path starts with `/`.

## Suggestions and tips

In the code logic, just remove(or don't add) extra slash at the beginning of path when saving path on Windows.

## Your Environment

* Webcamoid information: (version) (architecture) 9.0.0 64 bit - Qt 5.15.2
* Operating System information: (name) (numeric version and codename if applied) (architecture) Windows 11 Pro 22H2 64 bit
* Any other useful information: (logs, gdb backtrace, valgrind logs, screenshots, hardware, etc.) Lenovo FHD 300 Web Cam

dchmelik, 2023-05-12:
There seems to be no such 'search' button so I can't save; why not have easier classic/standard menus?

hipersayanX, 2023-05-12:
@dchmelik 

I don´t understand your complaining, the **Search** button is there and always was.

[image]

Also, a saving button is not necessary, all options are automatically saved unless explicitly required, it's the same exact design as Gnome apps.

[image]

hipersayanX, 2023-06-10:
Bug fixed in Webcamoid 9.1.0.

closed 2023-06-10 by hipersayanX