webcamoid/webcamoid #580: Adhere to the theme of the host.

rokejulianlockhart, 2023-02-15:
* [x] Search the [issues list](https://github.com/webcamoid/webcamoid/issues) for similar topics before opening a new one.
* If you know how to fix the problem, consider doing a [pull request](https://github.com/webcamoid/webcamoid/pulls) instead of opening a new issue.
* [x] Only report problems in [latest](https://github.com/webcamoid/webcamoid/releases) and [development](https://github.com/webcamoid/webcamoid/) version. Reporting issues related to older versions will be rejected.
* Use [gist](https://gist.github.com/) to post logs longer than 1024 characters.
* Webcamoid as project, is not affiliated or endorsed to any distribution, report packaging problems in their respective issue tracker.
* [x] Respect the templates, we need as much information as possible.
* Don't open an issue and disappear, we need you at least the first week to clear up missing information.
* Missing information makes useless and unsolvable an issue report.
* You are our debugger, eyes and hands, if an issue is not reproducible then you will be the only person able to solve it.
* [x] Take a seat and wait for your turn, as many others that has their issues open, or much better try fixing it your self and collaborate with the solution :smile:

## Current Behavior

When I invoke Webcamoid, its appearance is distinctly different to all of the other Qt-based software on my computer for no apparent reason.

## Expected Behavior

Webcamoid should adhere to the theme that its host desktop environment defines.

## Steps to Reproduce

Write the steps to reproduce the issue below:

1. `lookandfeeltool --apply org.kde.breezedark.desktop`
2. `webcamoid`

## Suggestions and tips

Adhere to the theme that is applied by KDE Plasma.

## Your Environment

* Webcamoid information: [9.0.0](https://github.com/webcamoid/webcamoid/commit/7d66553cdb1951429dae9924b3fe49afd321e689) 64-bit
* Operating System information:
  `Get-Content -Path /etc/*release | Select-String -Raw 'CPE_NAME'`
  ```
  CPE_NAME="cpe:/o:opensuse:tumbleweed:20230213"
  ```
  `PS /home/rokejulianlockhart/30741832/code/configuration> python3 -c 'import platform; print(platform.machine())'`
  ```
  x86_64
  ```
* Any other useful information: (logs, gdb backtrace, valgrind logs, screenshots, hardware, etc.)
  [image]
  [image]
  `flatpak run io.github.webcamoid.Webcamoid`
  [image]
  ```
  [2023-02-15 17:47:56.107, Webcamoid, 0x7fc3c48fe100, VideoOutputPicture.qml (73)] warning: qrc:/Webcamoid/share/qml/VideoOutputPicture.qml:73:13: QML Image: Protocol "" is unknown
  [2023-02-15 17:47:57.168, Webcamoid, 0x7fc342ffd640,  (0)] debug: Camera input frame format: AkCaps(mimeType="video/unknown",fourcc=QVariant(QString, "YUYV"),fps=QVariant(QString, "30/1"),height=QVariant(uint, 480),width=QVariant(uint, 640))
  ```

hipersayanX, 2023-02-15:
No and never, no f* way, all themes provided by Qt are horribly crafted and in most cases doesn't work as they are supposed to be working. Having a custom theme allow me to have a more precise and consistent design.

KDE Plasma, does you note something missing?

[image]

Gnome and others, we are not in 90's anymore.

[image]

Windows, squares everywhere.

[image]

Mac, white text over light gray box, sure the best design ever.

[image]

Android theme, does not even fit properly in an Android phone screen. 

[image]

rokejulianlockhart, 2023-02-16:
@hipersayanX, I'm not certain of what the problem in the first picture is, since even when using `kde-gtk-config` to synchronize Qt and GTK software's themes, nothing even vaguely similar that has ever appeared. I'm really hesitant to blame, I suppose, but might that be because of an issue with Webcamoid rather than Breeze?

Would you at least allow a toggle? I understand your desire to not have to deal with this issue since using one mandatory theme means no complaints about incompatible themes, but if every application were to use their own style, navigation and usability would obviously severely suffer:

One of my favourite aspects of the native Qt support of Plasma is how everything looks so consistent, and therefore is so wonderfully easy to navigate without ever using it previously. Going back to Windows is difficult, mostly for this reason.

hipersayanX, 2023-02-16:
@rokejulianlockhart

>I'm not certain of what the problem in the first picture was caused by

The problem is that KDE Plasma Custom Qml theme is always many versions outdated in relation to Qt official releases. in the first capture, buttons are not showing the icon because they are implementing an older API than the [required one to do so](https://doc.qt.io/qt-5/qml-qtquick-controls2-abstractbutton.html#icon.name-prop), it needs QtQuick.Controls 2.3 for showing the icon, that [was implemented in Qt 5.10](https://doc.qt.io/qt-5/qtquickcontrols-index.html#versions). It's always the same problem with KDE, they goes many versions backward Qt.

>Would you at least allow a toggle? I understand your desire to not have to deal with this issue since using one mandatory theme means no complaints about incompatible themes, but if every application were to use their own style, navigation and usability would obviously severely suffer:

Sure I can do, but right now I'm not putting all my bets into Qml, I've some other ideas in my mind.

>One of my favourite aspects of the native Qt support of Plasma is how everything looks so consistent

Webcamoid is a Qml app, KDE apps are mostly programmed with QtWidgets, Qml apps running with KDE Qml custom theme looks clunky and does not look native. Because again, neither Qt or KDE developers put any effort designing the Qml themes carefully.

RokeJulianLockhart, 2025-06-20:
> https://github.com/webcamoid/webcamoid/issues/580#issuecomment-1433887781

KDE's Breeze `QQuickStyle` has since improved (and Union shall remediate this entirely), whereas Qt 6 now implements a WinUI3 `QQuickStyle` (and maybe a `QStyle`, too). All that remains is https://github.com/flathub/org.gtk.Gtk3theme.Breeze/issues/170#issuecomment-2957604430 for `flatpak` to inherit this improved style.

hipersayanX, 2025-06-20:
My idea is more like adapting [the current theme](https://github.com/webcamoid/webcamoid/tree/master/StandAlone/share/themes/WebcamoidTheme) to imitate other OS themes, for example, for the current theme is based on [Material UI](https://m2.material.io/components/buttons). The problem is that most of theme are not as well defined as Material UI. 

# GNU/Linux

https://develop.kde.org/hig/
https://developer.gnome.org/hig/

# Windows

https://learn.microsoft.com/en-us/windows/apps/design/controls/buttons
https://developer.microsoft.com/en-us/fluentui#/controls/web

# MacOS

This is actually very well defined.

http://interface.free.fr/Archives/AquaHIGuidelines.pdf

# Fluent UI

https://react.fluentui.dev/?path=/docs/components-button-button--docs&globals=storybook_fluentui-react-addon_theme:web-dark;storybook_fluentui-react-addon_dir:ltr

The problem is that I have to implement 52 controls each time, 52 * 4 = 208 (discarding Fluent UI), and I'm not sure if I can use AI to speed up the thing 😟
Also because I want to reuse the code for customizing the color scheme and adapting for different screen DPIs like the current theme does, because the idea is to allow the user to switch between different themes, so for example can use KDE Breeze theme in Windows, os MacOS Aqua theme in Android.
But all of this is a titanic work, and I think that there are other priorities that users may like more.
I will it for sure, but much much later.

RokeJulianLockhart, 2025-06-21:
> The problem is that most of theme are not as well defined as Material UI.

@hipersayanX, indeed, most of them aren't meant to be re-implemented by 3rd-parties, unlike M3.

> I won't use Qt default themes because I still think they are horribly implemented.

To ensure that you understand, I'm not asking for this: I wouldn't want it either (I hate Fusion). Perhaps you're used to macOS or Windows, where the Qt themes might be embedded (IDK, because I don't much utilise those platforms), but in Qt-based DEs, like KDE Plasma (but even some GTK ones, like GNOME) provide a `QStyle` and `QQuickStyle` that all Qt applications can consume. As an example:

1.	1.	~~~
		import sys
		try:
		    from PyQt6.QtWidgets import QApplication, QStyleFactory
		    
		    app = QApplication(sys.argv)
		    
		    print('Available Qt6 QStyles:')
		    styles = QStyleFactory.keys()
		    for style in sorted(styles):
		        print(f'  - {style}')
		    
		    print(f'\nTotal QStyles found: {len(styles)}')
		        
		except ImportError as e:
		    print(f'PyQt6 QtWidgets not available: {e}')
		    
		try:
		    from PyQt6.QtQml import qmlRegisterType
		    from PyQt6.QtQuickControls2 import QQuickStyle
		    
		    print('\nAvailable Qt6 QQuickStyles:')
		    quick_styles = QQuickStyle.availableStyles()
		    for style in sorted(quick_styles):
		        print(f'  - {style}')
		    
		    print(f'\nTotal QQuickStyles found: {len(quick_styles)}')
		    
		except ImportError as e:
		    print(f'PyQt6 QtQuickControls2 not available: {e}')
		    
		except Exception as e:
		    print(f'Error getting QQuickStyles: {e}')
		~~~

	1.	<blockquote>

		~~~YAML
		Available Qt6 QStyles:
		  - Breeze
		  - Fusion
		  - Windows

		Total QStyles found: 3
		~~~

		</blockquote>

1.	1.	~~~py
		import sys
		try:
		    from PyQt6.QtGui import QGuiApplication
		    import os
		    
		    app = QGuiApplication(sys.argv)
		    
		    # Try to get QQuickStyle info
		    try:
		        import PyQt6.QtQuick
		        print('PyQt6.QtQuick available')
		    except:
		        print('PyQt6.QtQuick not available')
		    
		    # List QML Controls styles from filesystem
		    qml_controls_path = '/usr/lib64/qt6/qml/QtQuick/Controls'
		    if os.path.exists(qml_controls_path):
		        print('\nQQuickStyles found in filesystem:')
		        styles = []
		        for item in os.listdir(qml_controls_path):
		            path = os.path.join(qml_controls_path, item)
		            if os.path.isdir(path) and not item.startswith('.') and item not in ['designer', 'impl']:
		                styles.append(item)
		        
		        for style in sorted(styles):
		            print(f'  - {style}')
		        
		        print(f'\nTotal QQuickStyles found: {len(styles)}')
		    
		except Exception as e:
		    print(f'Error: {e}')
		~~~

	1.	<blockquote>

		~~~YAML
		PyQt6.QtQuick available

		QQuickStyles found in filesystem:
		  - Basic
		  - FluentWinUI3
		  - Fusion
		  - Imagine
		  - Material
		  - Universal

		Total QQuickStyles found: 6
		~~~

		</blockquote>

Inheriting these requires no effort on your part, except not manually overriding them.

hipersayanX, 2025-06-21:
>Inheriting these requires no effort on your part, except not manually overriding them.

Believe me, is not even near that easy.

[image]

If you want to use Breeze or your system theme in general just comment [this line](https://github.com/webcamoid/webcamoid/blob/d50ebf0af44da053e4aeb01739237bd708177123/StandAlone/src/mediatools.cpp#L633), compile the code and try if its that "easy".

RokeJulianLockhart, 2025-06-22:
> https://github.com/webcamoid/webcamoid/issues/580#issuecomment-1432221998

@hipersayanX, before I try to recompile without that line, I've tested it in its native form (not Flatpak or AppImage'd) on some platforms, and must note that I don't understand your criticism of Fusion in https://github.com/webcamoid/webcamoid/issues/580#issuecomment-1432221998, [^1] when the default theme doesn't appear to be an improvement upon it:

[^1]: <details>

      > [image]

      </details>

Windows | OpenSUSE
--------|---------
[image] | [image]

Additionally, in the case of Breeze:

1. Are icons defined as part of the `QQuickStyle`? I was under the assumption that `QIcon::fromTheme` is an explicit declaration, that can be made solely if the current theme returns an icon.

1. Irrespective, `breeze-icons` is a *massive* library, so did you definitely poll the correct icons? Considering that yours appear to be embedded, it might not be surprising that you'd need to refactor the icon code to work differently when consuming a theme.

I do understand now that my initial assumption that you'd not need to do much is incorrect, though. I use Plasma, and I always design my GUIs around theme compatibility, so I forgot that anyone embedded icons anymore.