webcamoid/webcamoid #583: FTBFS with ffmpeg-6.0

grover92000, 2023-03-18:
Latest git build fails with ffmpeg-6.0:

```
[ 25%] Building CXX object libAvKys/Plugins/VideoCapture/src/convert/ffmpeg/CMakeFiles/VideoCapture_ffmpeg.dir/src/convertvideoffmpeg.cpp.o
/tmp/build/webcamoid/libAvKys/Plugins/VideoCapture/src/convert/ffmpeg/src/convertvideoffmpeg.cpp: In member function ‘virtual bool ConvertVideoFFmpeg::init(const AkCaps&)’:
/tmp/build/webcamoid/libAvKys/Plugins/VideoCapture/src/convert/ffmpeg/src/convertvideoffmpeg.cpp:208:31: error: ‘AV_CODEC_CAP_TRUNCATED’ was not declared in this scope; did you mean ‘AV_CODEC_CAP_HARDWARE’?
  208 |     if (codec->capabilities & AV_CODEC_CAP_TRUNCATED)
      |                               ^~~~~~~~~~~~~~~~~~~~~~
      |                               AV_CODEC_CAP_HARDWARE
/tmp/build/webcamoid/libAvKys/Plugins/VideoCapture/src/convert/ffmpeg/src/convertvideoffmpeg.cpp:209:43: error: ‘AV_CODEC_FLAG_TRUNCATED’ was not declared in this scope; did you mean ‘AV_CODEC_FLAG_DROPCHANGED’?
  209 |         this->d->m_codecContext->flags |= AV_CODEC_FLAG_TRUNCATED;
      |                                           ^~~~~~~~~~~~~~~~~~~~~~~
      |                                           AV_CODEC_FLAG_DROPCHANGED
make[2]: *** [libAvKys/Plugins/VideoCapture/src/convert/ffmpeg/CMakeFiles/VideoCapture_ffmpeg.dir/build.make:118: libAvKys/Plugins/VideoCapture/src/convert/ffmpeg/CMakeFiles/VideoCapture_ffmpeg.dir/src/convertvideoffmpeg.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:4346: libAvKys/Plugins/VideoCapture/src/convert/ffmpeg/CMakeFiles/VideoCapture_ffmpeg.dir/all] Error 2
make: *** [Makefile:136: all] Error 2

```
AV_CODEC_CAP_TRUNCATED has been deprecated for a while and was removed in ffmpeg-6.0.  Fix should be easy enough:

```
--- convertvideoffmpeg.cpp.orig 2023-03-05 07:28:32.000000000 -0500
+++ convertvideoffmpeg.cpp      2023-03-18 15:45:57.997823270 -0400
@@ -205,8 +205,10 @@ bool ConvertVideoFFmpeg::init(const AkCa
         return false;
     }

+#if LIBAVCODEC_VERSION_MAJOR < 60
     if (codec->capabilities & AV_CODEC_CAP_TRUNCATED)
         this->d->m_codecContext->flags |= AV_CODEC_FLAG_TRUNCATED;
+#endif

 #ifdef CODEC_FLAG_EMU_EDGE
     if (codec->capabilities & CODEC_CAP_DR1)

```

hipersayanX, 2023-03-19:
This one was fixed in the [latest commit](https://github.com/webcamoid/webcamoid/commit/b5dfb2895b62ab0dc9e5807d6efa847868911df4#diff-709a98d74dfe92daa58a1546963ddf3e79de2a11870f46403708003931eaa33d).

hipersayanX, 2023-06-10:
Bug fixed in Webcamoid 9.1.0.

closed 2023-06-10 by hipersayanX