webcamoid/webcamoid #584: About MacOS support

hipersayanX, 2023-03-19:
About giving support to future MacOS versions, I've some things to say...

I will be clear, I will try to give MacOS support as long as possible, but at this point I still working with Mac Catalina that is the latest working version for me and can't test newer versions because none of them works, forget about buying a Mac because it's out of budget.

I will start migrating native MacOS code to third-party libraries, so for example instead of having a specific plugin for [CoreAudio](https://github.com/webcamoid/webcamoid/blob/440280381d67e142153fca2b4c7be6f99d37e56f/libAvKys/Plugins/AudioDevice/src/coreaudio/src/audiodevcoreaudio.cpp#L34), it will delegate the work to [SDL](https://github.com/webcamoid/webcamoid/blob/440280381d67e142153fca2b4c7be6f99d37e56f/libAvKys/Plugins/AudioDevice/src/sdl/src/audiodevsdl.cpp) or [PortAudio](http://www.portaudio.com/).

# Advantages

- I only need to test once in Linux.
- Any bug report in Mac will be delegated to those third-party libraries.
- I won't to constantly depend on a Mac developer to maintain the plugin, developers usually come and goes.
- Less work to do.

# Disadvantages

- Third-party usually lack a lot of features or does not works as expected, for example [SDL](https://github.com/Rust-SDL2/rust-sdl2/issues/901) which does not works in MacOS, or [PortAudio](https://github.com/PortAudio/portaudio/wiki/HotPlug) which is missing HotPlug support yet.
- Blind development, if no other developers step-up, I will depend on users to tell me if a fix works or not.

This changes could give a bit more time to Webcamoid in Mac, but do not expect to last long. You know what to do, step-up or let it die.

Hanzzsz, 2023-04-21:
Any  webcamoid version for MAC High Sierra 10.13?

hipersayanX, 2023-04-23:
@Hanzzsz Any of the [old versions](https://github.com/webcamoid/webcamoid/releases) maybe, but those versions doesn't have any kind of support.

hipersayanX, 2023-06-05:
# Maintainers for MacOS are welcome

As [news says](https://eclecticlight.co/2020/08/22/apple-silicon-macs-will-require-signed-code/) and many users reported me, Apple has enforced code signing for apps working with it's ARM processors, so for me it make no sense to release pre-compiled binaries anymore, therefore since Webcamoid 9.1 and forward you will have to compile it from source code or install it from [homebrew](https://brew.sh/).
Also, the current [Webcamoid package in homebrew](https://formulae.brew.sh/cask/webcamoid) is just a Cask that install the pre-compiled binaries, that formulae will need to be rewritten to compile straight from sources. The formulae is currently maintained by @miccal but I haven't any contact with him, and not sure if he want to keep maintaining it.
In resume, I need people to maintain Webcamoid's code and homebrew packages for Mac.