webcamoid/webcamoid #585: Complete failure to record video

davidgiven, 2023-03-25:
## Summary

I just used Webcamoid to record a bunch of video for a project. While collecting all the video files together for editing, I discover that all the files that Webcamoid recorded are empty, meaning I've lost all my video.

## Current Behavior

As above. I've tried this with both a USB HDMI capture card and a webcam. Webcamoid produces the same empty files each time.

## Expected Behavior

Files should contain video?

## Steps to Reproduce

1. Open Webcamoid.
2. Connect to video source. Notice that the video is displayed in the window.
3. Press the 'record' button. Notice the flashing 'recording' indicator with the increasing timer.
4. After a while, press the 'stop' button. Notice that a file is generated.
5. The file is always 264 bytes long and contains no video.

## Your Environment

* Webcamoid information: 9.0.0 amd64
* Operating System information: Debian testing

Enclosed is one of the generated files.

[Video 2023-03-25 19-36-48.webm](https://user-images.githubusercontent.com/100638/227735696-00d4fc46-c34f-4213-8e09-a15565ee0483.webm)

hipersayanX, 2023-03-29:
Does the terminal show any error like in #408?

hipersayanX, 2023-06-10:
Bug fixed in Webcamoid 9.1.0.

rodrigopedra, 2023-06-20:
I still have this still same issue on OpenSUSE Tumbleweed. Although my saved files always have 258 bytes.

I am using these video settings (I don't remember having changed them)

[image]

System Info

```text
Operating System: openSUSE Tumbleweed 20230617
KDE Plasma Version: 5.27.5
KDE Frameworks Version: 5.107.0
Qt Version: 5.15.10
Kernel Version: 6.3.7-1-default (64-bit)
Graphics Platform: X11
```

Package Info

```bash
~ $ zypper info webcamoid
Loading repository data...
Reading installed packages...

Information for package webcamoid:
----------------------------------
Repository     : multimedia:apps
Name           : webcamoid
Version        : 9.1.1-41.2
Arch           : x86_64
Vendor         : obs://build.opensuse.org/multimedia
Installed Size : 16,0 MiB
Installed      : Yes
Status         : up-to-date
Source package : webcamoid-9.1.1-41.2.src
Upstream URL   : https://webcamoid.github.io/
Summary        : Webcam applet for Plasma
Description    : 
    Webcam applet for Plasma.

    Features:
    * Take pictures with the webcam.
    * Record videos.
    * Manages multiple webcams.
    * Play/Stop capture, this saves resources while the applet is not in use.
    * Written in C++.
    * Qt based software.
    * Custom controls for each webcam.
    * Popup applet support (you can embed Webcamoid in the panel).
    * +50 video effects available.
    * Effects with live previews.
    * Translated to many languages.
    * Stand alone installation mode (use it as a normal program).
    * Use custom network and local files as capture devices.
    * Capture from desktop.
```

Running it from the terminal, has no relevant output

```bash
~ $ webcamoid
QML debugging is enabled. Only use this in a safe environment.
```

rodrigopedra, 2023-06-20:
As an additional note, I tested the version from `multimedia:apps` (as in above), also the AppImage from the website, and the flatpak version from flathub.

Only the flatpak version works as expected, but it reports as version: 9.0.0

[image]

I also tested with both ffmpeg versions 5 and 6. Both fail.

If you want me to test anything different, please let me know.

hipersayanX, 2023-06-20:
@rodrigopedra the ideal would by compiling in Debug mode and testing it with the same parameters. In release mode, the FFmmpeg's output is disabled.

rodrigopedra, 2023-06-20:
@hipersayanX I used the instructions on these pages to install dependencies, compile, and install:

https://github.com/webcamoid/webcamoid/wiki/Build-and-install-in-GNU-Linux-and-FreeBSD

https://github.com/webcamoid/webcamoid/blob/8cede577b78b8c4ef90bfbc0a4053dea13875ac4/ports/ci/linux-opensuse/install_deps.sh#L91

The only change I made on the compile script was changing `-DCMAKE_BUILD_TYPE=Release` to `-DCMAKE_BUILD_TYPE=Debug`

I started webcamoid from the CLI, tried recording a short video. The output video file is now 257 bytes long and doesn't play.

The CLI output is below

<details>
    <summary>Click to expand</summary>

    ~/code/plasma $ webcamoid
    QML debugging is enabled. Only use this in a safe environment.
    [swscaler @ 0x7f1714001300] deprecated pixel format used, make sure you did set range correctly
    [swscaler @ 0x7f171403ce00] deprecated pixel format used, make sure you did set range correctly
    [swscaler @ 0x7f1714049e40] deprecated pixel format used, make sure you did set range correctly
    [swscaler @ 0x7f1714049e40] deprecated pixel format used, make sure you did set range correctly
    [swscaler @ 0x7f1714056e80] deprecated pixel format used, make sure you did set range correctly
    [swscaler @ 0x7f1714056e80] deprecated pixel format used, make sure you did set range correctly
    [swscaler @ 0x7f1714056e80] deprecated pixel format used, make sure you did set range correctly
    [swscaler @ 0x7f1714056e80] deprecated pixel format used, make sure you did set range correctly
    [swscaler @ 0x7f1714056e80] deprecated pixel format used, make sure you did set range correctly
    [swscaler @ 0x7f1714056e80] deprecated pixel format used, make sure you did set range correctly
    [swscaler @ 0x7f1714056e80] deprecated pixel format used, make sure you did set range correctly
    [swscaler @ 0x7f1714056e80] deprecated pixel format used, make sure you did set range correctly
    [swscaler @ 0x7f1714056e80] deprecated pixel format used, make sure you did set range correctly
    [swscaler @ 0x7f1714056e80] deprecated pixel format used, make sure you did set range correctly
    [swscaler @ 0x7f1714056e80] deprecated pixel format used, make sure you did set range correctly
    [swscaler @ 0x7f1714056e80] deprecated pixel format used, make sure you did set range correctly
    [swscaler @ 0x7f1714056e80] deprecated pixel format used, make sure you did set range correctly
    [swscaler @ 0x7f1714056e80] deprecated pixel format used, make sure you did set range correctly
    [swscaler @ 0x7f1714056e80] deprecated pixel format used, make sure you did set range correctly
    [swscaler @ 0x7f1714056e80] deprecated pixel format used, make sure you did set range correctly
    [swscaler @ 0x7f1714056e80] deprecated pixel format used, make sure you did set range correctly
    [swscaler @ 0x7f1714056e80] deprecated pixel format used, make sure you did set range correctly
    [swscaler @ 0x7f1714056e80] deprecated pixel format used, make sure you did set range correctly
    [swscaler @ 0x7f1714056e80] deprecated pixel format used, make sure you did set range correctly
    [swscaler @ 0x7f1714056e80] deprecated pixel format used, make sure you did set range correctly
    [swscaler @ 0x7f1714056e80] deprecated pixel format used, make sure you did set range correctly
    [swscaler @ 0x7f1714056e80] deprecated pixel format used, make sure you did set range correctly
    [swscaler @ 0x7f1714056e80] deprecated pixel format used, make sure you did set range correctly
    [swscaler @ 0x7f1714056e80] deprecated pixel format used, make sure you did set range correctly
    [swscaler @ 0x7f1714056e80] deprecated pixel format used, make sure you did set range correctly
    [swscaler @ 0x7f1714056e80] deprecated pixel format used, make sure you did set range correctly
    [swscaler @ 0x7f1714056e80] deprecated pixel format used, make sure you did set range correctly
    [swscaler @ 0x7f1714056e80] deprecated pixel format used, make sure you did set range correctly
    [swscaler @ 0x7f1714056e80] deprecated pixel format used, make sure you did set range correctly
    [swscaler @ 0x7f1714056e80] deprecated pixel format used, make sure you did set range correctly
    [swscaler @ 0x7f1714056e80] deprecated pixel format used, make sure you did set range correctly
    [swscaler @ 0x7f1714056e80] deprecated pixel format used, make sure you did set range correctly
    [swscaler @ 0x7f1714056e80] deprecated pixel format used, make sure you did set range correctly
    [swscaler @ 0x7f1714056e80] deprecated pixel format used, make sure you did set range correctly
    [swscaler @ 0x7f1714056e80] deprecated pixel format used, make sure you did set range correctly
    [swscaler @ 0x7f1714056e80] deprecated pixel format used, make sure you did set range correctly
    [swscaler @ 0x7f1714056e80] deprecated pixel format used, make sure you did set range correctly
    [swscaler @ 0x7f1714056e80] deprecated pixel format used, make sure you did set range correctly
    [swscaler @ 0x7f1714056e80] deprecated pixel format used, make sure you did set range correctly
    [swscaler @ 0x7f1714056e80] deprecated pixel format used, make sure you did set range correctly
    Output #0, ipod, to '/home/rodrigo/Videos/Webcamoid/Video 2023-06-20 18-12-53.m4v':
    Stream #0:0: Audio: aac (LC), 44100 Hz, mono, fltp, 128 kb/s
    [ipod @ 0x20f90c0] Invalid packet stream index: 1
    [ipod @ 0x20f90c0] Invalid packet stream index: 1
[... cut ...]
    [swscaler @ 0x7f1714056e80] deprecated pixel format used, make sure you did set range correctly
    [swscaler @ 0x7f1714056e80] deprecated pixel format used, make sure you did set range correctly
    
    ~/code/plasma $ ls -la /home/rodrigo/Videos/Webcamoid/Video\ 2023-06-20\ 18-12-53.m4v
    -rw-r--r-- 1 rodrigo rodrigo 257 jun 20 18:12 '/home/rodrigo/Videos/Webcamoid/Video 2023-06-20 18-12-53.m4v'

</details>

Mind my local timezone is GMT-03:00m so the output timestamp might look weird but it is correct.

Let me know if you need any additional info.

rodrigopedra, 2023-06-20:
@hipersayanX I noticed I was using the iPod preset, as the last time I attempted to make it work I tried to change to that.

So I recorded a small screencast to show usage with the default MP4 preset

https://github.com/webcamoid/webcamoid/assets/5470108/3b220304-b94b-418d-8bf4-04f6c13ffa22

hipersayanX, 2023-06-24:
@rodrigopedra It's only recording the audio but not the video, will check.

hipersayanX, 2023-08-02:
@karstengit Which audio and video codecs did you tried?

Lalarian, 2023-10-01:
Hello! I have the same issue (On my Archlinux system and Webcamoid 9.1.1): Only 258 byte files are 'recorded'.
With the preset default codec (WebM) I could not even start recording, the record button was greyed out.
So I tried:
-MP4 / x264
-MP4 / libx265 H.265 / HEVC
-MP4 / HEVC (Intel QuickSync)
-MP4 / H.264 ( AVC / IntelQuickSync)

These four all gave the same result:
I could click the record button, and when starting it created the file with 44 bytes. During the supposed record time file size did not change. When I clicked stop, the file size increased to 258 bytes.

Next I tried:
-MKV / HEVC (IntelQuickSync)
-MKV / MPEG2-Video
for seeing if it's container or codec dependent, and both just created 4.7KByte files everytime.

Last I tried:
-OOG / libvpx  VP8
and it always created 4.2KByte files.
So it seems as if generally only header data is written to the files, while the video stream on screen is completely ignored.