webcamoid/webcamoid #587: No UI on startup

Chimychanga, 2023-04-01:
## Summary

Just installed webcamoid on a relatively clean OS install. functionality seems to work but all UI/buttons don't appear on the client. 

Spacebar still work as a method to take photos however settings and effects can't be used because they don't appear.

## Current Behavior

No UI shown. Pressing the spacebar flashes the screen and takes screenshot as expected.
[image]
fyi: webcam works fine, covered lens as it's not relevant to the problem

## Expected Behavior

UI for settings, snap, effects should show on startup

## Steps to Reproduce

1. yay -S webcamoid
2. Open up webcamoid
3. No UI

## Your Environment

* Webcamoid information: webcamoid-9.0.0-2 amd64
* Operating System information: EndeavourOS Linux x86_64

hipersayanX, 2023-04-01:
Did you tried installing ```webcamoid-git```?
The terminal show any output?
You are using Wayland or Xorg?
Graphics card?

Chimychanga, 2023-04-01:
Thanks for the quick reply,
1. I've just tried installing `webcamoid-git`, same thing unfortunately
2. when opening webcamoid through terminal it only shows 
`[2023-04-02 02:22:12.311, Webcamoid, 0x7ffa38008000, main.qml (169)] warning: qrc:/Webcamoid/share/qml/main.qml:169:5: QML ColumnLayout: Binding loop detected for property "width"`
3. Currently I'm using Xorg
4. `GPU 0: NVIDIA GeForce RTX 3080    Driver Version: 525.89.02    CUDA Version: 12.0`

hipersayanX, 2023-04-01:
Qml is not marking error. maybe Qml has some bug with nVidia cards? everything else is fine.
Try this, write the following file:

**test.qml**

```
import QtQuick 2.12
import QtQuick.Layouts 1.15
import QtQuick.Controls 2.12

ApplicationWindow {
    width: 320
    height: 240
    visible: true

    ColumnLayout {
      Button {
          text: "Click me"
      }
      Button {
          text: "Click me"
      }
    }
}
```

And run with:

```
qmlscene-qt5 test.qml
```

Chimychanga, 2023-04-01:
Seems to display correctly. 
Let me know if there's logs elsewhere I could be looking at.
[image]

hipersayanX, 2023-04-02:
>Let me know if there's logs elsewhere I could be looking at.

Those are the logs, and they are not marking any error. I've not any idea what could be causing your problem, I'm an Arch Linux user and can grant that Webcamoid works fine on it, but no idea what could be your problem, maybe try updating your system, maybe try using it in other desktops, try modifying the source code if you know how to program, but I can't help much on this.

Chimychanga, 2023-04-02:
All good, I'll keep digging and update this if I find a solution. It was working on Manjaro a couple months back so most likely OS specific. Thanks for attempting to help me out nonetheless

hipersayanX, 2023-04-02:
I'll try installing EndeavourOS, but I totally bet that it will work for me, because there is probably maybe some theme or whatever interfering with Qml or something not sure. 
In the past someone else reported me some similar problem (not the same), and it resulted to be some Qt/Qml theme causing the bug for they, but it worked fine for me when installed the fresh distro, so let see :man_shrugging:

Chimychanga, 2023-04-02:
I just tried booting into a live USB and it seemed to work... This seems like a me problem, oh well. 

Did a bit of digging and found https://forum.endeavouros.com/t/blank-malfunctioning-ui-elements-in-qt-applications-e-g-vnote-ghostwriter/26757 although they're on a different desktop (XFCE vs KDE)
tldr: they tried a different machine and issue went away so they suspect it was driver related

hipersayanX, 2023-04-02:
I was thinking, maybe applying ```export QT_QUICK_BACKEND=software``` before calling Webcamoid?

Chimychanga, 2023-04-02:
Definitely did something as startup took a bit longer, but same outcome

hipersayanX, 2023-04-02:
Yes, exactly as expected, it's working fine for me.

[image]

Tried with every single desktop.

[image]

So the bug is only triggered under very special conditions, I can't reproduce.

>Definitely did something as startup took a bit longer, but same outcome

Qml use OpenGL or Vulkan for UI rendering, with software rendering it draw everything without any kind of hardware acceleration, it's much slower but it's supposed to be granted to work.

>they tried a different machine and issue went away so they suspect it was driver related

Maybe try using an LTS kernel with a LTS nVidia driver, or just for the mere testing, using nouveau?

Chimychanga, 2023-04-03:
Hmm I see. I super tempted to reinstall my OS since I've only set it up ~ 3 weeks ago, so starting fresh shouldn't be too bad. I'll close the ticket if this resolves my issue

hipersayanX, 2023-04-03:
>I super tempted to reinstall my OS since I've only set it up ~ 3 weeks ago

Try this before doing that:

>Maybe try using an LTS kernel with a LTS nVidia driver, or just for the mere testing, using nouveau?

Maybe it works and you don't need to reinstall nothing.

midhun-kumar, 2023-05-04:
I too am having this issue,
Ubuntu, Plasma, nVidia 2080 Ti

Chimychanga, 2023-05-06:
tldr: Reinstall my OS multiple times. Ditched EOS and ended up on Arch however still not working. IMO seems like a rare bug in the community and alternative apps work so I don't see a rush to fix it.

When it works:
From my experience it works on a live image of EOS probably other distros. 
Also works on initial boot of Arch after install. 

When it breaks:
Once I restart after the initial install it magically breaks (This is after a fresh install without any installs/updates other than webcamoid, possibly dual monitor related? I have a horizontal ultrawide and vertical 1440p).

I've reinstalled 3 times and had the same issue. I can't seem to make this reproducible on other's hardware so I don't expect it to be fixed until someone can make it reproducible.

Chimychanga, 2023-05-06:
Quick update:
I'm not familiar with the UI but if I press "Tab" followed by "Enter".

A grey box flicks on the left side. So some UI elements are loading but not all.
[image]

[gif of behaviour in action](https://imgur.com/LgFPP2c)

Chimychanga, 2023-05-06:
Another quick update:
Using spectacle I think I found the UI. If i resize the window to a massive 3,912x5,143 I can just see the UI, so maybe it's showing but for whatever reason to a massive size?

At 3,912x5,143
[image]

At 11,647x7,980 with mouse hover
[image]

You can just see  "Record video" text in the middle right of the image so I suspect the scaling's gone to wack for whatever reason?

hipersayanX, 2023-05-06:
@Chimychanga @midhun-kumar What's your screen dpi and resolution? https://dpi.lv/
Also try running Webcamoid as:

```
QT_SCALE_FACTOR=0.25 webcamoid
```

Play with the number, with lower or higher values, so maybe you can see the UI.

Chimychanga, 2023-05-06:
Resolution: 3440 × 1440
Diagonal: 13.3″
280 pixels per inch

I think we're getting somewhere! See here at `QT_SCALE_FACTOR=0.01 webcamoid`
[image]

See here at `QT_SCALE_FACTOR=0.001 webcamoid`. Closer but still a bit weird compared to before though.
[image]

hipersayanX, 2023-05-06:
Then this is some bug in the scaling. Try writing this python script:

**dpi.py**

```
import sys
from PyQt5 import QtGui, QtWidgets

app = QtWidgets.QApplication(sys.argv)
screen = QtGui.QGuiApplication.primaryScreen()
print("PDPI", screen.physicalDotsPerInch())
print("LDPI", screen.logicalDotsPerInch())
```

and run it with:

```
python dpi.py
```

What's the result?

Chimychanga, 2023-05-06:
PDPI 61976.0
LDPI 104.01248859060402

hipersayanX, 2023-05-06:
Oh, that was the reason, The size of the widgets are calculated in **dp**, so

```
pixels = dp * physicalDotsPerInch / 160
```

and *physicalDotsPerInch* has an absurd value, probably because [High DPI](https://doc.qt.io/qt-5/highdpi.html) was enabled. The right would have been calculating it relative to logicalDotsPerInch.

Chimychanga, 2023-05-07:
Can confirm it's been fixed! Thanks @hipersayanX 

[image]

closed 2023-05-07 by Chimychanga

hipersayanX, 2023-05-07:
Nice :smile: , anyways I had to leave physical DPI for Android, otherwise the widgets will look too small, I hope no having the same problem with Android smart TVs, or the same problem with Linux working in small screens :worried: