webcamoid/webcamoid #590: windows 10 can't uninstall webcamoid

neuronetv, 2023-05-04:
I installed webcamoid using [webcamoid-installer-windows-daily-master-win64.exe](https://github.com/webcamoid/webcamoid/releases/download/daily-build/webcamoid-installer-windows-daily-master-win64.exe)
but its not in the control panel/programs or anywhere else to uninstall. I've also got revo uninstaller and its not listed in there either.
how do I unistall it completely? - including the virual camera driver.

neuronetv, 2023-05-04:
ok thankyou.
do you know why webcamoid hangs and stops responding after the akvirtualcamera is installed?
is there a fix for this?

neuronetv, 2023-05-05:
thankyou and i have installed the virtual camera driver manually using:
https://github.com/webcamoid/akvirtualcamera/releases/download/daily-build/akvirtualcamera-windows-daily-master-1.exe

the webcamoid app is no longer hanging and not responding it now works better  however the virtual camera driver does not seem to be working. no app on my windows 10 pc recognises a virtual camera driver. do I have to manually start it in some way? or is there something else I'm missing?

hipersayanX, 2023-05-05:
Did you created at least one camera from Webcamoid after installing the driver?

hipersayanX, 2023-05-05:
Let me confirm, you have a camera created like in here, right?

[image]

neuronetv, 2023-05-05:
thankyou I have done as you say and set the output to virtual camera (see screenshot 1).but windows 10 still does not see a virtual camera, and nothing in device manager.  when I first added the virtual camera I did get a box asking me for permission to run akvirtualcamera and I clicked yes.
screenshot 2 shows the settings I get when I click on 'configure output'.
still stumped by this. all windows sees is the droidcam client.
[image]
[image]

hipersayanX, 2023-05-05:
>but windows 10 still does not see a virtual camera

In which apps did you tested in? Did you tested it with for example Firefox or Chrome?

>and nothing in device manager.

The camera won't never appear in the device manager because technically isn't a driver but a DirectShow filter, the client app must support DirectShow capture otherwise it won't never work.

neuronetv, 2023-05-05:
ok thankyou.
I was testing the virtual camera in the windows 10 default camera app. I could never get the webcamoid image. I also tried testing in whatsapp video call but that didn't recognise any virtual webcam either.
I've just tested in google chrome and in the chrome settings I can select the virtual camera driver and it worked however during testing the webcamoid program kept closing all by itself. this happened about 5 times and after that it would open for a 3 seconds and then close again. it wouldn't run any more.

hipersayanX, 2023-05-05:
>I was testing the virtual camera in the windows 10 default camera app. I could never get the webcamoid image.

Microsoft has recently introduced [mfvirtualcamera.h](https://learn.microsoft.com/en-us/windows/win32/api/mfvirtualcamera/) which make possible to write a virtual camera for Media Foundation based apps, but it's only for Windows 11, not sure if Microsoft will backport the API to old versions of Windows. I'll try much later writing virtual camera with that API.

>I've just tested in google chrome and in the chrome settings I can select the virtual camera driver and it worked

Nice

>however during testing the webcamoid program kept closing all by itself.

Will check that. If you uninstall the virtual camera it runs again?

neuronetv, 2023-05-06:
yes I unistalled the akvirtualcamera and now webcamoid is more stable, the program is not closing. I can switch between inputs without any problems now however this is not much use as the whole point of what I want to do os broadcast live video and be able to switch sources for the other person to see.

hipersayanX, 2023-05-06:
I know, but now I have a point where to start debugging.

neuronetv, 2023-05-06:
ok great, looking forward to your next daily build

hipersayanX, 2023-05-11:
At the end, the NSIS installer wasn't properly configured to show the icon in the Control panel, this was fixed in the daily build, will close this issue later.
Will check the virtual camera later, after Webcamoid 9.1.0 release, since it's more a bug in that side.

hipersayanX, 2023-06-10:
Bug fixed in Webcamoid 9.1.0.

closed 2023-06-10 by hipersayanX