webcamoid/webcamoid #591: is there a webcamoid download for windows 7? 32bit

neuronetv, 2023-05-20:
and a virtual camera driver to go with it?

hipersayanX, 2023-05-21:
You have win32 packages:

```
webcamoid-installer-windows-daily-master-win32.exe 
webcamoid-portable-windows-daily-master-win32.zip 
```

in the [daily build](https://github.com/webcamoid/webcamoid/releases/tag/daily-build), maybe works, maybe does not, Idk, did not tested because Windows 7 is EOL and all CPU now a days are 64 bits. And the virtual camera supports 32 bits too.

neuronetv, 2023-05-21:
thankyou I installed webcamoid on windows 7 from the links you gave however the program crashes as soon as the virtualcameradriver is installed.

hipersayanX, 2023-05-21:
It works without the virtual camera?

neuronetv, 2023-05-21:
yeah

closed 2023-06-10 by hipersayanX