webcamoid/webcamoid #592: Validate Snap Maintenance

CapeCrusader321, 2023-05-22:
Hey there, i wish to publish, build and maintain webcamoid as a snap, but to register the webcamoid name for snap, i need upstream approval on whether they agree on letting me build and maintain the snap or they will do it by themself, so can i build and maintain the webcamoid snap on behalf of this project ?? (This process is needed to register the name to publish this app as snap)

CapeCrusader321, 2023-05-25:
Ok, i will surely update you once everything is ready.

CapeCrusader321, 2023-06-14:
@hipersayanX  Sorry for the delay, i have started initializing this again --> https://launchpad.net/webcamoid-snap ,the delay is due to the fact that kde snap frameworks being broken, but they are being fixed and am also trying workarounds, but am still working on this and will surely update you once its finalized.

CapeCrusader321, 2023-06-14:
@hipersayanX  Can you please reopen this ?? the snap builds are ready and i have applied for registration --> https://launchpad.net/webcamoid-snap

CapeCrusader321, 2023-06-15:
@hipersayanX  Done, i have published it although its still in edge and not in stable, will take some time to publish to stable.

[image]

CapeCrusader321, 2023-06-15:
Ok, but be informed it may have issues and may not yet work properly, thanks.

CapeCrusader321, 2023-06-15:
Please divert all snap related bugs to -->  https://launchpad.net/webcamoid-snap

CapeCrusader321, 2023-06-15:
@hipersayanX  sorry to say, if possible please dont publish the links of snap in the websiteas its causing some error on wayland devices and will take time to get fixed,thanks for your understanding.

hipersayanX, 2023-06-15:
@CapeCrusader321 It's so bad? I usually more in the side of "It's better something than nothing", in the line [Release early, release often](https://en.wikipedia.org/wiki/Release_early,_release_often).
Did you tested if the virtual camera features are working? there was some problems with Flatpak, so imagine you will have similar problems too.

CapeCrusader321, 2023-06-16:
@hipersayanX  Ahh, yes the problem is that the app is unable to access camera even though we camera and audio-record plug enabled, my assumption is that the app uses dbus and maybe its not communicating with the hardware since in the initial snap revisions i had not enabled dbus slot, but now i have enabled dbus(Is io.github.webcamoid.Webcamoid the correct dbus id ??) plus can you please give me a list of runtime dependencies needed by the app, no problem keep it in the website but i think users should be noticed that they are using a beta version of webcamoid as its not yet ready.

CapeCrusader321, 2023-06-16:
@hipersayanX  good news after dbus its working fine, just wait until these request for automatic connection to camera and audio-record plugs is gained, nice weare going to have an official snap soon. -->   https://forum.snapcraft.io/t/autoconnect-requests-for-webcamoid/35628

CapeCrusader321, 2023-06-16:
My guess about dbus was correct.

hipersayanX, 2023-06-16:
@CapeCrusader321 

>My guess about dbus was correct.

If I remember well, PluseAudio used D-Bus for working.
Until now, which plugins are working?

[image]

CapeCrusader321, 2023-06-16:
@hipersayanX  I have built ffmpeg from source with literally all options enabled plus staged vlc-plugin base and some other deps, i believe all plugins should work, is the image you shared of the snap version ??
Here are my screenshots on lxqt.
Though root method is empty.
[image]
[image]
[image]

[image]

CapeCrusader321, 2023-06-16:
@hipersayanX  The snap confinement is even stricter than of flatpak,  only way it can spawn is using xdg-portals and yes the webcamoid snap has an issue of not opening saved img and videos but still can use xdg-portals for save file location.

hipersayanX, 2023-06-16:
Ok, I need to add support for xdg-portals for those parts then :+1:

CapeCrusader321, 2023-06-16:
> Ok, I need to add support for xdg-portals for those parts then +1

Thats good i believe we will get autoconnect request by next month until then the snap will be super-ready.

capecrusader-121, 2023-07-08:
@hipersayanX The permissions have been granted, and i will make the snap public soon, currently it has some build issues since i was trying to build sdl from source(maybe foolish of me) but now i will move ahead without building sdl from source. also i have created a launchpad mirror of this repo if you dont mind ?
https://launchpad.net/webcamoid , its just vanilla import of this repo.
Thesnap package will be soon available.

hipersayanX, 2023-07-10:
>also i have created a launchpad mirror of this repo if you dont mind ?

Sure, no problem :+1:

capecrusader-121, 2023-07-14:
@hipersayanX  I Have made the snap public and released to candidate, the snap works but still has some issues that can be neglected, although i will ask you to consider making the app sandbox friendly that's working inside a confinement using xdg-portals and less demanding permissions.

capecrusader-121, 2023-07-19:
@hipersayanX  The Webcamoid snap s popularity is exploding, 132 downloads in three days.
[image]

hipersayanX, 2023-07-19:
Great! just a bit of polishing and it will be perfect.

hipersayanX, 2023-07-19:
I was working in supporting PipeWire for webcam capturing right now, so later I will try to add Snap to the [pipeline](https://github.com/webcamoid/webcamoid/tree/master/.github/workflows) and then start fixing the missing features.

capecrusader-121, 2023-07-20:
Nice.
If needed do check its launchpad home.

hipersayanX, 2023-08-13:
snapcraft worked for a few times, then started throwing the following error, and can't run anymore in my computer, even if I remove all packages:

```
Executing parts lifecycle...                                                                                                    
Cannot install all requested build packages: cmake, file, g++, gcc, git, gstreamer1.0-plugins-base, gstreamer1.0-plugins-good, libasound2-dev, libavcodec-dev, libavdevice-dev, libavformat-dev, libavutil-dev, libgl1-mesa-dev, libgstreamer-plugins-base1.0-dev, libgstreamer1.0-0, libjack-dev, libkmod-dev, libpipewire-0.3-dev, libpulse-dev, libqt5opengl5-dev, libqt5svg5-dev, libsdl2-dev, libswresample-dev, libswscale-dev, libusb-dev, libuvc-dev, libv4l-dev, libvlc-dev, libvlccore-dev, linux-libc-dev, make, pkg-config, portaudio19-dev, qml-module-qt-labs-folderlistmodel, qml-module-qt-labs-platform, qml-module-qt-labs-settings, qml-module-qtqml-models2, qml-module-qtquick-controls2, qml-module-qtquick-dialogs, qml-module-qtquick-extras, qml-module-qtquick-privatewidgets, qml-module-qtquick-templates2, qt5-qmake, qtdeclarative5-dev, qtmultimedia5-dev, qtquickcontrols2-5-dev, qttools5-dev-tools, qtwayland5, vlc-plugin-base
```

Also it fails to locate libavkys, even if set the rpath, also I have to set QT_PLUGIN_PATH for it to locate Qt plugins, and even if I set QML2_IMPORT_PATH it won't find Qml files. Also, launching from desktop menu it just crash straight.  
What I'm missing? [here](https://github.com/webcamoid/webcamoid/blob/f27231dad06b3bf0106dc8cce97c1a20744991b1/ports/ci/snap/build_prepare.sh#L41) is the configuration I'm using.

capecrusader-121, 2023-08-14:
@hipersayanX  Snapcraft usually has problems with using a lot of dependencies, i also see kde extension missing in the snap yaml, plus the resultant size is also too much, have you seen the original snap yaml from launchpad ?
Also to build snap files you can use a simple github action, see -->  https://github.com/ubuntu/gnome-calculator/blob/stable/.github/workflows/build.yml

Using extensions is must otherwise it will increase headache  for maintenance.

capecrusader-121, 2023-08-15:
@hipersayanX  I have started building the arm64buillds and as such now the snap webcamoid is available for amd64 and arm64.

hipersayanX, 2023-08-19:
@capecrusader-121 

>@hipersayanX Snapcraft usually has problems with using a lot of dependencies,

I tried installing g++, make, and few other packages (ignoring Qt, FFmpeg and the rest) just for testing and it give me the same error message with just 2 or 3 packages, tried cleaning the cache too with no luck.

>i also see kde extension missing in the snap yaml, plus the resultant size is also too much, have you seen the original snap yaml from launchpad ?

Yes, but not at all. It was weir to me that you were building everything from scratch instead of using the packakes in the repository, also I don't understand why were you disabling features.
So I tried building the Snap as I understand the system, for example, I don't understand why are you including KDE related things when Webcamoid does not depends on KDE.

>@hipersayanX I have started building the arm64buillds and as such now the snap webcamoid is available for amd64 and arm64.

Nice :+1: