webcamoid/webcamoid #596: akvirtual camera - Virus detected

AndyIreland, 2023-06-29:
[image]

n3storm, 2023-06-30:
In case this helps

[image]

https://www.virustotal.com/gui/file/a378824867ed32bb6c540523fa530b58634f068a3c0849b3b24a1cedda7e5545/detection

Notice AVG engine does not detect it.

hipersayanX, 2023-07-10:
The virtual camera is detected as a false positive because is a plugin loading inside other programs, if you have any doubt you can seen in the [release page](https://github.com/webcamoid/akvirtualcamera/releases/tag/9.1.1) the commit and the source code from were all binaries were built. Also, you can ask the anti-malware vendor to put the virtual camera in the white list.