webcamoid/webcamoid #597: Build errors need some consultation

capecrusader-121, 2023-07-09:
Hey there as you know i have been trying to snap webcamoid, and had done it successfully but the video function was broken due to lib mismatch, so i decided to build complete ffmpeg from source for webcamoid but webcamoid refuses to recognize the ffmpeg libs as existent even though the libs are detected at test/checks.

Is there anything i can do to fix it, the remaining option is to use older ffmpeg libs from system repo but they will create conflicts. 

Build Log For better understanding;

https://launchpadlibrarian.net/676309314/buildlog_snap_ubuntu_jammy_amd64_webcamoid_BUILDING.txt.gz

hipersayanX, 2023-07-10:
Did you added the include directories to the flags?

```
export CXXFLAGS=${CXXFLAGS} -I/path/to/ffmpeg-sources/include
cmake ...
```

capecrusader-121, 2023-07-11:
> 

Hey there is this to be added as a cmake parameter or build environment ?

hipersayanX, 2023-07-11:
>@hipersayanX I didnt add it since it wasnt mentioned in build instructions,is this to be added as a cmake parameter or build environment ?

You said you were compiling FFmpeg from sources:

>so i decided to build complete ffmpeg from source for webcamoid

Now you need to tell Cmake how to find the headers and libraries, that's why you need to pass extra flags to Cmake.
Yet, I don't understand why are you doing so since Webcamoid compile fine with the FFmpeg from Ubuntu repositories.

>Also all the ffmpeg libs are installed in usr(bin and lib/x86_64) directory
>
>Is this fine --> CXXFLAGS=${CXXFLAGS} -I/usr/bin/include

If you have installed FFmpeg in the /usr prefix, then that flag isn't necessary, that flag is only necessary when the headers and libraries are located in uncommon places.

>but then will this cover lib directory ?

No, for linking libraries you use:

```
export LDFLAGS=${LDFLAGS} -L/path/to/ffmpeg-binaries/lib -lavcodec -lavformat ...
```

capecrusader-121, 2023-07-11:
@hipersayanX  I am compiling ffmpeg from source beacause the app at-least on the snap side webcamoid  suffers from the following problems:
1) Makes snap size large and as such makes it slower to load.
2) Video record function etc. has crashes.

I mainly need help because ffmpeg has been successfully installed to /usr and webcamoid tests also detect it successfully but during build times webcamoid refuses to locate the build ffmpeg libs

capecrusader-121, 2023-07-12:
@hipersayanX  lol, here is the root cause of this issue i was accidentally remove usr/include folder after compiling ffmpeg that folder contains header files 🤣  and thats why this drama i have fixed that hope it gets fixed.

closed 2023-07-14 by capecrusader-121