webcamoid/webcamoid #598: Webcamoid crashing when using gstreamer

capecrusader-121, 2023-07-15:
I have successfully snapped webcamoid and staged ubuntu jammy gstreamer libs with it.
The app initially crashes when using gstreamer instead of ffmpeg in settings.
Main:

```
[2023-07-15 17:58:43.190, Webcamoid, 0x7f31127fc640,  (0)] warning: Error pushing buffer to GStreamer pipeline
tcache_thread_shutdown(): unaligned tcache chunk detected
Aborted (core dumped)
```
Full:

```
[2023-07-15 17:58:27.899, Webcamoid, 0x7f319bfff640,  (0)] warning: propsReply "An AppArmor policy prevents this sender from sending this message to this recipient; type=\"method_call\", sender=\":1.134\" (uid=1000 pid=12788 comm=\"/snap/webcamoid/44/usr/bin/webcamoid \" label=\"snap.webcamoid.webcamoid (enforce)\") interface=\"org.freedesktop.DBus.Properties\" member=\"GetAll\" error name=\"(unset)\" requested_reply=\"0\" destination=\"org.freedesktop.NetworkManager\" (uid=0 pid=1608 comm=\"/usr/sbin/NetworkManager --no-daemon \" label=\"unconfined\")"
[2023-07-15 17:58:27.899, Webcamoid, 0x7f319bfff640,  (0)] warning: nmReply "An AppArmor policy prevents this sender from sending this message to this recipient; type=\"method_call\", sender=\":1.134\" (uid=1000 pid=12788 comm=\"/snap/webcamoid/44/usr/bin/webcamoid \" label=\"snap.webcamoid.webcamoid (enforce)\") interface=\"org.freedesktop.NetworkManager\" member=\"GetDevices\" error name=\"(unset)\" requested_reply=\"0\" destination=\"org.freedesktop.NetworkManager\" (uid=0 pid=1608 comm=\"/usr/sbin/NetworkManager --no-daemon \" label=\"unconfined\")"
[2023-07-15 17:58:27.901, Webcamoid, 0x7f319bfff640,  (0)] warning: "Object path cannot be empty"
[2023-07-15 17:58:29.625, Webcamoid, 0x7f31e12775c0, VideoOutputPicture.qml (81)] warning: qrc:/Webcamoid/share/qml/VideoOutputPicture.qml:81:13: QML Image: Protocol "" is unknown

** (webcamoid:12788): WARNING **: 17:58:33.318: arnr-type is a no-op since control has been deprecated in libvpx
[2023-07-15 17:58:33.352, Webcamoid, 0x7f31e12775c0,  (0)] debug: State changed from NULL to READY
[2023-07-15 17:58:33.356, Webcamoid, 0x7f31e12775c0,  (0)] debug: State changed from NULL to READY
[2023-07-15 17:58:33.357, Webcamoid, 0x7f31e12775c0,  (0)] debug: State changed from NULL to READY
[2023-07-15 17:58:33.357, Webcamoid, 0x7f31e12775c0,  (0)] debug: State changed from NULL to READY
[2023-07-15 17:58:33.357, Webcamoid, 0x7f31e12775c0,  (0)] debug: State changed from NULL to READY
[2023-07-15 17:58:33.357, Webcamoid, 0x7f31e12775c0,  (0)] debug: State changed from NULL to READY
[2023-07-15 17:58:33.357, Webcamoid, 0x7f31e12775c0,  (0)] debug: State changed from NULL to READY
[2023-07-15 17:58:33.357, Webcamoid, 0x7f31e12775c0,  (0)] debug: State changed from NULL to READY
[2023-07-15 17:58:33.358, Webcamoid, 0x7f31e12775c0,  (0)] debug: State changed from NULL to READY
[2023-07-15 17:58:33.358, Webcamoid, 0x7f31e12775c0,  (0)] debug: State changed from NULL to READY
[2023-07-15 17:58:33.358, Webcamoid, 0x7f31e12775c0,  (0)] debug: State changed from NULL to READY
[2023-07-15 17:58:33.358, Webcamoid, 0x7f31e12775c0,  (0)] debug: State changed from NULL to READY
[2023-07-15 17:58:33.358, Webcamoid, 0x7f31e12775c0,  (0)] debug: State changed from NULL to READY
[2023-07-15 17:58:33.358, Webcamoid, 0x7f31e12775c0,  (0)] debug: State changed from NULL to READY
[2023-07-15 17:58:33.358, Webcamoid, 0x7f31e12775c0,  (0)] debug: State changed from NULL to READY
[2023-07-15 17:58:33.358, Webcamoid, 0x7f31e12775c0,  (0)] debug: State changed from NULL to READY
[2023-07-15 17:58:33.358, Webcamoid, 0x7f31e12775c0,  (0)] debug: State changed from NULL to READY
[2023-07-15 17:58:33.358, Webcamoid, 0x7f31e12775c0,  (0)] debug: State changed from READY to PAUSED
[2023-07-15 17:58:33.358, Webcamoid, 0x7f31e12775c0,  (0)] debug: Stream Status: queue0 is 0
[2023-07-15 17:58:33.358, Webcamoid, 0x7f31e12775c0,  (0)] debug: State changed from READY to PAUSED
[2023-07-15 17:58:33.359, Webcamoid, 0x7f31e12775c0,  (0)] debug: Stream Status: queue1 is 0
[2023-07-15 17:58:33.359, Webcamoid, 0x7f31e12775c0,  (0)] debug: Stream Status: queue0 is 1
[2023-07-15 17:58:33.364, Webcamoid, 0x7f31e12775c0,  (0)] debug: State changed from READY to PAUSED
[2023-07-15 17:58:33.365, Webcamoid, 0x7f31e12775c0,  (0)] debug: Stream Status: queue1 is 1
[2023-07-15 17:58:33.365, Webcamoid, 0x7f31e12775c0,  (0)] debug: State changed from READY to PAUSED
[2023-07-15 17:58:33.365, Webcamoid, 0x7f31e12775c0,  (0)] debug: State changed from READY to PAUSED
[2023-07-15 17:58:33.366, Webcamoid, 0x7f31e12775c0,  (0)] debug: State changed from READY to PAUSED
[2023-07-15 17:58:33.366, Webcamoid, 0x7f31e12775c0,  (0)] debug: State changed from READY to PAUSED
[2023-07-15 17:58:33.366, Webcamoid, 0x7f31e12775c0,  (0)] debug: State changed from READY to PAUSED
[2023-07-15 17:58:33.366, Webcamoid, 0x7f31e12775c0,  (0)] debug: State changed from READY to PAUSED
[2023-07-15 17:58:33.366, Webcamoid, 0x7f31e12775c0,  (0)] debug: State changed from READY to PAUSED
[2023-07-15 17:58:33.366, Webcamoid, 0x7f31e12775c0,  (0)] debug: State changed from READY to PAUSED
[2023-07-15 17:58:33.366, Webcamoid, 0x7f31e12775c0,  (0)] debug: State changed from READY to PAUSED
[2023-07-15 17:58:33.366, Webcamoid, 0x7f31e12775c0,  (0)] debug: State changed from READY to PAUSED
[2023-07-15 17:58:33.366, Webcamoid, 0x7f31e12775c0,  (0)] debug: Stream Status: video_0 is 0
[2023-07-15 17:58:33.366, Webcamoid, 0x7f31e12775c0,  (0)] debug: Stream Status: video_0 is 1
[2023-07-15 17:58:33.367, Webcamoid, 0x7f31e12775c0,  (0)] debug: State changed from READY to PAUSED
[2023-07-15 17:58:33.367, Webcamoid, 0x7f31e12775c0,  (0)] debug: Stream Status: audio_1 is 0
[2023-07-15 17:58:33.367, Webcamoid, 0x7f31e12775c0,  (0)] debug: State changed from READY to PAUSED
[2023-07-15 17:58:33.367, Webcamoid, 0x7f31e12775c0,  (0)] debug: Stream Status: audio_1 is 1
[2023-07-15 17:58:33.369, Webcamoid, 0x7f31e12775c0,  (0)] debug: Recalculating latency
[2023-07-15 17:58:34.817, Webcamoid, 0x7f31e12775c0,  (0)] debug: Recalculating latency
[2023-07-15 17:58:34.818, Webcamoid, 0x7f31e12775c0,  (0)] debug: Stream started
[2023-07-15 17:58:34.818, Webcamoid, 0x7f31e12775c0,  (0)] debug: State changed from READY to PAUSED
[2023-07-15 17:58:34.819, Webcamoid, 0x7f31e12775c0,  (0)] debug: State changed from READY to PAUSED
[2023-07-15 17:58:34.819, Webcamoid, 0x7f31e12775c0,  (0)] debug: ASYNC done
[2023-07-15 17:58:34.819, Webcamoid, 0x7f31e12775c0,  (0)] debug: Recalculating latency
[2023-07-15 17:58:34.819, Webcamoid, 0x7f31e12775c0,  (0)] debug: New clock: GstSystemClock
[2023-07-15 17:58:34.819, Webcamoid, 0x7f31e12775c0,  (0)] debug: State changed from PAUSED to PLAYING
[2023-07-15 17:58:34.819, Webcamoid, 0x7f31e12775c0,  (0)] debug: State changed from PAUSED to PLAYING
[2023-07-15 17:58:34.819, Webcamoid, 0x7f31e12775c0,  (0)] debug: State changed from PAUSED to PLAYING
[2023-07-15 17:58:34.819, Webcamoid, 0x7f31e12775c0,  (0)] debug: State changed from PAUSED to PLAYING
[2023-07-15 17:58:34.819, Webcamoid, 0x7f31e12775c0,  (0)] debug: State changed from PAUSED to PLAYING
[2023-07-15 17:58:34.819, Webcamoid, 0x7f31e12775c0,  (0)] debug: State changed from PAUSED to PLAYING
[2023-07-15 17:58:34.820, Webcamoid, 0x7f31e12775c0,  (0)] debug: State changed from PAUSED to PLAYING
[2023-07-15 17:58:34.820, Webcamoid, 0x7f31e12775c0,  (0)] debug: State changed from PAUSED to PLAYING
[2023-07-15 17:58:34.820, Webcamoid, 0x7f31e12775c0,  (0)] debug: State changed from PAUSED to PLAYING
[2023-07-15 17:58:34.820, Webcamoid, 0x7f31e12775c0,  (0)] debug: State changed from PAUSED to PLAYING
[2023-07-15 17:58:34.820, Webcamoid, 0x7f31e12775c0,  (0)] debug: State changed from PAUSED to PLAYING
[2023-07-15 17:58:34.820, Webcamoid, 0x7f31e12775c0,  (0)] debug: State changed from PAUSED to PLAYING
[2023-07-15 17:58:34.820, Webcamoid, 0x7f31e12775c0,  (0)] debug: State changed from PAUSED to PLAYING
[2023-07-15 17:58:34.820, Webcamoid, 0x7f31e12775c0,  (0)] debug: State changed from PAUSED to PLAYING
[2023-07-15 17:58:34.820, Webcamoid, 0x7f31e12775c0,  (0)] debug: State changed from PAUSED to PLAYING
[2023-07-15 17:58:34.820, Webcamoid, 0x7f31e12775c0,  (0)] debug: State changed from PAUSED to PLAYING
[2023-07-15 17:58:34.820, Webcamoid, 0x7f31e12775c0,  (0)] debug: State changed from PAUSED to PLAYING
[2023-07-15 17:58:43.190, Webcamoid, 0x7f31127fc640,  (0)] warning: Error pushing buffer to GStreamer pipeline
tcache_thread_shutdown(): unaligned tcache chunk detected
Aborted (core dumped)
```

I dont think this is snap specific as this involves app threading and clocks.
--> https://bugs.launchpad.net/webcamoid-snap/+bug/2027853

This only happens initially second time this doesnt happen but the video isnt either recorded properly with the following error ;
```
Error pushing buffer to GStreamer pipeline
```

capecrusader-121, 2023-07-15:
I think ffmpeg is the best option still this should be fixed after i added more gstreamer plugins.

hipersayanX, 2023-07-19:
I usually prefer FFmpeg instead of GStreamer because it's a lot much easier to package and to setup the pipeline. I have included GStreamer as an option just because some distributions (like Fedora and OpenSUSE) did not provided FFmpeg in their repositories, because possible problems with patents.

capecrusader-121, 2023-07-19:
@hipersayanX  is there any build flag to disable it completely ? if not do provide an option to disable it altogether, its a good riddance.

hipersayanX, 2023-07-19:
**NOGSTREAMER**, all major flags are located [here](https://github.com/webcamoid/webcamoid/blob/40acd570b8138cd3ba9348cce305ad455619767d/libAvKys/cmake/ProjectCommons.cmake#L35).

capecrusader-121, 2023-07-20:
@hipersayanX  I did try that but its termed unrecognized option/flag.
I added it it this way
```
          - -NOGSTREAMER
          - -NOMEDIAFOUNDATION
          - -NONDKCAMERA
          - -NONDKMEDIA
          - -NOVLC
          - -NODSHOW
          - -NOLIBUVC
```

closed 2023-07-20 by capecrusader-121