webcamoid/webcamoid #599: Hangs, then crash

Mrcubix, 2023-07-19:
# Reporting a problem?

yes

## Summary

App seems to crash on launch.

## Expected Behavior

What's supposed to be the expected behavior if the app does not work?

## Steps to Reproduce

- Download the last release (9.1.11) of the app, 
- Download the last release of the driver, 
- install the driver,
- Try starting the app,
- The windows shows up and cache seems to get filled, but crash instantly then after.

## Your Environment

* Webcamoid information: 9.1.11
* Operating System information: Windows 10 22H2 19044.3086
* Any other useful information:

GDB Logs: https://pastebin.com/RmGNjT4V

hipersayanX, 2023-07-19:
It keep crashing over and over after installing the virtual camera? if you uninstall the virtual camera, it works again?

Mrcubix, 2023-07-20:
it didn't work without, why would it work with, (and yes i tried both)

alcosml, 2023-07-26:
In my case, the win32 build kept crashing after creating the virtual camera output.
The win64 output also crashed after selecting video source (another virtual camera). I couldn't 'reset' it even after reinstalling - had to delete 'webcamoid' keys from the registry to fully 'reset'. From there, I was able to reinstall, select a working video source, select effects, create a virtual camera, and use its virtual camera instance on another app.