webcamoid/webcamoid #607: Please provide this via F-Droid.

RokeJulianLockhart, 2023-12-15:
# Want a new feature?

Some rationale is available at https://github.com/Tadashi-Hikari/Sapphire/issues/2#issue-1228766021.

However, I realize that the current state of this software when compiled for Android is objectively inadequate (as https://github.com/webcamoid/webcamoid/issues/169#issuecomment-1013526028 demonstrates). Consequently, addition of it to https://f-droid.org despite that would allow users of Webcamoid that potentially do not want to manually install each new version (since https://github.com/webcamoid/webcamoid/releases/download/daily-build/webcamoid-android-daily-master-1.apk demonstrates that the code currently compiles to `.APK`) to evaluate it easily.

RokeJulianLockhart, 2024-04-21:
> https://github.com/webcamoid/webcamoid/issues/607#issuecomment-1857844381

@hipersayanX, that doesn't assist anyone using https://calyxos.org/install/, for instance, who isn't able to utilize https://www.apkmirror.com/apk/google-inc/google-play-store/google-play-store-40-6-28-release/google-play-store-40-6-28-31-0-pr-625422852-android-apk-download/download/?forcebaseapk=true, especially since https://github.com/webcamoid/webcamoid/releases/tag/9.1.1 doesn't contain any release APKs, meaning that they can't even use https://github.com/ImranR98/Obtainium/blob/85718dc3a37d12ce13f596c950a8fb5c2b484df6/README.md#-obtainium.

hipersayanX, 2024-04-23:
@RokeJulianLockhart

I like F-Droid and I use it but, the first reason for rejecting the F-Droid package is because as I explained in my previous message, F-Droid force my to compile EVERYTHING from scratch, and Webcamoid depends on 2 really big libraries: Qt and FFmpeg, both requiring hundreds of dependencies for building ([Qt](https://aur.archlinux.org/packages?O=0&SeB=nd&K=android-aarch64-qt6&outdated=&SB=p&SO=d&PP=50&submit=Go), [FFmpeg](https://aur.archlinux.org/packages?O=0&SeB=m&K=hipersayan_x&outdated=&SB=p&SO=d&PP=50&submit=Go)), and the time for maintaining the build tree just for supporting F-Droid is absurd.
Seconds, Google Store allow me to monetize Webcamoid, F-Droid and others doesn't.
Third, Webcamoid is still [Free Software](https://github.com/webcamoid/webcamoid/blob/master/COPYING), anyone can create a F-Droid package if they want, of course I can help the maintainer but the maintainer and it's users are on their own, I fix only the bugs in my software.

RokeJulianLockhart, 2024-04-23:
> https://github.com/webcamoid/webcamoid/issues/607#issuecomment-2071286001

@hipersayanX, and (as https://github.com/webcamoid/webcamoid/issues/607#issuecomment-1857844357 mentions)  https://f-droid.org/en/tutorials/create-repo/ would be undesirable too?

hipersayanX, 2024-04-24:
I have already answered just bellow the comment you are citing.

RokeJulianLockhart, 2024-05-24:
> https://github.com/webcamoid/webcamoid/issues/607#issuecomment-2075603085

@hipersayanX, yes - apologies.

RokeJulianLockhart, 2025-04-19:
> https://github.com/webcamoid/webcamoid/issues/607#issuecomment-1857844381

@hipersayanX, is Webcamoid on the Play Store (yet)? I don't see it (at least, with my region set to the UK). Your payment method makes adding an APK source to Obtanium infeasible, so without F-Droid or Play Store, updating would be quite a chore.

hipersayanX, 2025-04-25:
@RokeJulianLockhart I was supposed to release in the Play Store in the latest version, but there were some some bugs with newer versions of Android and things that i was not happy with it yet to officially release it. i have almost solved all bugs in the daily build, so the next version for sure will be available there.