webcamoid/webcamoid #647: Add blur background

jaymendez, 2023-12-15:
Similar to other video conference tools, where they can blur the background and also adjust the blur strength.

henricook, 2025-03-07:
I know that +1s aren't ideal, but I felt that 4.5 years was long enough to add mine. I bought Webcamoid today and it's a very vanilla usage but all I was actually looking for was a slightly customisable bokeh effect that's less blurry in the background than MS Teams :-D

I'm still happy to have contributed but beyond the cool Kung Fu fighting setting I don't think I have much of a use for what I've bought (I could of course have done more research before buying but thought I'd just try it out and see).

hipersayanX, 2025-03-07:
@henricook There was only one person in this in this issue asking for this feature, and there were other priorities. I you want, I can start working in this issue right now if you set a good price in [Bountyhub](https://bountyhub.dev/bounties?repo=webcamoid/webcamoid) 😃 , my guess is that it will take at least between 1 to 2 moths to develop since I must test it in Android, GNU/Linux, and Windows, so imagine you were a developer and how much would you want to receive for that time of work.

henricook, 2025-03-08:
@hipersayanX I'm very sorry this came off that way, but it wasn't an attack or complaint. I completely understand and sympathise with the trade offs of project development, and just wanted to register my interest in this. Secretly I also hoped it might be one of those posts where someone responded and said "what are you talking about dude background blurring is possible with <this thing that I'd overlooked>" :-D

hipersayanX, 2025-03-09:
@henricook Right now I have to prioritize what increase to income and fix urgent bugs, and that I'm the only developer of the project, if more people [donated](https://webcamoid.github.io/donations), more people [become sponsors](https://github.com/sponsors/hipersayanX), the [sellings increase](https://payhip.com/Webcamoid), there will be more resources to hire more people to work in the project, I can also buy better hardware for work, right now I need at least a new laptop because I'm working in a Core 2 duo 15 years old laptop, for example you can donate at least 10% of you income which is not too much for you.
Right now my only priority is to launch it in Google Play Store so I can monetize with ads for those who want it for "free" which should give me another source of income. Companies sponsoring the project are also welcome.
The equation is simple: the more you pay the more you get.
Also, imagine that you were working in your own company, will you put all your attention in a heavily demanded issue with lots of comments and that probably increase the earning? or will you put your attention in a 4 years old issue with 2 comments?
Again, if you (or anyone else) want this feature you can pay to develop it, it does not represent to much money for you but represent a lot of money to me.