webcamoid/webcamoid #660: Please support RAW.

RokeJulianLockhart, 2023-12-15:
# Want a new feature?

Please support the "http://wikipedia.org/wiki/Raw_image_format" because the format allows retainment of approximately all of the information that was provided to what created the photograph, as "http://adobe.com/creativecloud/file-types/image/raw.html" states.

I suggest that implementation of this should utilize "http://helpx.adobe.com/camera-raw/digital-negative.html", whose file-extension is “`.DNG`”, so that the information is simultaneously standardized and extensible so that parsal and archival are more easy than they alternatively would be.

# Questions and other matters?

Approximately all professional software supports this format, which even includes software that is designed for mobile devices, as "http://reddit.com/comments/4uh4v5 demonstrates", so I believe that the lack of this ability within software that is as impressive as Webcamoid is weirdly unusual. The software that I have provided as demonstration appears to utilise "http://helpx.adobe.com/camera-raw/digital-negative.html".

I have not been able to locate how to activate this ability if Webcamoid supports it, potentially because I have solely evaluated "http://github.com/webcamoid/webcamoid/releases/download/daily-build/webcamoid-android-daily-master-1.apk", so I apologize if this is unnecessary.

hipersayanX, 2026-11-01:
Claude answer on this:

>Thanks for the suggestion. Webcamoid is designed for live video and streaming, not for still photography only. Webcams and capture devices deliver already processed frames, not RAW sensor data, and on top of that every captured frame goes through an OpenGL processing pipeline, so a RAW frame is never available at any stage. Saving RAW/DNG would also require using camera-specific APIs of each platform (for example Camera2 on Android), which is outside the project’s goals. For RAW photography, a dedicated photo application is the right tool. Closing as out of scope.

Closing.

closed 2026-11-01 by hipersayanX

beedell-roke, 2026-11-01:
In which case, if anyone else sees this, follow [`powerusers.codidact.com/posts/294945`](https://powerusers.codidact.com/posts/294945).