webcamoid/webcamoid #667: About MacOS support

hipersayanX, 2023-12-15:
About giving support to future MacOS versions, I've some things to say...

I will be clear, I will try to give MacOS support as long as possible, but at this point I still working with Mac Catalina that is the latest working version for me and can't test newer versions because none of them works, forget about buying a Mac because it's out of budget.

I will start migrating native MacOS code to third-party libraries, so for example instead of having a specific plugin for [CoreAudio](https://github.com/webcamoid/webcamoid/blob/440280381d67e142153fca2b4c7be6f99d37e56f/libAvKys/Plugins/AudioDevice/src/coreaudio/src/audiodevcoreaudio.cpp#L34), it will delegate the work to [SDL](https://github.com/webcamoid/webcamoid/blob/440280381d67e142153fca2b4c7be6f99d37e56f/libAvKys/Plugins/AudioDevice/src/sdl/src/audiodevsdl.cpp) or [PortAudio](http://www.portaudio.com/).

# Advantages

- I only need to test once in Linux.
- Any bug report in Mac will be delegated to those third-party libraries.
- I won't to constantly depend on a Mac developer to maintain the plugin, developers usually come and goes.
- Less work to do.

# Disadvantages

- Third-party usually lack a lot of features or does not works as expected, for example [SDL](https://github.com/Rust-SDL2/rust-sdl2/issues/901) which does not works in MacOS, or [PortAudio](https://github.com/PortAudio/portaudio/wiki/HotPlug) which is missing HotPlug support yet.
- Blind development, if no other developers step-up, I will depend on users to tell me if a fix works or not.

This changes could give a bit more time to Webcamoid in Mac, but do not expect to last long. You know what to do, step-up or let it die.

lalaRLH, 2024-09-04:
Have you considered a little hackintosh eg I have a HP Elitedesk 800 G2 35w miniPC. Most are intel wifi, but some are not, so just double check it's got intel card if you want to use wifi / bluetooth. 

I have compiled in full all drivers etc (kexts etc) required for it, and it runs latest macOS. The unit validates with apple for iServices and runs everything fine other than sleep (due to macOS not supporting the same sleep modes as the HP BIOS /  firmware aimed at windows). Else, you have a great little box - it's really a Mac mini. They sell anywhere from 50 to 100 bucks on eBay. I use it over my Mac mini as it's faster (due to better ssd etc I'd assume). I'd happily share with you the EFI (where drivers etc are loaded from - a little partition of 200 or so mb), and even walk you through it as required. The hardest bit is generating a bootable USB but that's easier now given you can generate online installers via Windows (eg it's only approx 500mb and then downloads macOS Sonoma via wifi and installs). 

How are you running the current macOS? Is it virtualised? Sonoma works on VMware Workstation, which is free now via Broadcom. You need to either use an Unlocker that "unhides" the hidden macOS menus, or manually amend it. Then you're away. 

I want to volunteer but legit I am swamped. I just know I'd be unreliable in the longer term. 

+ people need to figure out how to type a string into command line, plenty of apps are not notarised etc and arm users run the apps. Anyway, it's not worth the headache / education piece to be accused of being the latest and greatest info stealer :)

hipersayanX, 2024-09-04:
>Have you considered a little hackintosh eg I have a HP Elitedesk 800 G2 35w miniPC. Most are intel wifi, but some are not, so just double check it's got intel card if you want to use wifi / bluetooth.

Yes, but I haven't not money for buying a new PC right now, not even a PC capable of running MacOs. Payhip income is just enough to pay bills and a bit more, and a very few people have ever donated, and only one person have donated 25$ for MacOs support.

>How are you running the current macOS? Is it virtualised? Sonoma works on VMware Workstation, which is free now via Broadcom. You need to either use an Unlocker that "unhides" the hidden macOS menus, or manually amend it. Then you're away.

I was running MacOs in VirtualBox, and it was working more or less fine until (if I remember) Monterrey, then I was not able to install newer versions any more. Now, the PC in which I was working is working is broken (probably need reballing, not sure), and now I'm using a much older computer without virtualization support. [And there are still](https://github.com/webcamoid/webcamoid/issues/697) [people who thing](https://github.com/webcamoid/webcamoid/issues/687) [I must give my work for free](https://github.com/webcamoid/webcamoid/issues/695).

> I want to volunteer but legit I am swamped. I just know I'd be unreliable in the longer term.
> - people need to figure out how to type a string into command line, plenty of apps are not notarised etc [...]

MacOs binaries are right now being uploaded to the private repository, all sponsors can get access to it.

AnaJuliaCodes, 2024-10-09:
Can't you just make prebuilt binaries free like any other person would do?

hipersayanX, 2024-10-09:
@AnaJuliaCodes people download the program, < 1% of the people who downloaded the program informs about any bug, the rest just try and discard :man_shrugging:
Also,

1. I need money for living
2. With monetary support I can buy new hardware for, for example, buying a Mac for testing it my self.
3. Paying $2 for having access to the private repository is not to much and Mac users can afford that without any problem.

hipersayanX, 2025-03-07:
Added a [GoGetFunding campaing](https://gogetfunding.com/?p=9088892) for buying a MacBook but it will probably never get filled.