webcamoid/webcamoid #668: Complete failure to record video

davidgiven, 2023-12-15:
## Summary

I just used Webcamoid to record a bunch of video for a project. While collecting all the video files together for editing, I discover that all the files that Webcamoid recorded are empty, meaning I've lost all my video.

## Current Behavior

As above. I've tried this with both a USB HDMI capture card and a webcam. Webcamoid produces the same empty files each time.

## Expected Behavior

Files should contain video?

## Steps to Reproduce

1. Open Webcamoid.
2. Connect to video source. Notice that the video is displayed in the window.
3. Press the 'record' button. Notice the flashing 'recording' indicator with the increasing timer.
4. After a while, press the 'stop' button. Notice that a file is generated.
5. The file is always 264 bytes long and contains no video.

## Your Environment

* Webcamoid information: 9.0.0 amd64
* Operating System information: Debian testing

Enclosed is one of the generated files.

[Video 2023-03-25 19-36-48.webm](https://user-images.githubusercontent.com/100638/227735696-00d4fc46-c34f-4213-8e09-a15565ee0483.webm)

Pac000rrr, 2024-01-30:
Same here on Windows 10, and 9.1.1 precompiled binary version. It just creates a 257 bytes empty file but no video.

Thank you

wmbm, 2024-03-15:
Same for Ubuntu version 9.1.1. file size is 258 bytes

rodrigopedra, 2024-10-01:
@hipersayanX I can confirm the newly released version (9.2.0) fixes this issue.

Just compiled it and tested on openSUSE Leap 15.6

Thanks a lot!

hipersayanX, 2024-10-01:
Fixed in 9.2.0 release.

closed 2024-10-01 by hipersayanX