webcamoid/webcamoid #669: v9.1.10 on windows 10 locks up (not responding) when changing video source

neuronetv, 2023-12-15:
# Reporting a problem?
yes
## Summary
as subject line
Write here a brief description of the problem.
as subject line
## Current Behavior
same
Describe the problem the best as you can, don't omit information.
as subject line
## Expected Behavior
should change sources without lock up
How it should have work?
should not get 'not responding' error
## Steps to Reproduce
change video source
If Webcamoid crashed, try to reproduce the crash several times to be sure where is the problem. Write the steps to reproduce the issue below:
n/a
1.  Open this
2.  Click that
3.  Drag those
4.  ...

## Suggestions and tips
none
How would you solve the problem?
v9.1.10 is not useable
## Your Environment
windows 10 v21h2  build 19044,2486
* Webcamoid information: (version) (architecture)
* Operating System information: (name) (numeric version and codename if applied) (architecture)
* Any other useful information: (logs, gdb backtrace, valgrind logs, screenshots, hardware, etc.)
64 bit
# Want a new feature?
no
Describe your idea the best as you can, include sketches, mockups and diagrams if required.  
Be patient, take in mind that there may be other priorities. Your idea will be accepted if fit in project goals.

# Questions and other matters?

Go ahead!

hipersayanX, 2024-10-01:
Fixed in 9.2.0.

closed 2024-10-01 by hipersayanX

neuronetv, 2024-10-02:
where's the download for windows?
I went to https://webcamoid.github.io/#downloads and clicked
on AkVirtualCamera driver download but I got some github page with loads of
strange stuff on it. I couldn't find a download for windows anywhere

hipersayanX, 2024-10-02:
@neuronetv [here](https://payhip.com/b/YjJro) is the download for Windows.

neuronetv, 2024-10-02:
ok thanks