webcamoid/webcamoid #670: Hangs, then crash on startup

Mrcubix, 2023-12-15:
# Reporting a problem?

yes

## Summary

App seems to crash on launch.

## Expected Behavior

What's supposed to be the expected behavior if the app does not work?

## Steps to Reproduce

- Download the last release (9.1.1) of the app, 
- Download the last release of the driver, 
- install the driver,
- Try starting the app,
- The windows shows up and cache seems to get filled, but crash instantly then after.

## Your Environment

* Webcamoid information: 9.1.1
* Operating System information: Windows 10 22H2 19044.3086
* Any other useful information:

GDB Logs: https://pastebin.com/RmGNjT4V

karobelle, 2024-02-27:
@alcosml where are the webcamoid keys you deleted located? 

I am also crashing on win64 both install and portable. When I select OBS Virtual Webcam as the input source, it crashed and crashes every launch after that.

alcosml, 2024-02-28:
@karobelle, I found them through [RegScanner](https://www.nirsoft.net/utils/regscanner.html) by searching for "webcamoid" under HKCU & HKLM:
- HKCU\SOFTWARE\Webcamoid
- HKLM\SOFTWARE\Webcamoid

hipersayanX, 2024-10-01:
Fixed in 9.2.0.

closed 2024-10-01 by hipersayanX