webcamoid/webcamoid #677: I downloaded Installer for x86_64 CPU to Linux Mint. Double clicking on the file named 'unconfirmed896649.crdownload gives 'unknown file type' error. Please  help with this update to Webcamoid.

wildad1, 2024-01-10:
# README

Some considerations before doing anything:

* Search the [issues list](https://github.com/webcamoid/webcamoid/issues) for similar topics before opening a new one.
* If you know how to fix the problem, consider doing a [pull request](https://github.com/webcamoid/webcamoid/pulls) instead of opening a new issue.
* Only report problems in [latest](https://github.com/webcamoid/webcamoid/releases) and [development](https://github.com/webcamoid/webcamoid/) version. Reporting issues related to older versions will be rejected.
* Use [gist](https://gist.github.com/) to post logs longer than 1024 characters.
* Webcamoid as project, is not affiliated or endorsed to any distribution, report packaging problems in their respective issue tracker.
* Respect the templates, we need as much information as possible.
* Don't open an issue and disappear, we need you at least the first week to clear up missing information.
* Missing information makes useless and unsolvable an issue report.
* You are our debugger, eyes and hands, if an issue is not reproducible then you will be the only person able to solve it.
* Take a seat and wait for your turn, as many others that has their issues open, or much better try fixing it your self and collaborate with the solution :smile:

Choose one of the templates below that fit better your issue.

# Reporting a problem?

## Summary

Write here a brief description of the problem.

## Current Behavior

Describe the problem the best as you can, don't omit information.

## Expected Behavior

How it should have work?

## Steps to Reproduce

If Webcamoid crashed, try to reproduce the crash several times to be sure where is the problem. Write the steps to reproduce the issue below:

1.  Open this
2.  Click that
3.  Drag those
4.  ...

## Suggestions and tips

How would you solve the problem?

## Your Environment

* Webcamoid information: (version) (architecture)
* Operating System information: (name) (numeric version and codename if applied) (architecture)
* Any other useful information: (logs, gdb backtrace, valgrind logs, screenshots, hardware, etc.)

# Want a new feature?

Describe your idea the best as you can, include sketches, mockups and diagrams if required.  
Be patient, take in mind that there may be other priorities. Your idea will be accepted if fit in project goals.

# Questions and other matters?

Go ahead!

hipersayanX, 2024-01-11:
The installer file name is ```webcamoid-installer-linux-9.1.1-x64.run```, your download was not completed, download it again.

wildad1, 2024-01-11:
I finally got a complete download, but when I double click I still get 'unknown file type' message.

hipersayanX, 2024-01-11:
Check the MD5 and the file size:

```
$ md5sum webcamoid-installer-linux-9.1.1-x64.run
2529a5dbde34e67a17d2792e61d8b5e3  webcamoid-installer-linux-9.1.1-x64.run
$ stat --format=%s webcamoid-installer-linux-9.1.1-x64.run
118803034
```

it should give the same result as me.

wildad1, 2024-01-11:
Well, if I have to work that hard to add an update, I will just forget it. Should not be necessary. Thanks for your comments.

hipersayanX, 2024-01-11:
With the context you are giving me, I understand that the file you downloaded is or was broken (maybe because your internet connection) that's why I'm asking you to run those commands, there is no other way of solving the problem.

closed 2024-10-01 by hipersayanX