webcamoid/webcamoid #678: Output resolution does not change when source resolution changes

qxc86, 2024-01-14:
# README

Some considerations before doing anything:

* Search the [issues list](https://github.com/webcamoid/webcamoid/issues) for similar topics before opening a new one.
* If you know how to fix the problem, consider doing a [pull request](https://github.com/webcamoid/webcamoid/pulls) instead of opening a new issue.
* Only report problems in [latest](https://github.com/webcamoid/webcamoid/releases) and [development](https://github.com/webcamoid/webcamoid/) version. Reporting issues related to older versions will be rejected.
* Use [gist](https://gist.github.com/) to post logs longer than 1024 characters.
* Webcamoid as project, is not affiliated or endorsed to any distribution, report packaging problems in their respective issue tracker.
* Respect the templates, we need as much information as possible.
* Don't open an issue and disappear, we need you at least the first week to clear up missing information.
* Missing information makes useless and unsolvable an issue report.
* You are our debugger, eyes and hands, if an issue is not reproducible then you will be the only person able to solve it.
* Take a seat and wait for your turn, as many others that has their issues open, or much better try fixing it your self and collaborate with the solution :smile:

(I read all of that :) )

# Reporting a problem?

## Summary

Output resolution doesn't change if the source resolution changes.

## Current Behavior

Video output when recording or when being pulled from `/dev/video1` (webcamoid's output) stays 640x480, my webcam's default resolution, even if I change the resolution of the source before using the output. I did confirm that changing sources (to a video file) does change the output resolution, but when I re-open the webcam source it opens in 640x480 again, and the output gets locked back to that.

## Expected Behavior

Output should ideally change resolutions as the source resolution changes at any time, but at least when no output or recording is being done. 

## Steps to Reproduce

1. Open source with multiple resolutions
2. Confirm output resolution (I opened the device with ffplay, recording also works), then close any software using the output device file or stop recording
3. Change source resolution
4. Repeat step 2, will still be original resolution. 

## Suggestions and tips

This was previously fixed per the 8.5 patch notes as well as this bug report closed on 9.0 https://github.com/webcamoid/webcamoid/issues/379

Hopefully an easy fix. If I could force the webcam to open in a different resolution initially maybe I could get by, but I don't know how to do that. 

## Your Environment

* Webcamoid information: 9.1.1 amd64
* Operating System information: Devuan testing (Excalibur) amd64 (using packages from Debian Sid)
* Any other useful information: Only output I get is the following on starting the program:

```warning: qrc:/Webcamoid/share/qml/VideoOutputPicture.qml:81:13: QML Image: Protocol "" is unknown```

Looking in the file listed, I don't think it's related.

hipersayanX, 2024-01-15:
# For the video

Most video codecs only support compressing the video frames at a fixed resolution (ignoring possible hacks or extensions), so saving video frames of different resolutions in the same file is not possible, when you start recording, Webcamoid will take the resolution of the source currently in use, and weep it until you stop the recording, if you want to record at a different resolution, stop the current recording and start a new one.

# For the virtual camera

You can't change the video resolution while it's being in use because it's not technically possible, if you want a different resolution in the client program you have to close the client program, change the source, and open the client program again.
To clarify it, let's remove Webcamoid from the equation, try the following:

Create a virtual camera:

```sudo modprobe v4l2loopback video_nr=2 card_label="Virtual Camera"```

Play a video and send it to the virtual camera:

```ffmpeg -re -i Video1.mp4 -vcodec rawvideo -pix_fmt yuv420p -f v4l2 /dev/video2```

Use the client program to capture the virtual camera:

```ffplay /dev/video2```

Cntrl+C ffmpeg while the client client program is still capturing the frames. The client program will get stuck with the last frame captured, and waiting for more.
Now play a different video with a different resolution (shouldn't be the same resolution as Video1).

```ffmpeg -re -i Video2.mp4 -vcodec rawvideo -pix_fmt yuv420p -f v4l2 /dev/video2```

The result is that the client program won't get the frames of Video2, and still will get stuck with the last frame of Video1, because the video capture format is always fixated in the client, and this is not neither a bug in the client, the driver, or the source program, and obviously not a bug in Webcamoid.
Also, when using akvcam as driver, neither Webcamoid or akvacm can't change the behavior of the  client program, Webcamoid configures the default resolution to 640x480, but also offers higher resolutions, and others resolutions can be manually added too. If you want the client program to capture in a different resolution, just select the resolution that you want in the client program.

[image]

# In resume

The issue at #379 should have been fixed taking into account the previous considerations.

hipersayanX, 2024-01-15:
>- lsof /dev/video* with webcamoid stopped shows nothing using the webcam video device nor the virtual camera.
>- With webcamoid started, I run lsof again and see it accessing both the input and output devices, each two different times. Each has one with read write (two numbers followed by u) file descriptor, and another with mem file descriptor. This is the behavior I would expect with nothing using the virtual camera, thus it should be able to change resolutions.
>- At this point I change the resolution, then open the virtual camera device with ffplay, and it's still in 640x480. lsof confirms ffplay with two file descriptors opening /dev/video1, as expected.

As I understand, you tested it with v4l2loopback, right? You are missing another test, try running this command without Webcamoid running:

```
ffplay /dev/video2
```

this will be the output:

```
[video4linux2,v4l2 @ 0x7f05cc000c80] ioctl(VIDIOC_G_FMT): Invalid argument
/dev/video2: Invalid argument
```

And this is because v4l2loopback does not define any format if the source is not sending any frame, and the resolution can't be changed on fly because v4l2loopback is a pass-through driver. So that, once the source playback has started v4l2loopback plugin will take and keep the resolution of the output even if you switch to another source, and it will be like that until you stop the playback, switch to the wanted source, then playback again. If I did not have programmed like that, the client would have experienced the same behavior explained in the previous post.

>Recording still only outputs 640x480 unfortunately, and I'm not starting the recording until after the resolution has been changed.

Recording should take the resolution of the source when started, if it does not do that, it's probably a bug.

>For what it's worth, some settings like the input device resolution settings for the webcam, as well as the output device scaling mode, don't seem to save between program restarts. This may be intended behavior or a different bug.

More like I did not programmed that functionality yet because is not prioritary for now, I'm more focused on getting the Android APK working smooth (for attracting more users), or moving video processing to Vulkan or OpenGL (for performance improvement), or improving the plugin system (for attracting more developers).

>I've also had some issues changing output device settings under the "edit" submenu because I get an error saying another process is using the device, but the process is the webcamoid process I'm already in. This was happening before as well but I assumed it was unrelated, but since it's happened with both virtual camera drivers I figured I'd mention it in case it's relevant.

If I remember well this bug was probably solved in the daily build. And yes, this is because it is detecting Webcamoid as a client.

>Unfortunately akvcam seems to be a bit buggy on my system and occasionally it will hang any program that tries to grab the virtual camera device. This happened once before trying to run two webcamoids, for the record. Will keep messing with it for the time being.

I have fixed many bugs with latest release, yet it still may need some more, and I still need to optimize the code a lot.

>Sorry if all of this isn't clear, I'm writing and troubleshooting at the same time and doing my best to go back and try to make it comprehensible afterwords. :)

Don't worry, I understand that all this topic is complex to explain and understand.

qxc86, 2024-01-16:
>So that, once the source playback has started v4l2loopback plugin will take and keep the resolution of the output even if you switch to another source, and it will be like that until you stop the playback, switch to the wanted source, then playback again.

Ahh I found an interesting behavior of this.

- Open webcamoid, set source to not 640x480 (Screen 0)
- ffplay virtual camera, outputs expected resolution (screen resolution)
- stop ffplay
- change source in webcamoid to webcam
- start ffplay
- webcam at 640x480 (as expected)

But if I don't stop ffplay when changing sources, v4l2loopback gets messed up and stuck at the last resolution and I have to reload the driver for it to change resolutions. When it tries to change resolutions it stops playing back in ffplay, as expected, but it stays in an error state after restarting ffplay. Resolution wrong, screen pink, etc. Not a webcamoid bug at all but just interesting.

Speaking of which, I was going to say "akvcam has been working fine since I rebooted it!" but at some point it got messed up and started locking anything that touched it and I had to reboot again. Unless I find a way to un freeze it I'll have to avoid using it for the time being, force stopping the module and even deleting the virtual camera devices didn't fix the programs after they hung, nor did `kill -9`. I'll have to get it set up on another system for testing, one I don't mind rebooting often. 

I hope the bugs can be worked out soon because the features are nice. Unfortunately I'm not a driver developer or I'd give it a shot. 

**Getting back on track a bit,**

>Recording should take the resolution of the source when started, if it does not do that, it's probably a bug.

- Open webcamoid, with webcam as default device (in 640x480)
- Set webcam to desired resolution in settings
- Start recording

Always ends up 640x480.

Changing the source to a screen input and back *does* change the recording resolution. I'm not sure why it was stuck at 640x480 doing the using-two-webcamoid-processes, using virtual camera as input device to get the correct resolution. I know the resolution going into the second webcamoid process was not 480p, and the output from the second virutal device was not 480p. Maybe switching to a screen capture and back would have done it, but I didn't test that while akvcam was working. 

I did notice that if I change the device resolution in webcamoid, change input devices to screen 0 so the webcam device is no longer being used, and then ffplay the real webcam device, it does open in the last resolution I set it to. Not 640x480. This is to say, webcamoid is changing it to 640x480 every time it opens it. I think if it opened in the last saved resolution it might record and output at that as well, perhaps after restarting the program or something. The problems right now seem to revolve around everything setting to 640x480 when the webcam is opened and not changing, even when they are in a state in which it should be safe to do so. Not recording, output device not being used. 

I'll probably have to find another workaround for the time being. I can't figure out any way to get the output device to not end up 640x480 with the v4l2loopback driver and the webcam as the source device. Let me know if there's anything else I can test on my side.

hipersayanX, 2024-01-17:
> Speaking of which, I was going to say "akvcam has been working fine since I rebooted it!" but at some point it got messed up and started locking anything that touched it and I had to reboot again. [...]

Are you using the latest version? Result of ```uname -a```?

> - Open webcamoid, with webcam as default device (in 640x480)
> - Set webcam to desired resolution in settings
> - Start recording
>
>Always ends up 640x480.

Ok, will check then.

>Let me know if there's anything else I can test on my side.

You are going a lot much faster than the questions I can have :smile:

qxc86, 2024-01-21:
Kernel 6.5.0-4-amd64 and akvcam 1.2.4. Had to look at the akvcam install logs to get the version.

For what it's worth I found OBS can do what I need for my particular use case. Webcamoid is nicer UI wise but I wasn't making use of the plugins or other Webcamoid specific features. Also happens to help out my dependency situation, but that's an issue entirely of my own doing.

isAAAc, 2024-03-19:
Hello, o/ 

i installed the appimage yesterday, `webcamoid-portable-linux-9.1.1-x64.AppImage` on debian 12,  
and i'm encountering the same issue, 
local output looks ok, but when loged on bbb, resolution is wrong,  
NB: resolution is perfect if i don't use webcamoid...

any fix in next weeks ?

hipersayanX, 2024-03-20:
Maybe a solution would be allow to set a fixed (but configurable) output resolution for both the recording and the virtual camera? would that be ok?
But the problem of doing that is that it will impact performance and image quality seriously. If for example, if you set an output resolution of 1280x720, and the actual resolution of the camera is 640x480, it will render slower in the other side, swap the resolution and and you will get a HD video rendered into a low resolution one loosing quality.

isAAAc, 2024-03-29:
yes it is ok on my side, thx for your help, works fine

hipersayanX, 2024-10-01:
The video recording resolution is now fixed independently from the source resolution, it can be configured from the preferences.

[image]

closed 2024-10-01 by hipersayanX