webcamoid/webcamoid #679: Doesn't start properly on bookworm raspberry pi5 (installed via apt install webcamoid)

mgrouch, 2024-02-04:
Doesn't start properly on bookworm raspberry pi5 (installed via apt install webcamoid).

Keeps drawing screen infinitely in smaller and smaller windows.

hipersayanX, 2024-02-05:
Does it detect the cameras?

mgrouch, 2024-02-05:
There is no cameras. I’m just starting the app

hipersayanX, 2024-02-05:
Then it started properly, if you have not any camera installed, it will capture the screen as expected.

mgrouch, 2024-02-05:
It goes into crazy loop of capturing screen 
and keeps scaling images smaller and smaller. 

Look horrible. Like two mirrors facing each other. 
I doubt it’s expected

hipersayanX, 2024-02-06:
You are capturing the screen with the window maximized, how would you expect it to look like then?

>Look horrible. Like two mirrors facing each other.

Superstitious thinking? :laughing:

mgrouch, 2024-02-06:
It never stops capturing. It looks like it hanged in a loop

hipersayanX, 2024-02-06:
Also, another funny effect to try: if you have an external camera, point to the screen were th Webcamoid window is located, and make Webcamoid capture from that camera, and turn the camera a bit, you will be able to open a portal to another dimension. :laughing:

hipersayanX, 2024-02-06:
>It never stops capturing. It looks like it hanged in a loop

If you think this is a bug, you can report the "bug" to [X.Org](https://www.x.org/), Microsoft, Apple, Google, and every single operating system implementing a desktop capture.

hipersayanX, 2024-02-06:
Also open an issue in the FFmpeg tracker, because they have the same "bug":

```
ffplay  -f x11grab -i :0.0
```

GStreamer is also "bugged" :worried: 

```
gst-launch-1.0 ximagesrc use-damage=0 ! video/x-raw,framerate=30/1 ! videoscale method=0 ! ximagesink
```

mgrouch, 2024-02-06:
Here is the video of the issue

https://www.youtube.com/watch?v=UWLMJyIh5EA

from another user

hipersayanX, 2024-02-06:
He is using an old version, and is not detecting the camera or he hasn't a camera connected. You said that you have not a camera connected to your Raspberry Pi, that's why it's capturing from the screen.
The screen capture loop is not a bug.

hipersayanX, 2024-02-07:
Other examples of infinite loops:

https://support.google.com/meet/thread/131558562/multiple-screens-are-appearing-as-infinite-loops-while-presenting-the-screen?hl=en
https://helpdesk.thelessonspace.com/article/131-infinity-mirror-effect-with-screen-sharing
https://stackoverflow.com/questions/76232073/python-infinite-window-loop
https://www.reddit.com/r/geek/comments/q2oib/teamviewer_makes_it_pretty_easy_to_create_an/
https://www.youtube.com/watch?v=K_AYxbpykMQ
https://www.youtube.com/watch?v=oZIKlpvsFZU
https://www.youtube.com/watch?v=DoW4KWEeKj0

mgrouch, 2024-02-07:
They are bad. For most of sane users it’s impossible to understand what is going on in UI when something like this shows up

hipersayanX, 2024-02-07:
You are not the first one asking for this :laughing:
All streaming programs with screen sharing support can do the same, and most users are not streamers so it may seem confusing at first :laughing: 
Some users even think this effect consume more memory or CPU when it isn't.

mgrouch, 2024-02-07:
It’s a very trippy effect. Makes it unusable for a non trippy user.

closed 2024-10-01 by hipersayanX