webcamoid/webcamoid #681: Bought app from official website and it's missing filter from the free version.

MarkoKurtovic, 2024-02-07:
So I bought app because free version keeps crashing (can't even start it now). 

And I really liked filter in free version.

I bought app from official website for linux and I'm missing this filter.

Image with filter:

[image]

Please can someone help me to find this filter in new app or advice me how to install free app without crashing.

hipersayanX, 2024-02-07:
All video filters are there. The one  you are looking for is probably the Photocopy effect.

[image]

MarkoKurtovic, 2024-02-07:
It's not same filter
[image]

MarkoKurtovic, 2024-02-07:
that one I didn't have to setup, just to click on it and it worked.

I tried manipulation photoCopy but it's not the same.

hipersayanX, 2024-02-08:
Then it should be the Black & white effect, and its there too.

[image]

MarkoKurtovic, 2024-02-08:
Unfortunately it's not.

it's just gray.
[image]

hipersayanX, 2024-02-08:
The only effect that gives a black and white result without options is that. Are you not sure its just a room illumination thing? because in the first image you have a strong light focus in your side, and that can change the contrast of the image.
Maybe, did you combined two effects? for example Black & white with Gamma?

MarkoKurtovic, 2024-02-09:
Ahh shit, I did use two effect by mistake probably.

Can't figure it out which ones, I'm trying for 1hour combinations

closed 2024-10-01 by hipersayanX