webcamoid/webcamoid #682: Virtual video output does not persist over reboot (VirtualCamera: Output formats were not configured)

Rufmord, 2024-02-15:
# Virtual video output does not persist over reboot

## Summary

If creating a virtual video output device it works but is removed after a reboot.

## Current Behavior

I create a virtual video output and use it for video conferencing software. After a reboot I open Webcamoid again and have to create the virtual video output again.

## Expected Behavior

The output device in Webcamoid should persist even after a reboot.

## Steps to Reproduce

1. Install Webcamoid from the AUR
2. Open it
3. Select Menu -> Video -> Outputs and add an output
3.1 remove all formats
3.2 add a format `yuyv422`
3.3 1920 x 1080
3.4 Press ok and enter root password
5. Reboot the system
6. Output should be gone

If launching Webcamoid via terminal (root and not root) this is the output
```
cat << EOF > /etc/akvcam/config.ini
[General]
default_frame=/etc/akvcam/default_frame.bmp

[Cameras]
cameras\1\description=Virtual Camera
cameras\1\formats=1
cameras\1\mode=mmap, userptr
cameras\1\type=capture
cameras\1\videonr=2
cameras\2\description=Virtual Camera (out)
cameras\2\formats=2
cameras\2\mode=mmap, userptr
cameras\2\type=output
cameras\2\videonr=3
cameras\size=2

[Connections]
connections\1\connection=2:1
connections\size=1

[Formats]
formats\1\format=YUY2
formats\1\fps=30/1
formats\1\height=1080
formats\1\width=1920
formats\2\format=RGB24
formats\2\fps=30/1
formats\2\height=1080
formats\2\width=1920
formats\size=2
EOF
modprobe akvcam

[2024-02-15 16:50:09.976, Webcamoid, 0x7bc08daa4d00,  (0)] debug: VirtualCamera: Output formats were not configured
```
In the `/etc/akvcam/config.ini` the output of the terminal execution matches the file content and even after a reboot the file content persists. But in Webcamoid the output device is gone.

## My Environment

* Webcamoid information: 9.1.1 (installed from AUR)
* Operating System information: Garuda Linux 6.6.16-1-lts (64-bit)

hipersayanX, 2024-02-16:
Does the same happens with webcamoid-git?

Rufmord, 2024-02-17:
Removed webcamoid (AUR) and installed webcamoid-git from hipersayan_x. Same problem, reboot removes output devices.

hipersayanX, 2025-07-29:
Should have been fixed in Webcamoid 9.3.0, closing.

closed 2025-07-29 by hipersayanX