webcamoid/webcamoid #684: Video capture either crashes or the video is unable to be played

KydeSion, 2024-02-26:
I've been trying to capture video using webcamoid, with a Nexigo N60 FHD Webcam.  I can hit record and it starts counting, when I hit stop it either hangs and then crashes or the file that it creates can't be played.  I checked the file sizes and they vary based on the length of the video.  I'm using VLC as my playback program, I've tried a bunch of the different video formats some of which cause the program to crash and some create the unviewable videos.  Not sure where to go from here...

hipersayanX, 2024-02-27:
Hi, which operating system are you using? are you using Webcamoid from your distro repositories or the official binaries? are you using the latest version?

hipersayanX, 2024-02-27:
Also try the basic setup: set the camera resolution at 640x480, video format MP4, video codec libx264, audio codec AAC, does it works?

KydeSion, 2024-02-27:
> Hi, which operating system are you using? are you using Webcamoid from your distro repositories or the official binaries? are you using the latest version?

Windows 10 and yes using the latest version

KydeSion, 2024-02-28:
Ok, I tried it with the basic setup you requested and when I hit stop recording the program crashes and I am left with a 1kb file that can't be played, VLC will open it but it doesn't have any video to play if that makes sense.

hipersayanX, 2024-02-29:
Sorry for the late response, will check, but its going to take some time because I'm very busy with the Android port right now.

TheFalldog, 2024-04-28:
Any update? Having the same issue as a new user.

KydeSion, 2024-04-30:
I tested with a Logitech webcam and it worked great, but testing with a second different model Nexigo webcam, it seems like it might have something to do with different devices or drivers

hipersayanX, 2024-05-01:
I did a quick test and for me its working, both capture and recording with the basic setup, did tried with both MP4 and Webm. Maybe could it be a problem of a capturing a non standard resolution? (resolution not multiple of 32, ie. 1367x***)

@TheFalldog I'm planning to make a new release near to June (if possible), but I don't guarantee that this issue will be fixed because I don't know what is causing it, yet.

hipersayanX, 2024-05-05:
I been thinking, does the same situation happens when recording the screen and setting the audio source to silent?

KydeSion, 2024-05-06:
I am able to record the screen without any errors or issues

hipersayanX, 2024-05-07:
@KydeSion Try recording the camera with audio source to silent. If that works, try setting Matroska/libx264/PCM signed 16-bit little endian, and set the audio source to the mic.

KydeSion, 2024-05-07:
Resulted in the same error when attempting to play back the video.

KydeSion, 2024-05-08:
Yes the audio is set to silence, the FPS shows 30

hipersayanX, 2024-05-08:
I have a slight idea what the problem could be. Just in case, you are using MediaFoundation for camera capture, right?

hipersayanX, 2024-05-26:
I did applied the changes to the master branch, that maybe, could fix this problem. Once I finish with the current changes, I will ask you to test it before release.

hipersayanX, 2024-06-07:
@KydeSion @TheFalldog send me the receipt ticket to hipersayan D,O,T x A,T gmail D,O,T com, so I can send the files for testing.

hipersayanX, 2024-10-01:
Fixed in 9.2.0.

closed 2024-10-01 by hipersayanX