webcamoid/webcamoid #687: Try B4 Buy?

gregoryd73, 2024-03-21:
# README

Some considerations before doing anything:

* Search the [issues list](https://github.com/webcamoid/webcamoid/issues) for similar topics before opening a new one.
* If you know how to fix the problem, consider doing a [pull request](https://github.com/webcamoid/webcamoid/pulls) instead of opening a new issue.
* Only report problems in [latest](https://github.com/webcamoid/webcamoid/releases) and [development](https://github.com/webcamoid/webcamoid/) version. Reporting issues related to older versions will be rejected.
* Use [gist](https://gist.github.com/) to post logs longer than 1024 characters.
* Webcamoid as project, is not affiliated or endorsed to any distribution, report packaging problems in their respective issue tracker.
* Respect the templates, we need as much information as possible.
* Don't open an issue and disappear, we need you at least the first week to clear up missing information.
* Missing information makes useless and unsolvable an issue report.
* You are our debugger, eyes and hands, if an issue is not reproducible then you will be the only person able to solve it.
* Take a seat and wait for your turn, as many others that has their issues open, or much better try fixing it your self and collaborate with the solution :smile:

Choose one of the templates below that fit better your issue.

# Reporting a problem?

## Summary

Write here a brief description of the problem.

## Current Behavior

Describe the problem the best as you can, don't omit information.

## Expected Behavior

How it should have work?

## Steps to Reproduce

If Webcamoid crashed, try to reproduce the crash several times to be sure where is the problem. Write the steps to reproduce the issue below:

1.  Open this
2.  Click that
3.  Drag those
4.  ...

## Suggestions and tips

How would you solve the problem?

## Your Environment

* Webcamoid information: (version) (architecture)
* Operating System information: (name) (numeric version and codename if applied) (architecture)
* Any other useful information: (logs, gdb backtrace, valgrind logs, screenshots, hardware, etc.)

# Want a new feature?

Describe your idea the best as you can, include sketches, mockups and diagrams if required.  
Be patient, take in mind that there may be other priorities. Your idea will be accepted if fit in project goals.

# Questions and other matters?

Go ahead!

gregoryd73, 2024-03-21:
Paid the $1.50 US and downloaded and installed only to find that it crashes on startup. Running on Windows10, i7 computer. It would be nice to try before buy ... you might consider that?
Thank you.

hipersayanX, 2024-03-21:
Hi, the README already explains the bug, try deleting or moving the file ```webcamoid/lib/avkys/libVideoCapture_dshow.dll``` to another folder.
I will fix this for 9.2.0, but its taking too long because first I want to fix some bugs with the Android version.

>It would be nice to try before buy ... you might consider that?

Let say you have a restaurant and you made a living from it as the only source of income, will you allow people to eat before paying with the hope that maybe a good samaritan will maybe give you some pennies?
Also 1, right now my economical situation is too delicate to give up my work for free :beer: 
Also 2, [there is the source code](https://github.com/webcamoid/webcamoid), users can compile Webcamoid for free.

gregoryd73, 2024-03-22:
The restaurant analogy is not quite the same thing. However, I agree you
should not give your work away from free. You could put a timer of 24 hours
on it maybe?
Tnx,
Dennis - WU6X

closed 2024-10-01 by hipersayanX