webcamoid/webcamoid #691: Crash on startup on sonoma

theotheroracle, 2024-04-11:
## Summary

Crash on startup on sonoma

## Current Behavior

When trying to launch the application, i get a popup saying the application crashed . 

## Expected Behavior

It should have started up successfully .

## Steps to Reproduce

If Webcamoid crashed, try to reproduce the crash several times to be sure where is the problem. Write the steps to reproduce the issue below:

1.  Build it following the instructions from this page: https://github.com/webcamoid/webcamoid/wiki/Build-and-install-in-Mac
2.  Launch the .app that was created 
3.  The app crashes

## Suggestions and tips

I'll try to do some investigation into the issue myself, but i'm not an experienced macos developer .

## Your Environment

* Webcamoid information: (9.1.1) (x86)
* Operating System information: (MacOS) (14.4.1 (23E224)) (x86)
* Any other useful information: (logs, gdb backtrace, valgrind logs, screenshots, hardware, etc.)
https://gist.github.com/theotheroracle/378e6e1e7b531a7f1c433fd60ad9538d - crash
https://gist.github.com/theotheroracle/2ade5f7fbdb9d16479851c42075e1d8e - build-log

theotheroracle, 2024-04-27:
i ran the deploy script, launched the webcamoid-portable app, and got this crash:
https://gist.github.com/theotheroracle/557f2ff75f7dc05da96c499b7c71444c

hipersayanX, 2024-04-27:
Does the bundle have these files?

```
Webcamoid.app/Contents/Plugins/platforms/libqcocoa.dylib
Webcamoid.app/Contents/Plugins/platforms/libqminimal.dylib
Webcamoid.app/Contents/Plugins/platforms/libqoffscreen.dylib
```

closed 2024-10-01 by hipersayanX