webcamoid/webcamoid #692: Video is recorded but not saved properly

sameersharma2006, 2024-04-12:
Hey there snap maintainer here, i have opened this to seek some help regarding an issue that pests the snap builds.

Video is captured successfully but are saved as empty files having size of 0-15kb.

I have staged gstreamer & ffmpeg.

No specific logs indicate anything related to this.

Snap bug-tracker issue: https://gitlab.com/sameersharma2006/webcamoid-snap/-/issues/1

hipersayanX, 2024-04-12:
Try uncommenting  [this line](https://github.com/webcamoid/webcamoid/blob/62cbc374e1e25ff1c48eaa6d8e0edd63c95de6b7/libAvKys/Plugins/MultiSink/src/ffmpeg/src/mediawriterffmpeg.cpp#L313).
Also, which FFmpeg version are you compiling against?

sameersharma2006, 2024-04-13:
> Try uncommenting [this line](https://github.com/webcamoid/webcamoid/blob/62cbc374e1e25ff1c48eaa6d8e0edd63c95de6b7/libAvKys/Plugins/MultiSink/src/ffmpeg/src/mediawriterffmpeg.cpp#L313). 
Ok, will try that.

> Also, which FFmpeg version are you compiling against?

ffmpeg 6.1.1 , last ffmpeg version supported by release 9.1.1 as it won't build with ffmpeg7

sameersharma2006, 2024-04-13:
Logs (Maybe Relevant );

```
[2024-04-13 12:21:19.996, Webcamoid, 0x7f0cb82ba5c0, VideoCodecOptions.qml (209)] warning: qrc:/Webcamoid/share/qml/VideoCodecOptions.qml:209: Error: Cannot assign [undefined] to QString
[2024-04-13 12:21:19.997, Webcamoid, 0x7f0cb82ba5c0, VideoCodecOptions.qml (192)] warning: qrc:/Webcamoid/share/qml/VideoCodecOptions.qml:192: Error: Cannot assign [undefined] to QString
[matroska,webm @ 0x7f0c840099c0] Duplicate element
[matroska,webm @ 0x7f0c840099c0] 0x00 at pos 110 (0x6e) invalid as first byte of an EBML number
[matroska,webm @ 0x7f0c840099c0] Duplicate element
[matroska,webm @ 0x7f0c840099c0] 0x00 at pos 204 (0xcc) invalid as first byte of an EBML number
[matroska,webm @ 0x7f0c840099c0] Element at 0x67 ending at 0x7a2ec016d exceeds containing master element ending at 0x1413

```

hipersayanX, 2024-04-13:
The log is missing information, there is not an output similar to this before those lines?

```
Output #0, webm, to 'Video.webm':
  Metadata:
    major_brand     : mp42
    minor_version   : 0
    compatible_brands: isomavc1mp42
    encoder         : Lavf60.16.100
  Stream #0:0(und): Video: vp9, yuv420p(progressive), 480x360 [SAR 1:1 DAR 4:3], q=2-31, 24.84 fps, 1k tbn (default)
    Metadata:
      creation_time   : 2009-04-27T00:12:13.000000Z
      handler_name    : (C) 2007 Google Inc. v08.13.2007.
      vendor_id       : [0][0][0][0]
      encoder         : Lavc60.31.102 libvpx-vp9
    Side data:
      cpb: bitrate max/min/avg: 0/0/0 buffer size: 0 vbv_delay: N/A
  Stream #0:1(und): Audio: opus, 48000 Hz, stereo, flt, 96 kb/s (default)
    Metadata:
      creation_time   : 2009-04-27T00:12:12.000000Z
      handler_name    : (C) 2007 Google Inc. v08.13.2007.
      vendor_id       : [0][0][0][0]
      encoder         : Lavc60.31.102 libopus
```

hipersayanX, 2024-04-13:
Which other video codecs are available for Matroska?

sameersharma2006, 2024-04-14:
> The log is missing information, there is not an output similar to this before those lines?
>
> ```
> Output #0, webm, to 'Video.webm':
>   Metadata:
>     major_brand     : mp42
>     minor_version   : 0 [...]

Nope, nothing like that.
Here is the complete terminal output from beginning of the app to recording video.

```
[2024-04-14 09:29:00.693, Webcamoid, 0x7f67d9c325c0,  (0)] critical: Failed to create wl_display (No such file or directory)
[2024-04-14 09:29:00.707, Webcamoid, 0x7f67d9c325c0,  (0)] info: Could not load the Qt platform plugin "wayland" in "" even though it was found.
[2024-04-14 09:29:01.346, Webcamoid, 0x7f67d9c325c0,  (0)] warning: Qt: Session management error: None of the authentication protocols specified are supported
[2024-04-14 09:29:02.400, Webcamoid, 0x7f67d9c325c0,  (0)] debug: Added component "Webcamoid"
[2024-04-14 09:29:02.401, Webcamoid, 0x7f67d9c325c0,  (0)] debug: Added component "VirtualCamera"
[2024-04-14 09:29:04.215, Webcamoid, 0x7f67d9c325c0, VideoCodecOptions.qml (209)] warning: qrc:/Webcamoid/share/qml/VideoCodecOptions.qml:209: Error: Cannot assign [undefined] to QString
[2024-04-14 09:29:04.215, Webcamoid, 0x7f67d9c325c0, VideoCodecOptions.qml (192)] warning: qrc:/Webcamoid/share/qml/VideoCodecOptions.qml:192: Error: Cannot assign [undefined] to QString
[matroska,webm @ 0x7f67cc00fcc0] Duplicate element
[matroska,webm @ 0x7f67cc00fcc0] 0x00 at pos 100 (0x64) invalid as first byte of an EBML number
[matroska,webm @ 0x7f67cc00fcc0] Duplicate element
[matroska,webm @ 0x7f67cc00fcc0] 0x00 at pos 167 (0xa7) invalid as first byte of an EBML number
[matroska,webm @ 0x7f67cc00fcc0] Element at 0x5d ending at 0x77cec0163 exceeds containing master element ending at 0x1409
```

sameersharma2006, 2024-04-14:
> Which other video codecs are available for Matroska?

Hmm, It list many, few notable one's are;

I) In default config, that is at first launch after install no video format & codec appears being specified.

[image]

II) If i change video format to Matroska, the app automatically selects video & audio codec as ;

[image]

III) Available video codecs are ;

[image]

IV) Available audio codecs are ;

[image]

[image]

More audio codecs are available which i can't statically capture but most of the audio codecs seems useless & hoax .

hipersayanX, 2024-04-15:
The webm format is not available? are there other formats?

sameersharma2006, 2024-04-15:
> The webm format is not available? are there other formats beside Matroska?

[image]

[image]

[image]

sameersharma2006, 2024-04-15:
@hipersayanX 

This is what i meant by: In default config although not visible but if we click in the format configure settings it says webm although it has also no effect.

[image]

hipersayanX, 2024-04-15:
Try writing this line [here](https://github.com/webcamoid/webcamoid/blob/62cbc374e1e25ff1c48eaa6d8e0edd63c95de6b7/libAvKys/Plugins/MultiSink/src/multisinkelement.cpp#L114):

```
qDebug() << "FORMAT" << format << this->d->m_formatDescription[format];
```

and give me the output.
Also, just for the purpose of testing, did you tried with the master branch? does the same behavior happens there?

sameersharma2006, 2024-04-16:
> Try writing this line [here](https://github.com/webcamoid/webcamoid/blob/62cbc374e1e25ff1c48eaa6d8e0edd63c95de6b7/libAvKys/Plugins/MultiSink/src/multisinkelement.cpp#L114):
> 
> ```
> qDebug() << "FORMAT" << format << this->d->m_formatDescription[format];
> ```
> and give me the output. 

Ok i will do it.

> Also 1, does libvpx or any of it's variants appears in Matroska video codecs?

Nope

> I think that you are missing VP8/9 codecs in your FFmpeg build, and that's why your are missing the webm format.

Hmm, although am not using ffmpeg build flags for VP8/9, i will try explicitly using VP8/9 build flags for ffmpeg.

sameersharma2006, 2024-04-16:
@hipersayanX  After enabling libvpx support webm appears as the default format . See  👇

[image]

But, still it produces empty and hoax video files with no relevant output.

sameersharma2006, 2024-04-16:
@hipersayanX  After patching the app with your suggestion, logs from startup to video record are ;

```
[2024-04-16 10:02:42.934, Webcamoid, 0x7fa3cfd705c0,  (0)] critical: Failed to create wl_display (No such file or directory)
[2024-04-16 10:02:42.942, Webcamoid, 0x7fa3cfd705c0,  (0)] info: Could not load the Qt platform plugin "wayland" in "" even though it was found.
[2024-04-16 10:02:42.954, Webcamoid, 0x7fa3cfd705c0,  (0)] warning: Qt: Session management error: None of the authentication protocols specified are supported
[2024-04-16 10:02:43.461, Webcamoid, 0x7fa3cfd705c0,  (0)] debug: FORMAT "3g2" "3GP2 (3GPP2 file format)"
[2024-04-16 10:02:43.461, Webcamoid, 0x7fa3cfd705c0,  (0)] debug: FORMAT "3gp" "3GP (3GPP file format)"
[2024-04-16 10:02:43.461, Webcamoid, 0x7fa3cfd705c0,  (0)] debug: FORMAT "a64" "a64 - video for Commodore 64"
[2024-04-16 10:02:43.461, Webcamoid, 0x7fa3cfd705c0,  (0)] debug: FORMAT "ac3" "raw AC-3"
[2024-04-16 10:02:43.461, Webcamoid, 0x7fa3cfd705c0,  (0)] debug: FORMAT "adts" "ADTS AAC (Advanced Audio Coding)"
[2024-04-16 10:02:43.461, Webcamoid, 0x7fa3cfd705c0,  (0)] debug: FORMAT "adx" "CRI ADX"
[2024-04-16 10:02:43.461, Webcamoid, 0x7fa3cfd705c0,  (0)] debug: FORMAT "aiff" "Audio IFF"
[2024-04-16 10:02:43.461, Webcamoid, 0x7fa3cfd705c0,  (0)] debug: FORMAT "alaw" "PCM A-law"
[2024-04-16 10:02:43.461, Webcamoid, 0x7fa3cfd705c0,  (0)] debug: FORMAT "alp" "LEGO Racers ALP"
[2024-04-16 10:02:43.461, Webcamoid, 0x7fa3cfd705c0,  (0)] debug: FORMAT "amv" "AMV"
[2024-04-16 10:02:43.461, Webcamoid, 0x7fa3cfd705c0,  (0)] debug: FORMAT "apm" "Ubisoft Rayman 2 APM"
[2024-04-16 10:02:43.461, Webcamoid, 0x7fa3cfd705c0,  (0)] debug: FORMAT "apng" "Animated Portable Network Graphics"
[2024-04-16 10:02:43.462, Webcamoid, 0x7fa3cfd705c0,  (0)] debug: FORMAT "aptx" "raw aptX (Audio Processing Technology for Bluetooth)"
[2024-04-16 10:02:43.462, Webcamoid, 0x7fa3cfd705c0,  (0)] debug: FORMAT "aptx_hd" "raw aptX HD (Audio Processing Technology for Bluetooth)"
[2024-04-16 10:02:43.462, Webcamoid, 0x7fa3cfd705c0,  (0)] debug: FORMAT "argo_asf" "Argonaut Games ASF"
[2024-04-16 10:02:43.462, Webcamoid, 0x7fa3cfd705c0,  (0)] debug: FORMAT "asf" "ASF (Advanced / Active Streaming Format)"
[2024-04-16 10:02:43.462, Webcamoid, 0x7fa3cfd705c0,  (0)] debug: FORMAT "asf_stream" "ASF (Advanced / Active Streaming Format)"
[2024-04-16 10:02:43.462, Webcamoid, 0x7fa3cfd705c0,  (0)] debug: FORMAT "ass" "SSA (SubStation Alpha) subtitle"
[2024-04-16 10:02:43.462, Webcamoid, 0x7fa3cfd705c0,  (0)] debug: FORMAT "ast" "AST (Audio Stream)"
[2024-04-16 10:02:43.462, Webcamoid, 0x7fa3cfd705c0,  (0)] debug: FORMAT "au" "Sun AU"
[2024-04-16 10:02:43.462, Webcamoid, 0x7fa3cfd705c0,  (0)] debug: FORMAT "avi" "AVI (Audio Video Interleaved)"
[2024-04-16 10:02:43.462, Webcamoid, 0x7fa3cfd705c0,  (0)] debug: FORMAT "avm2" "SWF (ShockWave Flash) (AVM2)"
[2024-04-16 10:02:43.462, Webcamoid, 0x7fa3cfd705c0,  (0)] debug: FORMAT "caf" "Apple CAF (Core Audio Format)"
[2024-04-16 10:02:43.463, Webcamoid, 0x7fa3cfd705c0,  (0)] debug: FORMAT "crc" "CRC testing"
[2024-04-16 10:02:43.463, Webcamoid, 0x7fa3cfd705c0,  (0)] debug: FORMAT "dash" "DASH Muxer"
[2024-04-16 10:02:43.463, Webcamoid, 0x7fa3cfd705c0,  (0)] debug: FORMAT "daud" "D-Cinema audio"
[2024-04-16 10:02:43.463, Webcamoid, 0x7fa3cfd705c0,  (0)] debug: FORMAT "dfpwm" "raw DFPWM1a"
[2024-04-16 10:02:43.463, Webcamoid, 0x7fa3cfd705c0,  (0)] debug: FORMAT "dirac" "raw Dirac"
[2024-04-16 10:02:43.463, Webcamoid, 0x7fa3cfd705c0,  (0)] debug: FORMAT "dnxhd" "raw DNxHD (SMPTE VC-3)"
[2024-04-16 10:02:43.463, Webcamoid, 0x7fa3cfd705c0,  (0)] debug: FORMAT "dv" "DV (Digital Video)"
[2024-04-16 10:02:43.463, Webcamoid, 0x7fa3cfd705c0,  (0)] debug: FORMAT "dvd" "MPEG-2 PS (DVD VOB)"
[2024-04-16 10:02:43.463, Webcamoid, 0x7fa3cfd705c0,  (0)] debug: FORMAT "eac3" "raw E-AC-3"
[2024-04-16 10:02:43.463, Webcamoid, 0x7fa3cfd705c0,  (0)] debug: FORMAT "f32be" "PCM 32-bit floating-point big-endian"
[2024-04-16 10:02:43.463, Webcamoid, 0x7fa3cfd705c0,  (0)] debug: FORMAT "f32le" "PCM 32-bit floating-point little-endian"
[2024-04-16 10:02:43.463, Webcamoid, 0x7fa3cfd705c0,  (0)] debug: FORMAT "f4v" "F4V Adobe Flash Video"
[2024-04-16 10:02:43.463, Webcamoid, 0x7fa3cfd705c0,  (0)] debug: FORMAT "f64be" "PCM 64-bit floating-point big-endian"
[2024-04-16 10:02:43.463, Webcamoid, 0x7fa3cfd705c0,  (0)] debug: FORMAT "f64le" "PCM 64-bit floating-point little-endian"
[2024-04-16 10:02:43.463, Webcamoid, 0x7fa3cfd705c0,  (0)] debug: FORMAT "film_cpk" "Sega FILM / CPK"
[2024-04-16 10:02:43.463, Webcamoid, 0x7fa3cfd705c0,  (0)] debug: FORMAT "fits" "Flexible Image Transport System"
[2024-04-16 10:02:43.463, Webcamoid, 0x7fa3cfd705c0,  (0)] debug: FORMAT "flac" "raw FLAC"
[2024-04-16 10:02:43.464, Webcamoid, 0x7fa3cfd705c0,  (0)] debug: FORMAT "flv" "FLV (Flash Video)"
[2024-04-16 10:02:43.464, Webcamoid, 0x7fa3cfd705c0,  (0)] debug: FORMAT "framecrc" "framecrc testing"
[2024-04-16 10:02:43.464, Webcamoid, 0x7fa3cfd705c0,  (0)] debug: FORMAT "framehash" "Per-frame hash testing"
[2024-04-16 10:02:43.464, Webcamoid, 0x7fa3cfd705c0,  (0)] debug: FORMAT "framemd5" "Per-frame MD5 testing"
[2024-04-16 10:02:43.464, Webcamoid, 0x7fa3cfd705c0,  (0)] debug: FORMAT "g722" "raw G.722"
[2024-04-16 10:02:43.464, Webcamoid, 0x7fa3cfd705c0,  (0)] debug: FORMAT "g723_1" "raw G.723.1"
[2024-04-16 10:02:43.464, Webcamoid, 0x7fa3cfd705c0,  (0)] debug: FORMAT "g726" "raw big-endian G.726 (\"left-justified\")"
[... cut ...]
[2024-04-16 10:02:43.495, Webcamoid, 0x7fa3cfd705c0,  (0)] debug: Added component "VirtualCamera"
[matroska,webm @ 0x7fa39c00a780] Duplicate element
[matroska,webm @ 0x7fa39c00a780] 0x00 at pos 100 (0x64) invalid as first byte of an EBML number
[matroska,webm @ 0x7fa39c00a780] Duplicate element
[matroska,webm @ 0x7fa39c00a780] 0x00 at pos 167 (0xa7) invalid as first byte of an EBML number
[matroska,webm @ 0x7fa39c00a780] Element at 0x5d ending at 0x40ec0100000066 exceeds containing master element ending at 0x1409
```

hipersayanX, 2024-04-17:
The problem is that I'm missing the FFmpeg configuration line, and I can't do blind debugging. Are you sure that did you enabled this line?

```
av_log_set_level(AV_LOG_TRACE);
```

The message should be printed by ```av_dump_format``` [here](https://github.com/webcamoid/webcamoid/blob/62cbc374e1e25ff1c48eaa6d8e0edd63c95de6b7/libAvKys/Plugins/MultiSink/src/ffmpeg/src/mediawriterffmpeg.cpp#L1385).

hipersayanX, 2024-04-17:
Try selecting MP4, MPEG-4 part 2, ACC, and try recording, does it works?

sameersharma2006, 2024-04-18:
@hipersayanX  Nope as i said before any combination of codecs & formats results in null video files.

Same is for your suggested combination.

[image]

P.S.  As before no logs are generated relating to video encoding/decoding.

sameersharma2006, 2024-04-18:
@hipersayanX  Can't  say how much impact this will have, but i have now added support for libx264 & libx265 in the ffmpeg build of webcamoid
 https://gitlab.com/sameersharma2006/webcamoid-snap/-/commit/bbd50cc0cafdc33a5d9efb03d0073c3043274ef3

hipersayanX, 2024-04-18:
If you select MP4 format, it show a different error message or it is still the same?

hipersayanX, 2024-04-18:
Also try replacing this line 

```
// Print recording information.
av_dump_format(this->d->m_formatContext,
               0,
               this->m_location.toStdString().c_str(),
               1);
```

with:

```
qDebug() << "LOG_START";
// Print recording information.
av_dump_format(this->d->m_formatContext,
               0,
               this->m_location.toStdString().c_str(),
               1);
qDebug() << "LOG_END";
```

sameersharma2006, 2024-04-19:
> If you select MP4 format, it show a different error message or it is still the same?

Nope, no logs even with mp4.

sameersharma2006, 2024-04-19:
> Also try replacing this line
>
> ```
> // Print recording information.
> av_dump_format(this->d->m_formatContext,
>                0,
>                this->m_location.toStdString().c_str(),
>                1); [...]

Which file contains this code snippet ?

sameersharma2006, 2024-04-19:
Something of Interest ?

When i replace ffmpeg with gstreamer in general settings, only sound is recorded (Very Poorly) but not video, P.S. v4l wont capture videos and even photos when ffmpeg is replaced by gstreamer.

And as always No Logs...

sameersharma2006, 2024-04-22:
> Also try replacing this line
>
> ```
> // Print recording information.
> av_dump_format(this->d->m_formatContext,
>                0,
>                this->m_location.toStdString().c_str(),
>                1); [...]

Nope, still no ffmpeg related logs are available.
Terminal outputs after the suggested change ;

```
[2024-04-22 12:10:04.713, Webcamoid, 0x7f44357875c0,  (0)] critical: Failed to create wl_display (No such file or directory)
[2024-04-22 12:10:04.726, Webcamoid, 0x7f44357875c0,  (0)] info: Could not load the Qt platform plugin "wayland" in "" even though it was found.
[2024-04-22 12:10:04.766, Webcamoid, 0x7f44357875c0,  (0)] warning: Qt: Session management error: None of the authentication protocols specified are supported
[2024-04-22 12:10:05.765, Webcamoid, 0x7f44357875c0,  (0)] debug: FORMAT "3g2" "3GP2 (3GPP2 file format)"
[2024-04-22 12:10:05.765, Webcamoid, 0x7f44357875c0,  (0)] debug: FORMAT "3gp" "3GP (3GPP file format)"
[2024-04-22 12:10:05.765, Webcamoid, 0x7f44357875c0,  (0)] debug: FORMAT "a64" "a64 - video for Commodore 64"
[2024-04-22 12:10:05.765, Webcamoid, 0x7f44357875c0,  (0)] debug: FORMAT "ac3" "raw AC-3"
[2024-04-22 12:10:05.765, Webcamoid, 0x7f44357875c0,  (0)] debug: FORMAT "adts" "ADTS AAC (Advanced Audio Coding)"
[2024-04-22 12:10:05.766, Webcamoid, 0x7f44357875c0,  (0)] debug: FORMAT "adx" "CRI ADX"
[2024-04-22 12:10:05.766, Webcamoid, 0x7f44357875c0,  (0)] debug: FORMAT "aiff" "Audio IFF"
[2024-04-22 12:10:05.766, Webcamoid, 0x7f44357875c0,  (0)] debug: FORMAT "alaw" "PCM A-law"
[2024-04-22 12:10:05.766, Webcamoid, 0x7f44357875c0,  (0)] debug: FORMAT "alp" "LEGO Racers ALP"
[2024-04-22 12:10:05.766, Webcamoid, 0x7f44357875c0,  (0)] debug: FORMAT "amv" "AMV"
[2024-04-22 12:10:05.766, Webcamoid, 0x7f44357875c0,  (0)] debug: FORMAT "apm" "Ubisoft Rayman 2 APM"
[2024-04-22 12:10:05.766, Webcamoid, 0x7f44357875c0,  (0)] debug: FORMAT "apng" "Animated Portable Network Graphics"
[2024-04-22 12:10:05.766, Webcamoid, 0x7f44357875c0,  (0)] debug: FORMAT "aptx" "raw aptX (Audio Processing Technology for Bluetooth)"
[2024-04-22 12:10:05.766, Webcamoid, 0x7f44357875c0,  (0)] debug: FORMAT "aptx_hd" "raw aptX HD (Audio Processing Technology for Bluetooth)"
[2024-04-22 12:10:05.766, Webcamoid, 0x7f44357875c0,  (0)] debug: FORMAT "argo_asf" "Argonaut Games ASF"
[2024-04-22 12:10:05.766, Webcamoid, 0x7f44357875c0,  (0)] debug: FORMAT "asf" "ASF (Advanced / Active Streaming Format)"
[2024-04-22 12:10:05.766, Webcamoid, 0x7f44357875c0,  (0)] debug: FORMAT "asf_stream" "ASF (Advanced / Active Streaming Format)"
[2024-04-22 12:10:05.766, Webcamoid, 0x7f44357875c0,  (0)] debug: FORMAT "ass" "SSA (SubStation Alpha) subtitle"
[2024-04-22 12:10:05.766, Webcamoid, 0x7f44357875c0,  (0)] debug: FORMAT "ast" "AST (Audio Stream)"
[2024-04-22 12:10:05.766, Webcamoid, 0x7f44357875c0,  (0)] debug: FORMAT "au" "Sun AU"
[2024-04-22 12:10:05.766, Webcamoid, 0x7f44357875c0,  (0)] debug: FORMAT "avi" "AVI (Audio Video Interleaved)"
[2024-04-22 12:10:05.767, Webcamoid, 0x7f44357875c0,  (0)] debug: FORMAT "avm2" "SWF (ShockWave Flash) (AVM2)"
[2024-04-22 12:10:05.767, Webcamoid, 0x7f44357875c0,  (0)] debug: FORMAT "caf" "Apple CAF (Core Audio Format)"
[2024-04-22 12:10:05.767, Webcamoid, 0x7f44357875c0,  (0)] debug: FORMAT "crc" "CRC testing"
[2024-04-22 12:10:05.767, Webcamoid, 0x7f44357875c0,  (0)] debug: FORMAT "dash" "DASH Muxer"
[2024-04-22 12:10:05.767, Webcamoid, 0x7f44357875c0,  (0)] debug: FORMAT "daud" "D-Cinema audio"
[2024-04-22 12:10:05.767, Webcamoid, 0x7f44357875c0,  (0)] debug: FORMAT "dfpwm" "raw DFPWM1a"
[2024-04-22 12:10:05.767, Webcamoid, 0x7f44357875c0,  (0)] debug: FORMAT "dirac" "raw Dirac"
[2024-04-22 12:10:05.767, Webcamoid, 0x7f44357875c0,  (0)] debug: FORMAT "dnxhd" "raw DNxHD (SMPTE VC-3)"
[2024-04-22 12:10:05.767, Webcamoid, 0x7f44357875c0,  (0)] debug: FORMAT "dv" "DV (Digital Video)"
[2024-04-22 12:10:05.767, Webcamoid, 0x7f44357875c0,  (0)] debug: FORMAT "dvd" "MPEG-2 PS (DVD VOB)"
[2024-04-22 12:10:05.767, Webcamoid, 0x7f44357875c0,  (0)] debug: FORMAT "eac3" "raw E-AC-3"
[2024-04-22 12:10:05.767, Webcamoid, 0x7f44357875c0,  (0)] debug: FORMAT "f32be" "PCM 32-bit floating-point big-endian"
[2024-04-22 12:10:05.767, Webcamoid, 0x7f44357875c0,  (0)] debug: FORMAT "f32le" "PCM 32-bit floating-point little-endian"
[2024-04-22 12:10:05.767, Webcamoid, 0x7f44357875c0,  (0)] debug: FORMAT "f4v" "F4V Adobe Flash Video"
[2024-04-22 12:10:05.767, Webcamoid, 0x7f44357875c0,  (0)] debug: FORMAT "f64be" "PCM 64-bit floating-point big-endian"
[2024-04-22 12:10:05.767, Webcamoid, 0x7f44357875c0,  (0)] debug: FORMAT "f64le" "PCM 64-bit floating-point little-endian"
[2024-04-22 12:10:05.767, Webcamoid, 0x7f44357875c0,  (0)] debug: FORMAT "film_cpk" "Sega FILM / CPK"
[2024-04-22 12:10:05.767, Webcamoid, 0x7f44357875c0,  (0)] debug: FORMAT "fits" "Flexible Image Transport System"
[2024-04-22 12:10:05.767, Webcamoid, 0x7f44357875c0,  (0)] debug: FORMAT "flac" "raw FLAC"
[2024-04-22 12:10:05.767, Webcamoid, 0x7f44357875c0,  (0)] debug: FORMAT "flv" "FLV (Flash Video)"
[2024-04-22 12:10:05.767, Webcamoid, 0x7f44357875c0,  (0)] debug: FORMAT "framecrc" "framecrc testing"
[2024-04-22 12:10:05.767, Webcamoid, 0x7f44357875c0,  (0)] debug: FORMAT "framehash" "Per-frame hash testing"
[2024-04-22 12:10:05.767, Webcamoid, 0x7f44357875c0,  (0)] debug: FORMAT "framemd5" "Per-frame MD5 testing"
[2024-04-22 12:10:05.767, Webcamoid, 0x7f44357875c0,  (0)] debug: FORMAT "g722" "raw G.722"
[... cut ...]
[2024-04-22 12:10:35.804, Webcamoid, 0x7f44357875c0,  (0)] debug: LOG_END
[matroska,webm @ 0x7f4404c0d1c0] Duplicate element
[matroska,webm @ 0x7f4404c0d1c0] 0x00 at pos 100 (0x64) invalid as first byte of an EBML number
[matroska,webm @ 0x7f4404c0d1c0] Duplicate element
[matroska,webm @ 0x7f4404c0d1c0] 0x00 at pos 167 (0xa7) invalid as first byte of an EBML number
[matroska,webm @ 0x7f4404c0d1c0] Element at 0x5d ending at 0x40ec0100000066 exceeds containing master element ending at 0x1409
Segmentation fault
```

sameersharma2006, 2024-04-22:
@hipersayanX  Does master branch builds with QT5 ?, as Qt6 SDKs are not currently matured in the snap side.

If yes, i will try building with master branch to see if any changes occur.

hipersayanX, 2024-04-23:
Its weird the av_dump_format is not printing nothing, I did not studied enough the FFmpeg source code, but could it be that av_log_set_level is setting AV_LOG_TRACE as a environment variable and the Snap environment is blocking it :confused: :question: 

>Does master branch builds with QT5 ?, as Qt6 SDKs are not currently matured in the snap side.
>If yes, i will try building with master branch to see if any changes occur.

Webcamoid 9.1 does not build with Qt6, only the master branch does and it is the future 9.2, which right now I'm stepping on the accelerator for releasing it as fast as I can.

Also

>Nope, no logs even with mp4.

This answer is not clear to me yet. When you set MP4, the messages in the console are still referencing Matroska, like this

```
[matroska,webm @ 0x7f43e0015480] Duplicate element
[matroska,webm @ 0x7f43e0015480] 0x00 at pos 100 (0x64) invalid as first byte of an EBML number
[matroska,webm @ 0x7f43e0015480] Duplicate element
[matroska,webm @ 0x7f43e0015480] 0x00 at pos 167 (0xa7) invalid as first byte of an EBML number
[matroska,webm @ 0x7f43e0015480] Element at 0x5d ending at 0x40ec0100000066 exceeds containing master element ending at 0x1409
```

or shows a different message?

hipersayanX, 2024-04-23:
Also, can you try packaging this simple program and see if this works?

```
#include <stdio.h>
#include <stdlib.h>

int main(void)
{
    char *msg = NULL;
    setenv("msg", "Hello world!", 1);
    msg = getenv("msg");

    if (msg)
        printf("%s\n", msg);

    return 0;
}
```

sameersharma2006, 2024-04-23:
> Also
>
>
> This answer is not clear to me yet. When you set MP4, the messages in the console are still referencing Matroska, like this
>
> ```
> [matroska,webm @ 0x7f43e0015480] Duplicate element [...]

This is the setting for MP4 ;

[image]

The above lines appear before changing to mp4, after changing to mp4 no logs appear even the above highlighted one.
The above appears with mkv, webm not in mp4.

MP4 doesn't have any output specific to it.

sameersharma2006, 2024-04-23:
> Also, can you try packaging this simple program and see if this works?
> 
> ```
> #include <stdio.h>
> #include <stdlib.h>
> 
> int main(void)
> {
>     char *msg = NULL;
>     setenv("msg", "Hello world!", 1);
>     msg = getenv("msg");
> 
>     if (msg)
>         printf("%s\n", msg);
> 
>     return 0;
> }
> ```

Ok.

sameersharma2006, 2024-04-23:
@hipersayanX  I packaged the simple program as snap, it gives an output of `Hello World !` .

Saved it as `exp.cpp` file, did g++ exp.cpp , dumped the output -> `a.out` into snap and executed snap, give output-> `Hello World !`

hipersayanX, 2024-04-24:
Maybe the messages are printed in the stderr instead of stdout (or vice-versa)?

 ```
#include <stdio.h>
#include <stdlib.h>

int main(void)
{
    fprintf(stdout, "%s", "Hello from stdout\n"); 
    fprintf(stderr, "%s", "Hello from stderr\n");

    return 0;
}
```

sameersharma2006, 2024-04-25:
@hipersayanX  I did modify the code as per your suggestion.

Now the output is 
```
Hello from stdout
Hello from stderr
```
I am attaching a tar ball containing the code, snap yaml & the snap itself

[exp.tar.gz](https://github.com/webcamoid/webcamoid/files/15106179/exp.tar.gz)

hipersayanX, 2024-04-27:
I you write this right after  ```av_log_set_level```, does it works?

```
av_log(NULL, AV_LOG_INFO, "%s\n", "Hello world");
```

sameersharma2006, 2024-04-28:
> I you write this right after `av_log_set_level`, does it works?
> 
> ```
> av_log(NULL, AV_LOG_INFO, "%s\n", "Hello world");
> ```

Ok i will add a patch for this.

sameersharma2006, 2024-04-30:
@hipersayanX  Added patch viz. https://gitlab.com/sameersharma2006/webcamoid-snap/-/commit/6c5f3639172e82a7e285581529f2672adf6c5c44

sameersharma2006, 2024-04-30:
@hipersayanX  After adding that line, still nothing appears not even the hello world.

1) With webm
```
[2024-04-30 16:46:09.814, Webcamoid, 0x7f2f038b05c0,  (0)] critical: Failed to create wl_display (No such file or directory)
[2024-04-30 16:46:09.838, Webcamoid, 0x7f2f038b05c0,  (0)] info: Could not load the Qt platform plugin "wayland" in "" even though it was found.
[2024-04-30 16:46:09.857, Webcamoid, 0x7f2f038b05c0,  (0)] warning: Qt: Session management error: None of the authentication protocols specified are supported
[2024-04-30 16:46:10.330, Webcamoid, 0x7f2f038b05c0,  (0)] debug: FORMAT "3g2" "3GP2 (3GPP2 file format)"
[2024-04-30 16:46:10.330, Webcamoid, 0x7f2f038b05c0,  (0)] debug: FORMAT "3gp" "3GP (3GPP file format)"
[2024-04-30 16:46:10.330, Webcamoid, 0x7f2f038b05c0,  (0)] debug: FORMAT "a64" "a64 - video for Commodore 64"
[2024-04-30 16:46:10.331, Webcamoid, 0x7f2f038b05c0,  (0)] debug: FORMAT "ac3" "raw AC-3"
[2024-04-30 16:46:10.331, Webcamoid, 0x7f2f038b05c0,  (0)] debug: FORMAT "adts" "ADTS AAC (Advanced Audio Coding)"
[2024-04-30 16:46:10.331, Webcamoid, 0x7f2f038b05c0,  (0)] debug: FORMAT "adx" "CRI ADX"
[2024-04-30 16:46:10.331, Webcamoid, 0x7f2f038b05c0,  (0)] debug: FORMAT "aiff" "Audio IFF"
[2024-04-30 16:46:10.331, Webcamoid, 0x7f2f038b05c0,  (0)] debug: FORMAT "alaw" "PCM A-law"
[2024-04-30 16:46:10.331, Webcamoid, 0x7f2f038b05c0,  (0)] debug: FORMAT "alp" "LEGO Racers ALP"
[2024-04-30 16:46:10.331, Webcamoid, 0x7f2f038b05c0,  (0)] debug: FORMAT "amv" "AMV"
[2024-04-30 16:46:10.331, Webcamoid, 0x7f2f038b05c0,  (0)] debug: FORMAT "apm" "Ubisoft Rayman 2 APM"
[2024-04-30 16:46:10.331, Webcamoid, 0x7f2f038b05c0,  (0)] debug: FORMAT "apng" "Animated Portable Network Graphics"
[2024-04-30 16:46:10.331, Webcamoid, 0x7f2f038b05c0,  (0)] debug: FORMAT "aptx" "raw aptX (Audio Processing Technology for Bluetooth)"
[2024-04-30 16:46:10.331, Webcamoid, 0x7f2f038b05c0,  (0)] debug: FORMAT "aptx_hd" "raw aptX HD (Audio Processing Technology for Bluetooth)"
[2024-04-30 16:46:10.331, Webcamoid, 0x7f2f038b05c0,  (0)] debug: FORMAT "argo_asf" "Argonaut Games ASF"
[2024-04-30 16:46:10.331, Webcamoid, 0x7f2f038b05c0,  (0)] debug: FORMAT "asf" "ASF (Advanced / Active Streaming Format)"
[2024-04-30 16:46:10.332, Webcamoid, 0x7f2f038b05c0,  (0)] debug: FORMAT "asf_stream" "ASF (Advanced / Active Streaming Format)"
[2024-04-30 16:46:10.332, Webcamoid, 0x7f2f038b05c0,  (0)] debug: FORMAT "ass" "SSA (SubStation Alpha) subtitle"
[2024-04-30 16:46:10.332, Webcamoid, 0x7f2f038b05c0,  (0)] debug: FORMAT "ast" "AST (Audio Stream)"
[2024-04-30 16:46:10.332, Webcamoid, 0x7f2f038b05c0,  (0)] debug: FORMAT "au" "Sun AU"
[2024-04-30 16:46:10.332, Webcamoid, 0x7f2f038b05c0,  (0)] debug: FORMAT "avi" "AVI (Audio Video Interleaved)"
[2024-04-30 16:46:10.332, Webcamoid, 0x7f2f038b05c0,  (0)] debug: FORMAT "avm2" "SWF (ShockWave Flash) (AVM2)"
[2024-04-30 16:46:10.332, Webcamoid, 0x7f2f038b05c0,  (0)] debug: FORMAT "caf" "Apple CAF (Core Audio Format)"
[2024-04-30 16:46:10.332, Webcamoid, 0x7f2f038b05c0,  (0)] debug: FORMAT "crc" "CRC testing"
[2024-04-30 16:46:10.332, Webcamoid, 0x7f2f038b05c0,  (0)] debug: FORMAT "dash" "DASH Muxer"
[2024-04-30 16:46:10.332, Webcamoid, 0x7f2f038b05c0,  (0)] debug: FORMAT "daud" "D-Cinema audio"
[2024-04-30 16:46:10.332, Webcamoid, 0x7f2f038b05c0,  (0)] debug: FORMAT "dfpwm" "raw DFPWM1a"
[2024-04-30 16:46:10.332, Webcamoid, 0x7f2f038b05c0,  (0)] debug: FORMAT "dirac" "raw Dirac"
[2024-04-30 16:46:10.332, Webcamoid, 0x7f2f038b05c0,  (0)] debug: FORMAT "dnxhd" "raw DNxHD (SMPTE VC-3)"
[2024-04-30 16:46:10.332, Webcamoid, 0x7f2f038b05c0,  (0)] debug: FORMAT "dv" "DV (Digital Video)"
[2024-04-30 16:46:10.332, Webcamoid, 0x7f2f038b05c0,  (0)] debug: FORMAT "dvd" "MPEG-2 PS (DVD VOB)"
[2024-04-30 16:46:10.332, Webcamoid, 0x7f2f038b05c0,  (0)] debug: FORMAT "eac3" "raw E-AC-3"
[2024-04-30 16:46:10.332, Webcamoid, 0x7f2f038b05c0,  (0)] debug: FORMAT "f32be" "PCM 32-bit floating-point big-endian"
[2024-04-30 16:46:10.333, Webcamoid, 0x7f2f038b05c0,  (0)] debug: FORMAT "f32le" "PCM 32-bit floating-point little-endian"
[2024-04-30 16:46:10.333, Webcamoid, 0x7f2f038b05c0,  (0)] debug: FORMAT "f4v" "F4V Adobe Flash Video"
[2024-04-30 16:46:10.333, Webcamoid, 0x7f2f038b05c0,  (0)] debug: FORMAT "f64be" "PCM 64-bit floating-point big-endian"
[2024-04-30 16:46:10.333, Webcamoid, 0x7f2f038b05c0,  (0)] debug: FORMAT "f64le" "PCM 64-bit floating-point little-endian"
[2024-04-30 16:46:10.333, Webcamoid, 0x7f2f038b05c0,  (0)] debug: FORMAT "film_cpk" "Sega FILM / CPK"
[2024-04-30 16:46:10.333, Webcamoid, 0x7f2f038b05c0,  (0)] debug: FORMAT "fits" "Flexible Image Transport System"
[2024-04-30 16:46:10.333, Webcamoid, 0x7f2f038b05c0,  (0)] debug: FORMAT "flac" "raw FLAC"
[2024-04-30 16:46:10.333, Webcamoid, 0x7f2f038b05c0,  (0)] debug: FORMAT "flv" "FLV (Flash Video)"
[2024-04-30 16:46:10.333, Webcamoid, 0x7f2f038b05c0,  (0)] debug: FORMAT "framecrc" "framecrc testing"
[2024-04-30 16:46:10.333, Webcamoid, 0x7f2f038b05c0,  (0)] debug: FORMAT "framehash" "Per-frame hash testing"
[2024-04-30 16:46:10.333, Webcamoid, 0x7f2f038b05c0,  (0)] debug: FORMAT "framemd5" "Per-frame MD5 testing"
[2024-04-30 16:46:10.333, Webcamoid, 0x7f2f038b05c0,  (0)] debug: FORMAT "g722" "raw G.722"
[2024-04-30 16:46:10.333, Webcamoid, 0x7f2f038b05c0,  (0)] debug: FORMAT "g723_1" "raw G.723.1"
[2024-04-30 16:46:10.333, Webcamoid, 0x7f2f038b05c0,  (0)] debug: FORMAT "g726" "raw big-endian G.726 (\"left-justified\")"
[... cut ...]
[2024-04-30 16:47:28.993, Webcamoid, 0x7f3e4d78b5c0,  (0)] debug: FORMAT "wav" "WAV / WAVE (Waveform Audio)"
[2024-04-30 16:47:28.993, Webcamoid, 0x7f3e4d78b5c0,  (0)] debug: FORMAT "webm" "WebM"
[2024-04-30 16:47:28.993, Webcamoid, 0x7f3e4d78b5c0,  (0)] debug: FORMAT "webvtt" "WebVTT subtitle"
[2024-04-30 16:47:28.993, Webcamoid, 0x7f3e4d78b5c0,  (0)] debug: FORMAT "wsaud" "Westwood Studios audio"
[2024-04-30 16:47:28.993, Webcamoid, 0x7f3e4d78b5c0,  (0)] debug: FORMAT "wtv" "Windows Television (WTV)"
[2024-04-30 16:47:28.993, Webcamoid, 0x7f3e4d78b5c0,  (0)] debug: FORMAT "wv" "raw WavPack"
```

hipersayanX, 2024-05-01:
Try removing  ```--disable-debug``` in ffmpeg autotools-configure-parameters.

sameersharma2006, 2024-05-02:
@hipersayanX  Nope even ater removing that ffmpeg flag, the output is similar to what i posted previously.

No logs, the hello world snippet also doesn't seem to work.

hipersayanX, 2024-05-02:
Tbh I'm out of ideas, it seems to be a bug on your FFmpeg build but I don't know what could it be :confused:

sameersharma2006, 2024-05-03:
@hipersayanX  Hmm,my ffmpeg build is from source and i am using that same build instruction in celluloid,mpv,audacious snaps all works fine.

hipersayanX, 2024-05-03:
In the last case, the problem could be the version of FFmpeg that you are building against. You are building against FFmpeg 7.0 and Webcamoid 9.1.1 was released 1 year ago, and FFmpeg constantly make many changes to their API.
The release versions are working fine and it was build in Ubuntu Focal, that is [FFmpeg 4.2.7](https://packages.ubuntu.com/focal/ffmpeg). Maybe try downgrading to FFmpeg 4.2.7?

sameersharma2006, 2024-05-04:
@hipersayanX  Will that have any security trade-offs, i mean security vulnerabilities (CVEs)  ? Nevertheless i think  your prediction is true given the fact that webcamoid snap actually used to worked last year with older ffmpeg.

hipersayanX, 2024-05-04:
>Will that have any security trade-offs, i mean security vulnerabilities (CVEs) ?

If that the case, apply the required patches.

>The focal thing can not apply here as the snap is being built atop core22(Jammy), core20 kde/qt support ended long before.

That is not a problem, the build environment is irrelevant here. Try building with the known working version, ultimately try upgrading to a newer one. If I remember well on that time Arch had ffmpeg 5.x and it was working fine there.

Also, if possible, I will try to make a new release for June, but not sure because there are some bugs I still need to fix.

sameersharma2006, 2024-05-05:
@hipersayanX  Unfortunately it still doesn't work.

I tested with 4.2.9 (Bugfix release with similar code to 4.2.7) & ffmpeg-5.1.4 

Both produces hoax files .

I think the best bet is waiting till new release & then re-visit these situation.

I am not sure why is this happening AFAIK last year webcamoid snap did record video with only mkv format, but still it atleast recorded video.

* Just a suggestion & feature-request:

1) Consider providing a cmake build flag to disable video recording feature altogether & trim the app to photo capturing utility with effects.

hipersayanX, 2024-05-05:
>I think the best bet is waiting till new release & then re-visit these situation.

There is not any guarantee that this gonna be fixed for the next release because:

1. Is not reproducible (it works for me).
2. Is not debugable (the logs are missing).

Instead of waiting for the next release, try building from master branch and check if the situation still persists.

sameersharma2006, 2024-05-06:
@hipersayanX  Ok, started building webcamoid master branch with qt6 libs viz. https://git.launchpad.net/webcamoid-snap?h=canary

hipersayanX, 2024-05-07:
>You mean to say the snap works for you ? or you are talking about native packaging formats / root build & install.

I'm talking about the binaries I distribute and the source code. I did tried adding a [Snap build](https://github.com/webcamoid/webcamoid/tree/efd54c1c34a5f9b4dafdec2e5ead094124db1658/ports/ci/snap) to the actions, but the build always failed, so I have removed it.

sameersharma2006, 2024-05-08:
@hipersayanX  If you are building snaps without helpers or extensions, you also probably need to add a build-environment part for building your app along with command scripts, hooks etc.

Nevertheless i will continue debugging against master branch.

hipersayanX, 2024-05-08:
>@hipersayanX If you are building snaps without helpers or extensions, you also probably need to add a build-environment part for building your app along with command scripts, hooks etc.

I don't know how Snap packages work, I just took code from here and there. Also, isn't possible to use ffmpeg from Ubuntu repositories instead of building it from scratch? or in general, any pre-build binary package from repositories ?

sameersharma2006, 2024-05-09:
One can use ffmpeg from ubuntu repos but it's a bit chaotic as it will include managing dependencies & will also increase app size leading to slower startups.

hipersayanX, 2024-10-01:
Fixed in 9.2.0.

closed 2024-10-01 by hipersayanX