webcamoid/webcamoid #693: Video Source Options not saving (VIdeo Format and Resolution)

lumifoxbox, 2024-04-14:
Is there any way to make the video format settings save like in the registery or in the program?, everytime i open the app it defaults to YUYV and 640p and i have to change it.

lumifoxbox, 2024-04-14:
I have the windows camera app unistalled too, does that make a difference?

hipersayanX, 2024-04-15:
As far as I remember, the camera resolution should be saved and restored each time, the recording takes the resolution from the actual source as explained in #678.

>I have the windows camera app unistalled too, does that make a difference?

No, Webcamoid has not any relation with other apps.

lumifoxbox, 2024-04-15:
> As far as I remember, the camera resolution should be saved and restored each time, the recording takes the resolution from the actual source as explained in #678.
> 
> > I have the windows camera app unistalled too, does that make a difference?
> 
> No, Webcamoid has not any relation with other apps.

I noticed another post saying it wasnt working with their razer kiyo pro (which is what i was using) so i tried my logitech C290 which also defaults to the same YUYV 640p settings. I reinstalled windows about a week ago, is it possible i'm missing some kind of codec? (i tried installing ffmpeg yesterday and it diddnt help)? I bought the paid version from your website, would that be any different from the github version?

lumifoxbox, 2024-04-15:
ffmpeg -f dshow -list_options true -i video="Razer Kiyo Pro Ultra" and
ffmpeg -f dshow -list_options true -i video="HD Pro Webcam C920"  seem to both be showing that they're getting recognised as being able to do the full range of resolutions and formats

hipersayanX, 2024-04-15:
>I noticed another post saying it wasnt working with their razer kiyo pro (which is what i was using) so i tried my logitech C290 which also defaults to the same YUYV 640p settings.

So the problem is that when you open Wecamoid, the format is always YUYV 640x480, right? will check, but other resolutions works fine, right?

>I reinstalled windows about a week ago, is it possible i'm missing some kind of codec?

All codecs are included in the binary release, the codecs can't be installed or uninstalled, all codecs are built-in.

>(i tried installing ffmpeg yesterday and it diddnt help)?

Again, Webcamoid does not interact with any other software in your system, reinstalling the system or other apps is irrelevant.

>I bought the paid version from your website, would that be any different from the github version?

All the binaries are automatically generated by github actions in [the corresponding tag](https://github.com/webcamoid/webcamoid/actions/runs/5274146044).

lumifoxbox, 2024-04-15:
Seems to be havin the same issue on my laptop too with the intergrated camera

> So the problem is that when you open Wecamoid, the format is always YUYV 640x480, right? will check, but other resolutions works fine, right?

Yeah it works fine after changing, i guess i'm just being picky cause i'm dumb and bought the $500 razer webcam literally just to take product shots for my ebay store using a light box next to me, but for some reason the razer software has no way to take photos or videos (like logitechs does) and the windows camera removes all of razers settings, Your software seems to work fine but i'm a bit spontanious with when i do things and going through settings every time just makes me lazy.

Ironically before finding your program i was trying to figure out how to code something myself using ffmpeg commands to just take a snapshot with a batch file but couldnt get it to work without being a super low resolution

lumifoxbox, 2024-04-18:
> Again, Webcamoid does not interact with any other software in your system, reinstalling the system or other apps is irrelevant.

Is this something i could manually force via the registery, i cant seem to find the value for it, but i know the settings are stored there?

Do you think the portable version would make any difference?

hipersayanX, 2024-04-18:
I was checking, and the code for saving the stream (the format you have selected for the camera) [is there](https://github.com/webcamoid/webcamoid/blob/7eec80270286262290b59ed8dd8e021fdf0ef6b9/libAvKys/Plugins/VideoCapture/src/videocaptureelement.cpp#L134), in Windows it's located in the regedit at ```HKEY_CURRENT_USER\Software\Webcamoid\Webcamoid\VideoCapture\devices\<number of the device>\stream=X```, try changing the format of the camera, close Webcamoid, and check if the value of the stream is changing, alternatively, change the stream manually and check if Webcamoid can catch the format.
Just to be clear, each device is composed by 3 properties:

**description**: This is the camera human readable name that is showed when you open the video inputs tab.  
**id**: This is the ID assigned by the operating system, I'm not sure if Windows assign a different ID with each reboot (which could be causing the bug that you are describing).
**stream**: Webcamoid stores all video formats (Width x Height FPS) into an array and the **stream* is an index of that array, so when testing try not using big numbers, something between 0-9 is ok, again just for the purpose of doing manual testing.

lumifoxbox, 2024-04-20:
> I was checking, and the code for saving the stream (the format you have selected for the camera) [is [...]

I think i figured out!, Webcamoid itself doesnt seem to be checking the register values when it starts, I was using Procmon to figure out exactly what number the process sets to and it seems to be saving the values in register regardless of what i set them to, but the program doesnt actually read this value when it launches (in my case it would be stream 9), the program also isnt forcing the registery stream value to change back to 0 either when it starts, so i guess the value isnt being addressed and its defaulting to 0 until changed again (i can see it change to 4 (MJPG 650x4080) then 9 (MJPG 3830x2160 when i change after launching but its still on 9 on startup)

hipersayanX, 2024-04-23:
> the program also isnt forcing the registery stream value to change back to 0 either when it starts, so i guess the value isnt being addressed and its defaulting to 0 until changed again (i can see [...]

Perfect, will check then!
Does it happens only with the streams or in general? for example, selecting a different video source or recording video format, adding effects, then returning back to it's default (camera controls does not count, because Webcamoid does not save them yet).
Also, sorry for the big delay of my responses but I'm very busy with the Android port right now.

lumifoxbox, 2024-04-23:
> Perfect, will check then! Does it happens only with the streams or in general? for example, selecting a different video source or recording video format, adding effects, then returning back to it's default (camera controls does not count, because Webcamoid does not save them yet). 

Tested it now, Seems to happen with just streams, it keeps whatever video source and effect and edits to the effect I last used (for example i changed the effect "ascii art" and the symbols used to "TEST" and it saves it, and i can see it update in the registery)  , as well as other preferences like file format and image directory and turning things like flash off. 

> Also, sorry for the big delay of my responses but I'm very busy with the Android port right now.

Its fine you're already 100 times more helpful than razer support just by actually responding heh, i was super close to buying the portable version yesterday but i figured it might also use the registry to save info anyways and i should be patient and wait for a response

hipersayanX, 2024-04-23:
When you close and open Webcamoid, the camera device ID (the string bellow the description) changes? (I imagine the answer is no, but just in case).

hipersayanX, 2024-04-23:
>i was super close to buying the portable version yesterday but i figured it might also use the registry to save info anyways and i should be patient and wait for a response

Both the installer and the portable version, contains the exact same binaries, the only difference is on how the user want to use it.

lumifoxbox, 2024-04-23:
> When you close and open Webcamoid, the camera device ID (the string bellow the description) changes? (I imagine the answer is no, but just in case).

Its a number like ||?|usb#vid_1532 and yeah it seems to be staying the same, the extra long id in regedit isnt changing either

hipersayanX, 2024-04-24:
>Its a number like ||?|usb#vid_1532 and yeah it seems to be staying the same, the extra long id in regedit isnt changing either

Yes, that is. Will check this is definitely a bug.

hipersayanX, 2024-10-01:
Fixed in 9.2.0.

closed 2024-10-01 by hipersayanX