webcamoid/webcamoid #694: Video recording results in almost empty files

fhenryco, 2024-04-23:
I have a MJPG webcam i'm trying to record. I see that there is another issue that looks like mine. But it is ultra technical and i understand nothing. I Launched the app image on linux in terminal and get message: 
main libvlc error: stale plugins cache: modified /tmp/.mount_webcamv93IPi/lib/vlc/plugins/video_output/libyuv_plugin.so

then when i launch the recording
 i see many messages  : Invalid packet stream index: 1

hipersayanX, 2024-04-24:
>I Launched the app image on linux in terminal and get message:
main libvlc error: stale plugins cache: modified /tmp/.mount_webcamv93IPi/lib/vlc/plugins/video_output/libyuv_plugin.so

This message can be ignored.

>then when i launch the recording
>i see many messages : Invalid packet stream index: 1

Try selecting in the video recording settings: MP4, libx264, AAC.

bnisevic, 2024-06-04:
I have the same issue. Can not find video config which works. Does anybody have working conf for video recording on Ubuntu 22.04?

hipersayanX, 2024-10-01:
Fixed in 9.2.0

closed 2024-10-01 by hipersayanX