webcamoid/webcamoid #695: NOT FREE

GlydetheSurfer, 2024-04-24:
If you guys say it's free.. why is it requiring me to pay $1.50 for it? What an absolute lie.

[image]

hipersayanX, 2024-04-24:
Webcamoid is Free as  in Free Speech, not as in free beer.

https://www.gnu.org/philosophy/free-sw.html.en

closed 2024-04-24 by GlydetheSurfer

hipersayanX, 2024-04-24:
>they said you just started charging money for it... so it WAS once "actually" free.

It was, yes. But I had to charge for it because I need the income. And anyways, actually any other FOSS project can do the same I'm doing, for example [Ardour is doing the same I'm doing](https://community.ardour.org/download).

>and idk if you are thinking about deleting this post.. at LEAST I got my message out!

I don't, as long as keep respectful.