webcamoid/webcamoid #696: Blank image (in fact white) from camera

callegar, 2024-05-08:
## Summary

Starting the application, I do not get any image from the camera, just a blank (in fact white) screen.

## Current Behavior

Encountering the issue with a locally compiled arch package (built using the AUR PKGBUILD and the 9.1.1 sources).

Situation is strange because:

1. the same package works fine on two other computers of mine, I have only a single laptop where I am encountering this behavior.
2. the flatpak from flathub (currently at 9.0 and indicated as deprecated since it is built against obsolete flatpak frameworks) works fine on that machine.
3. in a few occasions I have been able to play with the camera controls (in some other cases the app froze), switching the camera away from  MJPG to YUYV gave me an image (though unusable due to the too slow refresh).

## Expected Behavior

Image from the camera expected.

## Steps to Reproduce

Start webcamoid

## Suggestions and tips

No clue

## Your Environment

* Webcamoid information: 9.1.1 on x86-64
* Operating System information: Manjaro linux (arch derivative)

callegar, 2024-05-08:
Just noticed that also on the hosts where webcamoid works, the screen freezes as soon as one selects "screen 0" as the source. If you try to switch back to another source, the screen remain frozen. The rest of the application works, though: for instance I have a camera that "clicks" when the configuration is changed.

Don't know if the things can be related.

hipersayanX, 2024-05-08:
In Preferences/General options/Video capture, try switching the values either V4L2, V4L2 wrapper, or Qt. Does the same happens with other programs?

nucino, 2025-08-15:
Hi, same white screen, with old version 8.8.0 everything worked ok but with paid "upgrade" to 9.3.0 I get the following errors when running from terminal:

"Cannot load library /home/felipe/.Webcamoid/lib/qt/plugins/avkys/libVideoCapture_ffmpeg.so: (/home/felipe/.Webcamoid/lib/qt/plugins/avkys/../../../libpango-1.0.so.0: undefined symbol: hb_ot_layout_get_horizontal_baseline_tag_for_script)"

..
..
..

info: V4L2 video device node found: "/dev/video0"
[2025-08-15 17:20:31.454, Webcamoid, 0x74a8b9ba5f00,  (0)] info: Detected camera: "Integrated_Webcam_HD: Integrate" ( "/dev/video0" )
...
..
info: V4L2 video device node found: "/dev/video1"
[2025-08-15 17:20:31.454, Webcamoid, 0x74a8b9ba5f00,  (0)] info: Detected camera: "Integrated_Webcam_HD: Integrate" ( "/dev/video1" )

Regards,

hipersayanX, 2025-08-15:
And I'm not sure at all that the command above will work at all, because pango library is already loading harfbuzz and hb_ot_layout_get_horizontal_baseline_tag_for_script function should be defined there, and it is not because is and older version than required. You should try upgrading to Ubuntu 24.04 LTS at least.
Also, in the download options its already explained the minimum requirements:

[image]

I use gcc and glibc as reference since it are the only common points across every single distro.

nucino, 2025-08-15:
> I have updated the description for the minimum required version for the
> main distros.

Hi,

I am not updating my OS soon, will try to update GCC however,  anyway is it
available the appimage version of newest webcamoid?

Regards,

Felipe Rodriguez

hipersayanX, 2025-08-16:
>I am not updating my OS soon, will try to update GCC however,

Is not about upgrading GCC, GCC is just a reference of how old the system must be. Is must better to just update to Ubuntu 24.04 that is the last LTS.

>anyway is it available the appimage version of newest webcamoid?

Webcamoid 9.3.0 is the latest version of Webcamoid, all the packages has the same requirements.

The biggest problem with supporting Ubuntu 22.04 is that [it provides up to Qt 6.2](https://pkgs.org/search/?q=libqt6core6) which is a very old version of Qt, the API difference with Qt 6.9 is too big.