webcamoid/webcamoid #697: How to download?

murolem, 2024-05-11:
How to download the windows binary? 
There's no way to download anything in the store without buying https://payhip.com/Webcamoid
Is this software free?

hipersayanX, 2024-05-11:
>How to download the windows binary?
>There's no way to download anything in the store without buying https://payhip.com/Webcamoid

You can buy it with Paypal or credit card. Also you can download the [source code](https://github.com/webcamoid/webcamoid) and [compile it yourself](https://github.com/webcamoid/webcamoid/wiki/Raw-build-and-install) for free.

>Is this software free?

Webcamoid is [Free as in Free Speech](https://www.gnu.org/philosophy/free-sw.en.html).

Merquis00, 2024-07-18:
>Is this software free?

Short answer: no, you have to pay for it.
less-short answer: you can fork the code and build it yourself.

avachon100510, 2024-08-05:
This is completely absurd. This is open-source and should therefore be free! But paying for pre-compiled binaries? This is dumb.

avachon100510, 2024-08-05:
Furthermore, a guest considered this method of obtaining the binaries through payment a violation of the GPL license. However... Clause 6 states that: if you distribute binaries, you must provide access to the source code with no additional charge. The source code is accessible, so nothing wrong so far. Therefore, it ain't a violation of this clause... which is... surprising. But locking the binaries behind a paywall is completely dumb, defeating the purpose of being a "free" software.

One user stated in AlternativesTo website:

> Surveillance program.
>
>
> Webcamoid and Ryonakano - GitHub sourced - records everything in filing system. Vids, docs, photos, everything. Find them in > RUN USR directories. When issues raised, Dev's hostile, delete comments queries, and blocked. Uninstalled still running.

By Shane-Hanson. 

This could raise some red flags, and this has to be addressed at once.

hipersayanX, 2024-08-06:
@avachon100510

>This is open-source and should therefore be free!

[image]
[image]

You are confusing ```open source``` with ```free software``` and ```Free Software```. 
```open source``` means that the original source code of a compiled program is accessible and readable by a human. A program can be ```open source``` but it's license can be still privative not respecting the [4 freedoms](https://www.gnu.org/philosophy/free-sw.en.html). For example, [The price of Persia](https://github.com/jmechner/Prince-of-Persia-Apple-II).
```free software``` or basically freeware or shareware, is just a software that owner does not change for it's usage, but it can still be closed source and privative software. For example, WhatsApp. 
```Free Software``` is the software in which the license respects the 4 freedoms:

0. The freedom to run the program as you wish, for any purpose (freedom 0).
1. The freedom to study how the program works, and change it so it does your computing as you wish (freedom 1). Access to the source code is a precondition for this.
2. The freedom to redistribute copies so you can help others (freedom 2).
3. The freedom to distribute copies of your modified versions to others (freedom 3). By doing this you can give the whole community a chance to benefit from your changes. Access to the source code is a precondition for this.

Webcamoid is ```Free Software```.

>But paying for pre-compiled binaries? This is dumb.

Some example of paid ```Free Software``` :

- [Krita](https://store.steampowered.com/app/280680/Krita/)
- [LibreOffice](https://apps.microsoft.com/detail/9pb80dcfp83w?hl=en-US&gl=US)
- [Ardour](https://community.ardour.org/download)
- [Red Hat Enterprise Linux](https://developers.redhat.com/products/rhel/download)
- [SUSE Linux Enterprise](https://www.suse.com/products/desktop/)

Your are paying for the ready-made binaries, support, and probably new features not available in the release versions, in my case when a user report a bug I send the daily build with the patches and the features not available in the release. Those who pay are actually the user who are helping to fix the bugs in Webcamoid, meanwhile the users who use it for free are totally silent (except for those who made very technical reports, they are f*cking gods :ok_hand:).

>Furthermore, a guest considered this method of obtaining the binaries through payment a violation of the GPL license. 

[Here is the license](https://github.com/webcamoid/webcamoid/blob/master/COPYING), can you tell me which clause I'm infringing? you can also ask a lawyer, the [Free Software Foundation](https://www.fsf.org), even better you can ask Richard Stallman (rms at gnu dot org), he most probably will complain about Webcamoid providing non-free codecs, or Payhip not behind Free Software, or github owned by Microsoft, or using Paypal as a paymet method, but he will never complain about Webcamoid behind paid.

>However... Clause 6 states that: if you distribute binaries, you must provide access to the source code with no additional charge.

And here is [the full source code](https://github.com/webcamoid/webcamoid/).

>The source code is accessible, so nothing wrong so far. Therefore, it ain't a violation of this clause... which is... surprising.

Exactly.

> One user stated in AlternativesTo website:
>
>
> By Shane-Hanson.
>
> This could raise some red flags, and this has to be addressed at once.

[image]
[image]

Every build is has a build-info.txt file with the following information:

```
Commit hash of the git repository
Build log URL to the github action where it was build

Information of the operating system where it was build

A list of the packages providing the binary files distributed with Webcamoid
```

From there you can do an exact trace to the source code fr where the binaries were build. Now, if some random twat is claiming that Webcamoid is some sort of spyware, there is the code he can use to probe it, but I can safely bet he can't connect 2 neurons, and in the last case he can request a malware analysis to some anti-malware company. If you don't trust the source code you can make a full analysis of the binaries, and you won't never find any difference between the binaries build from the source code (with the same build environment) and the binaries distributed by me. 
The funny thing is that I can sue the guy and win because both the proofs and the truth is in my side because he know he is lying and Webcamoid is Free Software.

closed 2024-10-01 by hipersayanX