webcamoid/webcamoid #698: Webcamoid, 0xffff9699e4a0,  (0)] warning: QQmlApplicationEngine failed to load component

BENOIT92200, 2024-05-15:
# README

Hi
Thks for advice !

Well..still have this messages:
bt92200@BTRPI-5:~/Downloads$ ./webcamoid-portable-linux-9.1.1-arm64.appimage
[2024-05-13 18:50:33.305, Webcamoid, 0xffff9699e4a0,  (0)] warning: QQmlApplicationEngine failed to load component
[2024-05-13 18:50:33.305, Webcamoid, 0xffff9699e4a0, main.qml (23)] warning: qrc:/Webcamoid/share/qml/main.qml:23:1: module "QtQuick.Layouts" is not installed
[2024-05-13 18:50:33.305, Webcamoid, 0xffff9699e4a0, main.qml (21)] warning: qrc:/Webcamoid/share/qml/main.qml:21:1: module "QtQuick.Window" is not installed
[2024-05-13 18:50:33.305, Webcamoid, 0xffff9699e4a0, main.qml (-1)] warning: qrc:/Webcamoid/share/qml/main.qml: module "QtQml.WorkerScript" is not installed
[2024-05-13 18:50:33.305, Webcamoid, 0xffff9699e4a0, main.qml (23)] warning: qrc:/Webcamoid/share/qml/main.qml:23:1: module "QtQuick.Layouts" is not installed
[2024-05-13 18:50:33.305, Webcamoid, 0xffff9699e4a0, main.qml (21)] warning: qrc:/Webcamoid/share/qml/main.qml:21:1: module "QtQuick.Window" is not installed
[2024-05-13 18:50:33.305, Webcamoid, 0xffff9699e4a0, main.qml (-1)] warning: qrc:/Webcamoid/share/qml/main.qml: module "QtQml.WorkerScript" is not installed
[2024-05-13 18:50:33.305, Webcamoid, 0xffff9699e4a0, main.qml (23)] warning: qrc:/Webcamoid/share/qml/main.qml:23:1: module "QtQuick.Layouts" is not installed
[2024-05-13 18:50:33.305, Webcamoid, 0xffff9699e4a0, main.qml (21)] warning: qrc:/Webcamoid/share/qml/main.qml:21:1: module "QtQuick.Window" is not installed
[2024-05-13 18:50:33.305, Webcamoid, 0xffff9699e4a0, main.qml (-1)] warning: qrc:/Webcamoid/share/qml/main.qml: module "QtQml.WorkerScript" is not installed

hipersayanX, 2024-05-16:
Plugins seems to be build statically linked, therefore the deploy tools can't detect the dependencies.

BENOIT92200, 2024-05-16:
So ?
I am not an expert..... Please advise on what exactly to do about that.
Brgds.

hipersayanX, 2024-05-16:
I will make a new release with the fix.

BENOIT92200, 2024-05-17:
Ohhhh..
C U soon  😎

hipersayanX, 2024-05-26:
Right now I'm trying to add the DEB and RPM package formats (I had problem packaging it with AppImage), once done I will port the changes to 9.1.1 code and release. Once done, later,  I will ask you test it before release.

hipersayanX, 2024-10-01:
Fixed in 9.2.0.

closed 2024-10-01 by hipersayanX