webcamoid/webcamoid #701: [Feature Request] Dark mode toggle

carafelix, 2024-06-23:
# Want a new feature?

Feature: Add a dark/light toggle to the preferences menu

Because: Installing Webcamoid on Ubuntu 24 does not auto-recognize user default theme

hipersayanX, 2024-06-24:
>Installing Webcamoid on Ubuntu 24 does not auto-recognize user default theme

Are you sure? [Ubuntu default theme is actually light](https://github.com/ubuntu/yaru), in KDE can also catch the theme.

carafelix, 2024-06-24:
> Are you sure? 

[image]

version: 9.1.1-1build6
installed from the official Ubuntu repo

A little toggle won't hurt :P

hipersayanX, 2024-06-24:
But, in the screenshot, it's actually catching the theme, if you are referring to the window frame, that is not controlled by Webcamoid, that is controlled by the window manager, I can't fix that.

>A little toggle won't hurt :P

Yes, I know, that is a planned feature anyway, but it was strange to me that Webcamoid was not catching the theme.

hipersayanX, 2024-06-24:
Will keep opened since, this is still a valid feature request.

Greenbriggs9, 2025-07-24:
> @Greenbriggs9 the color scheme selection is available in the Daily Build, only available for [github sponsors](https://github.com/sponsors/hipersayanX). This feature will be available in the next stable release.
I paid for webcamoid yesterday. 
What is the difference between the paid (latest version), and the unpaid version? 

I didn't notice any difference at all

hipersayanX, 2025-07-24:
Uninstall the system version and execute the AppImage as explained in the
wiki
https://github.com/webcamoid/webcamoid/wiki/Installing-binary-packages-in-GNU-Linux#appimage

hipersayanX, 2025-07-29:
Color schemes added in Webcamoid 9.3.0, including dark themes, closing.