webcamoid/webcamoid #702: Fails to build with LTO due to ODR violations

eli-schwartz, 2024-07-07:
I tried to build with the following *FLAGS to optimize the build: `-flto=4 -Werror=odr -Werror=lto-type-mismatch -Werror=strict-aliasing`

Note the -Werror=* flags are used to help detect cases where the compiler tries to optimize by assuming UB cannot exist in the source code -- if it does exist, ordinarily the code would be miscompiled, and this says to make the miscompilation a fatal error.

I got this error:
```
[1/99] cd /var/tmp/portage/media-video/webcamoid-9999/work/webcamoid-9999_build/StandAlone/ManPages/src && /usr/bin/gzip -n -c9 webcamoid.1 > webcamoid.1.gz
[2/99] : && /usr/bin/x86_64-pc-linux-gnu-g++ -fPIC -pipe -march=native -fstack-protector-all -O2 -fdiagnostics-color=always -frecord-gcc-switches -flto=4 -Werror=odr -Werror=lto-type-mismatch -Werror=strict-aliasing  -Wformat -Werror=format-security  -Wl,-O1 -Wl,--as-needed -Wl,-z,pack-relative-relocs -flto=4 -Werror=odr -Werror=lto-type-mismatch -Werror=strict-aliasing -Wl,--defsym=__gentoo_check_ldflags__=0 -shared -Wl,-soname,libavkys.so.9 -o build/lib64/libavkys.so.9.2.0 libAvKys/Lib/CMakeFiles/avkys.dir/avkys_autogen/mocs_compilation.cpp.o libAvKys/Lib/CMakeFiles/avkys.dir/src/ak.cpp.o libAvKys/Lib/CMakeFiles/avkys.dir/src/akaudiocaps.cpp.o libAvKys/Lib/CMakeFiles/avkys.dir/src/akaudioconverter.cpp.o libAvKys/Lib/CMakeFiles/avkys.dir/src/akaudiopacket.cpp.o libAvKys/Lib/CMakeFiles/avkys.dir/src/akcaps.cpp.o libAvKys/Lib/CMakeFiles/avkys.dir/src/akcolorcomponent.cpp.o libAvKys/Lib/CMakeFiles/avkys.dir/src/akcolorconvert.cpp.o libAvKys/Lib/CMakeFiles/avkys.dir/src/akcolorplane.cpp.o libAvKys/Lib/CMakeFiles/avkys.dir/src/akcompressedvideocaps.cpp.o libAvKys/Lib/CMakeFiles/avkys.dir/src/akcompressedvideopacket.cpp.o libAvKys/Lib/CMakeFiles/avkys.dir/src/akelement.cpp.o libAvKys/Lib/CMakeFiles/avkys.dir/src/akfrac.cpp.o libAvKys/Lib/CMakeFiles/avkys.dir/src/akmultimediasourceelement.cpp.o libAvKys/Lib/CMakeFiles/avkys.dir/src/akpacket.cpp.o libAvKys/Lib/CMakeFiles/avkys.dir/src/akpacketbase.cpp.o libAvKys/Lib/CMakeFiles/avkys.dir/src/akplugininfo.cpp.o libAvKys/Lib/CMakeFiles/avkys.dir/src/akpluginmanager.cpp.o libAvKys/Lib/CMakeFiles/avkys.dir/src/aksubtitlecaps.cpp.o libAvKys/Lib/CMakeFiles/avkys.dir/src/aksubtitlepacket.cpp.o libAvKys/Lib/CMakeFiles/avkys.dir/src/akunit.cpp.o libAvKys/Lib/CMakeFiles/avkys.dir/src/akvideocaps.cpp.o libAvKys/Lib/CMakeFiles/avkys.dir/src/akvideoconverter.cpp.o libAvKys/Lib/CMakeFiles/avkys.dir/src/akvideoformatspec.cpp.o libAvKys/Lib/CMakeFiles/avkys.dir/src/akvideomixer.cpp.o libAvKys/Lib/CMakeFiles/avkys.dir/src/akvideopacket.cpp.o libAvKys/Lib/CMakeFiles/avkys.dir/src/qml/akcolorizedimage.cpp.o libAvKys/Lib/CMakeFiles/avkys.dir/src/qml/akpalette.cpp.o libAvKys/Lib/CMakeFiles/avkys.dir/src/qml/akpalettegroup.cpp.o libAvKys/Lib/CMakeFiles/avkys.dir/src/qml/aktheme.cpp.o libAvKys/Lib/CMakeFiles/avkys.dir/src/qml/akutils.cpp.o libAvKys/Lib/CMakeFiles/avkys.dir/avkys_autogen/EWIEGA46WW/qrc_AvKys.cpp.o  /usr/lib64/libQt6Quick.so.6.7.1  /usr/lib64/libQt6Widgets.so.6.7.1  /usr/lib64/libQt6QmlModels.so.6.7.1  /usr/lib64/libQt6Qml.so.6.7.1  /usr/lib64/libQt6QmlBuiltins.a  /usr/lib64/libQt6Network.so.6.7.1  /usr/lib64/libQt6OpenGL.so.6.7.1  /usr/lib64/libQt6Gui.so.6.7.1  /usr/lib64/libGLX.so  /usr/lib64/libOpenGL.so  /usr/lib64/libQt6Core.so.6.7.1 && :
FAILED: build/lib64/libavkys.so.9.2.0 
: && /usr/bin/x86_64-pc-linux-gnu-g++ -fPIC -pipe -march=native -fstack-protector-all -O2 -fdiagnostics-color=always -frecord-gcc-switches -flto=4 -Werror=odr -Werror=lto-type-mismatch -Werror=strict-aliasing  -Wformat -Werror=format-security  -Wl,-O1 -Wl,--as-needed -Wl,-z,pack-relative-relocs -flto=4 -Werror=odr -Werror=lto-type-mismatch -Werror=strict-aliasing -Wl,--defsym=__gentoo_check_ldflags__=0 -shared -Wl,-soname,libavkys.so.9 -o build/lib64/libavkys.so.9.2.0 libAvKys/Lib/CMakeFiles/avkys.dir/avkys_autogen/mocs_compilation.cpp.o libAvKys/Lib/CMakeFiles/avkys.dir/src/ak.cpp.o libAvKys/Lib/CMakeFiles/avkys.dir/src/akaudiocaps.cpp.o libAvKys/Lib/CMakeFiles/avkys.dir/src/akaudioconverter.cpp.o libAvKys/Lib/CMakeFiles/avkys.dir/src/akaudiopacket.cpp.o libAvKys/Lib/CMakeFiles/avkys.dir/src/akcaps.cpp.o libAvKys/Lib/CMakeFiles/avkys.dir/src/akcolorcomponent.cpp.o libAvKys/Lib/CMakeFiles/avkys.dir/src/akcolorconvert.cpp.o libAvKys/Lib/CMakeFiles/avkys.dir/src/akcolorplane.cpp.o libAvKys/Lib/CMakeFiles/avkys.dir/src/akcompressedvideocaps.cpp.o libAvKys/Lib/CMakeFiles/avkys.dir/src/akcompressedvideopacket.cpp.o libAvKys/Lib/CMakeFiles/avkys.dir/src/akelement.cpp.o libAvKys/Lib/CMakeFiles/avkys.dir/src/akfrac.cpp.o libAvKys/Lib/CMakeFiles/avkys.dir/src/akmultimediasourceelement.cpp.o libAvKys/Lib/CMakeFiles/avkys.dir/src/akpacket.cpp.o libAvKys/Lib/CMakeFiles/avkys.dir/src/akpacketbase.cpp.o libAvKys/Lib/CMakeFiles/avkys.dir/src/akplugininfo.cpp.o libAvKys/Lib/CMakeFiles/avkys.dir/src/akpluginmanager.cpp.o libAvKys/Lib/CMakeFiles/avkys.dir/src/aksubtitlecaps.cpp.o libAvKys/Lib/CMakeFiles/avkys.dir/src/aksubtitlepacket.cpp.o libAvKys/Lib/CMakeFiles/avkys.dir/src/akunit.cpp.o libAvKys/Lib/CMakeFiles/avkys.dir/src/akvideocaps.cpp.o libAvKys/Lib/CMakeFiles/avkys.dir/src/akvideoconverter.cpp.o libAvKys/Lib/CMakeFiles/avkys.dir/src/akvideoformatspec.cpp.o libAvKys/Lib/CMakeFiles/avkys.dir/src/akvideomixer.cpp.o libAvKys/Lib/CMakeFiles/avkys.dir/src/akvideopacket.cpp.o libAvKys/Lib/CMakeFiles/avkys.dir/src/qml/akcolorizedimage.cpp.o libAvKys/Lib/CMakeFiles/avkys.dir/src/qml/akpalette.cpp.o libAvKys/Lib/CMakeFiles/avkys.dir/src/qml/akpalettegroup.cpp.o libAvKys/Lib/CMakeFiles/avkys.dir/src/qml/aktheme.cpp.o libAvKys/Lib/CMakeFiles/avkys.dir/src/qml/akutils.cpp.o libAvKys/Lib/CMakeFiles/avkys.dir/avkys_autogen/EWIEGA46WW/qrc_AvKys.cpp.o  /usr/lib64/libQt6Quick.so.6.7.1  /usr/lib64/libQt6Widgets.so.6.7.1  /usr/lib64/libQt6QmlModels.so.6.7.1  /usr/lib64/libQt6Qml.so.6.7.1  /usr/lib64/libQt6QmlBuiltins.a  /usr/lib64/libQt6Network.so.6.7.1  /usr/lib64/libQt6OpenGL.so.6.7.1  /usr/lib64/libQt6Gui.so.6.7.1  /usr/lib64/libGLX.so  /usr/lib64/libOpenGL.so  /usr/lib64/libQt6Core.so.6.7.1 && :
/var/tmp/portage/media-video/webcamoid-9999/work/webcamoid-9999/libAvKys/Lib/src/akvideoconverter.cpp:69:6: error: type ‘AlphaMode’ violates the C++ One Definition Rule [-Werror=odr]
   69 | enum AlphaMode
      |      ^
/var/tmp/portage/media-video/webcamoid-9999/work/webcamoid-9999/libAvKys/Lib/src/akvideopacket.cpp:47:6: note: an enum with different value name is defined in another translation unit
   47 | enum AlphaMode
      |      ^
/var/tmp/portage/media-video/webcamoid-9999/work/webcamoid-9999/libAvKys/Lib/src/akvideoconverter.cpp:71:5: note: name ‘AlphaMode_AI_AO’ differs from name ‘AlphaMode_AO’ defined in another translation unit
   71 |     AlphaMode_AI_AO,
      |     ^
/var/tmp/portage/media-video/webcamoid-9999/work/webcamoid-9999/libAvKys/Lib/src/akvideopacket.cpp:49:5: note: mismatching definition
   49 |     AlphaMode_AO,
      |     ^
lto1: some warnings being treated as errors
lto-wrapper: fatal error: /usr/bin/x86_64-pc-linux-gnu-g++ returned 1 exit status
compilation terminated.
/usr/libexec/gcc/x86_64-pc-linux-gnu/ld: error: lto-wrapper failed
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
```

Originally reported downstream: https://bugs.gentoo.org/927104
Full build log: [build.log](https://github.com/user-attachments/files/16121296/build.log)

Note: the original report was with 9.1.1 and my reproducer builds webcamoid from git master:

```
 * Checking out https://github.com/webcamoid/webcamoid.git to /var/tmp/portage/media-video/webcamoid-9999/work/webcamoid-9999 ...
git checkout --quiet master
GIT update -->
   repository:               https://github.com/webcamoid/webcamoid.git
   at the commit:            ee95266cc97535539496c68557058eed1cd3c7ea
```

https://github.com/webcamoid/webcamoid/blob/ee95266cc97535539496c68557058eed1cd3c7ea/libAvKys/Lib/src/akvideopacket.cpp#L47-L51
https://github.com/webcamoid/webcamoid/blob/ee95266cc97535539496c68557058eed1cd3c7ea/libAvKys/Lib/src/akvideoconverter.cpp#L69-L75

hipersayanX, 2024-07-08:
Will check, thanks!

hipersayanX, 2024-10-01:
Fixed issues in 9.2.0 and added **ENABLE_IPO** for tracking this issues.

closed 2024-10-01 by hipersayanX