webcamoid/webcamoid #703: No longer works - main libvlc error: stale plugins cache: modified

ratsalad, 2024-07-15:
00005560afe098e0] main libvlc error: stale plugins cache: modified /home/jim/Webcamoid/lib/vlc/plugins/audio_filter/libaudiobargraph_a_plugin.so
[00005560afe098e0] main libvlc error: stale plugins cache: modified /home/jim/Webcamoid/lib/vlc/plugins/audio_filter/libremap_plugin.so
[00005560afe098e0] main libvlc error: stale plugins cache: modified /home/jim/Webcamoid/lib/vlc/plugins/audio_filter/libtospdif_plugin.so
[00005560afe098e0] main libvlc error: stale plugins cache: modified /home/jim/Webcamoid/lib/vlc/plugins/audio_filter/libheadphone_channel_mixer_plugin.so
[00005560afe098e0] main libvlc error: stale plugins cache: modified /home/jim/Webcamoid/lib/vlc/plugins/audio_filter/libkaraoke_plugin.so
[00005560afe098e0] main libvlc error: stale plugins cache: modified /home/jim/Webcamoid/lib/vlc/plugins/audio_filter/libspatializer_plugin.so
[00005560afe098e0] main libvlc error: stale plugins cache: modified /home/jim/Webcamoid/lib/vlc/plugins/audio_filter/libgain_plugin.so
[00005560afe098e0] main libvlc error: stale plugins cache: modified /home/jim/Webcamoid/lib/vlc/plugins/audio_filter/libtrivial_channel_mixer_plugin.so
[00005560afe098e0] main libvlc error: stale plugins cache: modified /home/jim/Webcamoid/lib/vlc/plugins/audio_filter/libequalizer_plugin.so
[00005560afe098e0] main libvlc error: stale plugins cache: modified /home/jim/Webcamoid/lib/vlc/plugins/audio_filter/libscaletempo_plugin.so
[00005560afe098e0] main libvlc error: stale plugins cache: modified /home/jim/Webcamoid/lib/vlc/plugins/audio_filter/libmono_plugin.so
[00005560afe098e0] main libvlc error: stale plugins cache: modified /home/jim/Webcamoid/lib/vlc/plugins/audio_filter/libsamplerate_plugin.so
[00005560afe098e0] main libvlc error: stale plugins cache: modified /home/jim/Webcamoid/lib/vlc/plugins/audio_filter/libaudio_format_plugin.so
[00005560afe098e0] main libvlc error: stale plugins cache: modified /home/jim/Webcamoid/lib/vlc/plugins/audio_filter/libstereo_widen_plugin.so
[00005560afe098e0] main libvlc error: stale plugins cache: modified /home/jim/Webcamoid/lib/vlc/plugins/audio_filter/libmad_plugin.so
[00005560afe098e0] main libvlc error: stale plugins cache: modified /home/jim/Webcamoid/lib/vlc/plugins/audio_filter/libspeex_resampler_plugin.so
[00005560afe098e0] main libvlc error: stale plugins cache: modified /home/jim/Webcamoid/lib/vlc/plugins/audio_filter/libchorus_flanger_plugin.so
[00005560afe098e0] main libvlc error: stale plugins cache: modified /home/jim/Webcamoid/lib/vlc/plugins/audio_filter/libscaletempo_pitch_plugin.so
[00005560afe098e0] main libvlc error: stale plugins cache: modified /home/jim/Webcamoid/lib/vlc/plugins/audio_filter/libspatialaudio_plugin.so
[00005560afe098e0] main libvlc error: stale plugins cache: modified /home/jim/Webcamoid/lib/vlc/plugins/audio_filter/libsimple_channel_mixer_plugin.so
[00005560afe098e0] main libvlc error: stale plugins cache: modified /home/jim/Webcamoid/lib/vlc/plugins/audio_filter/libnormvol_plugin.so
[00005560afe098e0] main libvlc error: stale plugins cache: modified /home/jim/Webcamoid/lib/vlc/plugins/audio_filter/libparam_eq_plugin.so
[00005560afe098e0] main libvlc error: stale plugins cache: modified /home/jim/Webcamoid/lib/vlc/plugins/audio_filter/libsoxr_plugin.so
[00005560afe098e0] main libvlc error: stale plugins cache: modified /home/jim/Webcamoid/lib/vlc/plugins/audio_filter/libugly_resampler_plugin.so
[00005560afe098e0] main libvlc error: stale plugins cache: modified /home/jim/Webcamoid/lib/vlc/plugins/audio_filter/libdolby_surround_decoder_plugin.so
[00005560afe098e0] main libvlc error: stale plugins cache: modified /home/jim/Webcamoid/lib/vlc/plugins/audio_filter/libcompressor_plugin.so
[00005560afe098e0] main libvlc error: stale plugins cache: modified /home/jim/Webcamoid/lib/vlc/plugins/access_output/libaccess_output_shout_plugin.so
[00005560afe098e0] main libvlc error: stale plugins cache: modified /home/jim/Webcamoid/lib/vlc/plugins/access_output/libaccess_output_file_plugin.so
[00005560afe098e0] main libvlc error: stale plugins cache: modified /home/jim/Webcamoid/lib/vlc/plugins/access_output/libaccess_output_dummy_plugin.so
[00005560afe098e0] main libvlc error: stale plugins cache: modified /home/jim/Webcamoid/lib/vlc/plugins/access_output/libaccess_output_http_plugin.so
[00005560afe098e0] main libvlc error: stale plugins cache: modified /home/jim/Webcamoid/lib/vlc/plugins/access_output/libaccess_output_livehttp_plugin.so
[00005560afe098e0] main libvlc error: stale plugins cache: modified /home/jim/Webcamoid/lib/vlc/plugins/access_output/libaccess_output_udp_plugin.so
[00005560afe098e0] main libvlc error: stale plugins cache: modified /home/jim/Webcamoid/lib/vlc/plugins/access_output/libaccess_output_rist_plugin.so
[00005560afe098e0] main libvlc error: stale plugins cache: modified /home/jim/Webcamoid/lib/vlc/plugins/lua/liblua_plugin.so
[00005560afe098e0] main libvlc error: stale plugins cache: modified /home/jim/Webcamoid/lib/vlc/plugins/audio_output/libsndio_plugin.so
[00005560afe098e0] main libvlc error: stale plugins cache: modified /home/jim/Webcamoid/lib/vlc/plugins/audio_output/libpulse_plugin.so
[00005560afe098e0] main libvlc error: stale plugins cache: modified /home/jim/Webcamoid/lib/vlc/plugins/audio_output/libalsa_plugin.so
[00005560afe098e0] main libvlc error: stale plugins cache: modified /home/jim/Webcamoid/lib/vlc/plugins/audio_output/libamem_plugin.so
[00005560afe098e0] main libvlc error: stale plugins cache: modified /home/jim/Webcamoid/lib/vlc/plugins/audio_output/libadummy_plugin.so
[00005560afe098e0] main libvlc error: stale plugins cache: modified /home/jim/Webcamoid/lib/vlc/plugins/audio_output/libafile_plugin.so
[00005560afe098e0] main libvlc error: stale plugins cache: modified /home/jim/Webcamoid/lib/vlc/plugins/video_filter/libcroppadd_plugin.so
[00005560afe098e0] main libvlc error: stale plugins cache: modified /home/jim/Webcamoid/lib/vlc/plugins/video_filter/libball_plugin.so
[00005560afe098e0] main libvlc error: stale plugins cache: modified /home/jim/Webcamoid/lib/vlc/plugins/video_filter/libblend_plugin.so
[00005560afe098e0] main libvlc error: stale plugins cache: modified /home/jim/Webcamoid/lib/vlc/plugins/video_filter/libgradfun_plugin.so
[00005560afe098e0] main libvlc error: stale plugins cache: modified /home/jim/Webcamoid/lib/vlc/plugins/video_filter/libhqdn3d_plugin.so
[00005560afe098e0] main libvlc error: stale plugins cache: modified /home/jim/Webcamoid/lib/vlc/plugins/video_filter/libposterize_plugin.so
[00005560afe098e0] main libvlc error: stale plugins cache: modified /home/jim/Webcamoid/lib/vlc/plugins/video_filter/libmirror_plugin.so
[00005560afe098e0] main libvlc error: stale plugins cache: modified /home/jim/Webcamoid/lib/vlc/plugins/video_filter/libcolorthres_plugin.so
[00005560afe098e0] main libvlc error: stale plugins cache: modified /home/jim/Webcamoid/lib/vlc/plugins/video_filter/liboldmovie_plugin.so
[00005560afe098e0] main libvlc error: stale plugins cache: modified /home/jim/Webcamoid/lib/vlc/plugins/video_filter/libanaglyph_plugin.so
[00005560afe098e0] main libvlc error: stale plugins cache: modified /home/jim/Webcamoid/lib/vlc/plugins/video_filter/libpsychedelic_plugin.so
[00005560afe098e0] main libvlc error: stale plugins cache: modified /home/jim/Webcamoid/lib/vlc/plugins/video_filter/libgaussianblur_plugin.so
[00005560afe098e0] main libvlc error: stale plugins cache: modified /home/jim/Webcamoid/lib/vlc/plugins/video_filter/libbluescreen_plugin.so
[00005560afe098e0] main libvlc error: stale plugins cache: modified /home/jim/Webcamoid/lib/vlc/plugins/video_filter/libsharpen_plugin.so
[00005560afe098e0] main libvlc error: stale plugins cache: modified /home/jim/Webcamoid/lib/vlc/plugins/video_filter/libfreeze_plugin.so
[00005560afe098e0] main libvlc error: stale plugins cache: modified /home/jim/Webcamoid/lib/vlc/plugins/video_filter/libmotiondetect_plugin.so
[00005560afe098e0] main libvlc error: stale plugins cache: modified /home/jim/Webcamoid/lib/vlc/plugins/video_filter/libpostproc_plugin.so
[00005560afe098e0] main libvlc error: stale plugins cache: modified /home/jim/Webcamoid/lib/vlc/plugins/video_filter/libvhs_plugin.so
[00005560afe098e0] main libvlc error: stale plugins cache: modified /home/jim/Webcamoid/lib/vlc/plugins/video_filter/libscene_plugin.so
[00005560afe098e0] main libvlc error: stale plugins cache: modified /home/jim/Webcamoid/lib/vlc/plugins/video_filter/libgrain_plugin.so
[... cut ...]
[00005560afe098e0] main libvlc error: stale plugins cache: modified /home/jim/Webcamoid/lib/vlc/plugins/services_discovery/libxcb_apps_plugin.so
[00005560afe098e0] main libvlc error: stale plugins cache: modified /home/jim/Webcamoid/lib/vlc/plugins/services_discovery/libmtp_plugin.so
[00005560afe098e0] main libvlc error: stale plugins cache: modified /home/jim/Webcamoid/lib/vlc/plugins/services_discovery/libavahi_plugin.so
[00005560afe098e0] main libvlc error: stale plugins cache: modified /home/jim/Webcamoid/lib/vlc/plugins/services_discovery/libupnp_plugin.so

hipersayanX, 2024-07-16:
These errors doesn't affects the normal run of Webcamoid, the error are just telling that the cache files has been moved to a different place from where it supposed to be. The messages disappears after the next run.
The messages can't fixed, only hidden.

ratsalad, 2024-07-16:
Ok. I just posted what was presented after running /home/jim/Webcamoid/webcamoid.sh because I thought it was somehow related to webcamoid not starting, but if they are irrelevant to wecamoid not loading what other information could I provide as it isn't working anymore. 

I just uncommented 
# Enable plugin debugging
export QT_DEBUG_PLUGINS=1

If this output is of any use:

 [2024-07-16 17:46:42.772, Webcamoid, 0x7f0189c0d7c0,  (0)] warning: Found metadata in lib /home/jim/Webcamoid/lib/avkys/libVirtualCamera.so, metadata=
{
    "IID": "org.avkys.plugin",
    "MetaData": {
        "plugins": [
            {
                "depends": [
                    "VideoSink/VirtualCamera/Impl/*"
                ],
                "description": "Create a virtual webcam to share with other programs",
                "id": "VideoSink/VirtualCamera",
                "implements": [
                    "Element"
                ],
                "name": "VirtualCamera",
                "type": "qtplugin"
            }
        ],
        "type": "WebcamoidPluginsCollection"
    },
    "archreq": 0,
    "className": "VirtualCamera",
    "debug": false,
    "version": 331520
}

[2024-07-16 17:46:42.772, Webcamoid, 0x7f0189c0d7c0,  (0)] debug: loaded library "/home/jim/Webcamoid/lib/avkys/libVirtualCamera.so"
[2024-07-16 17:46:42.772, Webcamoid, 0x7f0189c0d7c0,  (0)] warning: Found metadata in lib /home/jim/Webcamoid/lib/avkys/libVirtualCamera_v4l2lb.so, metadata=
{
    "IID": "org.avkys.plugin",
    "MetaData": {
        "plugins": [
            {
                "depends": [
                    "VideoFilter/Flip",
                    "VideoFilter/SwapRB"
                ],
                "description": "Create a virtual webcam using v4l2loopback driver",
                "id": "VideoSink/VirtualCamera/Impl/v4l2loopback",
                "implements": [
                    "VirtualCameraImpl"
                ],
                "name": "v4l2loopback",
                "priority": 900,
                "type": "qtplugin"
            }
        ],
        "type": "WebcamoidPluginsCollection"
    },
    "archreq": 0,
    "className": "Plugin",
    "debug": false,
    "version": 331520
}

[20

hipersayanX, 2024-07-16:
Just in case, are you using Wayland?

ratsalad, 2024-07-16:
I just reinstalled mint xfce and a RX580 so I don't think Wayland is an issue. Webcamoid was working after I reinstalled, but now it is not.

ratsalad, 2024-07-16:
I had It working for a couple of months and it just recently quit working. I am not quite certain, but may have quit working when the latest linux mint kernel was installed.

5.15.0-116-generic #126-Ubuntu SMP Mon Jul 1 10:14:24 UTC 2024 x86_64 x86_64 x86_64 GNU/Linux

hipersayanX, 2024-07-16:
Wayland is the actual issue in your case, because when Webcamoid 9.1.1 was packaged (1 year ago), it didn't take into account the possible conflicts with Wayland libraries. This bug was already fixed in the daily build (next Webcamoid 9.2), but I can't make a new release until I fix the bugs with Android.

hipersayanX, 2024-07-16:
The binaries works perfectly fine in X11.

ratsalad, 2024-07-17:
Does Linux Mint 21.3 XFCE even include Wayland support? as it says: "Linux Mint 21.3, the next major release of the popular Ubuntu-based Linux distro, will include experimental Wayland support for the Cinnamon desktop"

https://www.omgubuntu.co.uk/2023/10/linux-mint-21-3-experimental-wayland-support

It doesn't like Wayland is even in use.

echo $XDG_SESSION_TYPE
x11

ratsalad, 2024-07-17:
Linux Mint 21.3 XFCE is using X11 so..

It might be working ok for others, but webcamoid is not working perfectly in X11 for me.

hipersayanX, 2024-07-17:
It show any error message (outside VLC related ones), or the program just get stuck in the command line without starting the UI?

ratsalad, 2024-07-18:
I am not sure what is happening as it just quite working about a week ago. 

In a shell, It just seems to just stop loading. 

I did have what might not be a related issue where the last kernel updated was causing issues for Virtualbox. I just removed the kernel that was causing the issue.

hipersayanX, 2024-07-19:
I'm pretty sure this is due to a  conflict with some X11 library.
Try the following, in Webcamoid's ```lib``` folder create a ```backup``` folder and move the following files there:

```
libICE.so.6
libXcomposite.so.1
```

and try starting webcamoid.
Does it:

- Start successfully?
- Fails with dependency missing?
- Keep stuck in terminal?

ratsalad, 2024-07-19:
Still stuck with the last few lines being:

[2024-07-18 22:50:47.116, Webcamoid, 0x7092ee3cd7c0,  (0)] debug: loaded library "/home/jim/.Webcamoid/lib/qt/plugins/imageformats/libqjpeg.so"
[2024-07-18 22:50:47.116, Webcamoid, 0x7092ee3cd7c0,  (0)] debug: loaded library "/home/jim/.Webcamoid/lib/qt/plugins/imageformats/libqsvg.so"

hipersayanX, 2024-07-24:
Sorry for not answering before. Can you send me the buy ticket so I can send you the daily build?

ratsalad, 2024-07-24:
It isn't real obvious to me what the buy ticket is or what email I would send something to. Is the buy ticket from Payhip?

hipersayanX, 2024-07-24:
Yes, the contact mail is in the ticket.

ratsalad, 2024-07-25:
I just forwarded the paypal receipt to the email address listed on it.

ratsalad, 2024-07-26:
OK I just decided to try the daily release using a video file or url and it doesn't seem to work with the any of the ones I tried so only the webcam output and image files are working for me.

hipersayanX, 2024-07-26:
I don't understand, the video playback doesn't works? if that the case I need the files or the URL to test.

ratsalad, 2024-07-26:
I load a video and there is nothing displayed in the in the window and there is no video output. I can make a video if you like, but ! essentially videos aren't working with the version I am using.

hipersayanX, 2024-07-26:
Try this: Go to Preferences -> General Options -> Video playback -> Select either VLC or FFmpeg

ratsalad, 2024-07-26:
I checked and VLC is already the default choice, but I don't see any other options, The dropdown doesn't seem to work like it should or there is no other options to choose from like FFMPEG.

hipersayanX, 2024-07-26:
Give me a screenshot of the General options section, and the console log.

ratsalad, 2024-07-26:
[image]

hipersayanX, 2024-07-26:
I need the logs too.

ratsalad, 2024-07-26:
Yeah I see that, but could you just suggest the best way to get them?

hipersayanX, 2024-07-26:
Oh, wait

>Linux Mint 21 is based on Ubuntu 22.04. Make sure to read the [Ubuntu release notes](https://discourse.ubuntu.com/t/jammy-jellyfish-release-notes/24668). [source](https://www.linuxmint.com/rel_virginia.php)

You must use Linux Mint 22 or higher.

hipersayanX, 2024-07-26:
>Yeah I see that, but could you just suggest the best way to get them?

run Webcamoid from terminal:

```
cd ~/Webcamoid/bin
./webcamoid
```

copy all output after that commands and upload to [github gist](https://gist.github.com).

hipersayanX, 2024-07-26:
My guess right now is that Webcamoid is not loading the FFmmpeg libraries because you are using an outdated version of Mint.

ratsalad, 2024-07-26:
The new version of Linux Mint 22 was only released yesterday so I am not able to upgrade just yet so I may just wait until then. I also haven't had time to see if Wayland will be integrated into the new release.

ratsalad, 2024-07-26:
I was also thinking of the screen command I used last time to capture the output instead. I just have look back a the command line  options as I don't use it too much and can't remember what I did.

hipersayanX, 2024-07-26:
>The new version of Linux Mint 22 was only released yesterday so I am not able to upgrade just yet so I may just wait until then.

 Again, you don't need to install LM, just run it in live mode and test webcamoid there.

ratsalad, 2024-07-26:
Yeah I could do that, but I think I will just wait until I upgrade my system instead.

ratsalad, 2024-07-28:
I have upgraded to Mint 22 and the majority of the issues I was experiencing have disappeared.

closed 2024-07-28 by ratsalad