webcamoid/webcamoid #705: Flatpak distribution uses end-of-life runtimes(org.kde.Platform, org.freedesktop.Platform.GL.default)

brlin-tw, 2024-07-22:
## Summary

Flatpak distribution uses end-of-life runtimes

## Current Behavior

When running the `flatpak update` command the following warning message is printed:

```output
Info: runtime org.kde.Platform branch 5.15-21.08 is end-of-life, with reason:
   We strongly recommend moving to the latest stable version of the Platform and SDK
Info: applications using this runtime:
   io.github.webcamoid.Webcamoid

Info: runtime org.freedesktop.Platform.GL.default branch 21.08 is end-of-life, with reason:
   org.freedesktop.Platform 21.08 is no longer receiving fixes and security updates. Please update to a supported runtime version.
Info: applications using this extension:
   io.github.webcamoid.Webcamoid
```

It appears that currently, the Webcamoid Flatpak uses EoL runtimes and may have security issues, also refer to the FlatHub listing warning:

[image]

## Expected Behavior

Flatpak uses maintained runtimes.

## Steps to Reproduce

1. Run the `flatpak update` command in a terminal while the Flatpak is installed.

## Suggestions and tips

I suggest updating the package recipe, unfortunately, I'm not familiar with Flatpak and am currently unable to help on this one.

## Your Environment

* Webcamoid information: 9.0.0 amd64
* Operating System information: Ubuntu 24.04 AMD64
* Any other useful information: 
    + Flatpak 1.14.6

hipersayanX, 2024-07-24:
Sorry for not answering before, can you send me the buy ticket (to my mail, **DO NOT** post it here), so I can send you the daily build?

brlin-tw, 2024-07-24:
> Sorry for not answering before, can you send me the buy ticket (to my mail, DO NOT post it here), so I can send you the daily build?

Hi, thanks for getting back to me. I'm not sure what you mean by 'buy ticket', could you please clarify?

hipersayanX, 2024-07-24:
Did you bought the flatpak from Payhip or from Flathub?

brlin-tw, 2024-09-18:
> Did you bought the flatpak from Payhip or from Flathub?

Apologies for the ignorance.  I don't have access to the system at the moment however I believe that the Flatpak distribution I used is from Flathub.

hipersayanX, 2024-09-19:
Ok, because this issue was fixed in the daily build.

hipersayanX, 2024-10-01:
Fixed in 9.2.0.

closed 2024-10-01 by hipersayanX