webcamoid/webcamoid #706: Not installed error

nicosalto, 2024-07-23:
I downloaded the latest flatpack for the software on the official website and installed it with 
flatpak install --user webcamoid-*.flatpak

and then i tried to launch it with 
flatpak run io.github.webcamoid

but then i get the error
error: app/io.github.webcamoid/x86_64/master not installed

iam on Linux Mint 21 Cinnamon ( 5.4.12 ) on an intel CPU with an nvidia GPU.

Iam sure its a mistake on my part, let me know how to proceed please.

nicosalto, 2024-07-23:
ok i figured it out the correct command to launch it is 
 flatpak run io.github.webcamoid.Webcamoid

hipersayanX, 2024-07-24:
Sorry for not answering before. 

>ok i figured it out the correct command to launch it is
>flatpak run io.github.webcamoid.Webcamoid

Different to raw packages (installer, portables, et-al), flatpaks are containerized packages, executables can't be  run directly you must invoke the command with ```flatpak run``` always. This is explained in the README note when downloading the flatpak.

closed 2024-10-01 by hipersayanX