webcamoid/webcamoid #707: low fps above 640x480 resolution using windows 7 64bit

jwaves, 2024-08-06:
I am using windows 7 with a good computer. I am getting low fps on any resolution above 640x480. I tested Debut webcam software with windows 7 and my webcam and  I can get 30fps @ 2048x1536. So I know windows 7 and my camera are okay. In webcamoid I only have the option for 5 fps at resolutions above 800x600 in the Video options menu. Any advice how to get Webcamoid to increase its framerate ?

hipersayanX, 2024-08-06:
In Windows 7, Webcamoid depends of DirectShow for capturing the camera, if DirectShow does not offers MJPG or H264 formats I can't do nothing. Media Foundation may work probably (not sure) but you need Window 8 if I remember well.

jwaves, 2024-08-11:
So I am still not sure what a good next step to try and get mjpeg or other video format to work instead of the yuyv video format ?

hipersayanX, 2024-08-11:
If you want I can send you the daily build to test it, there are some bugs I have fixed recently related to frame rate detection in Windows, not sure if these fixes addresses your case specifically, may be works maybe not, but let's try. Send me the buy ticket to my mail (**NOT HERE**), and I will send you the daily build.

jwaves, 2024-08-12:
I bought a week ago, what email do I use ?

hipersayanX, 2024-08-12:
hipersayan _dot_ x _at_ gmail _dot_ com (if I remember well, the mail also appears in the ticket).

closed 2024-10-01 by hipersayanX