webcamoid/webcamoid #711: Webcamoid 9.1.1 purchases portable version doesn't start on Windows 11

lukamrkonjic, 2024-09-16:
# README

Some considerations before doing anything:

* Search the [issues list](https://github.com/webcamoid/webcamoid/issues) for similar topics before opening a new one.
* If you know how to fix the problem, consider doing a [pull request](https://github.com/webcamoid/webcamoid/pulls) instead of opening a new issue.
* Only report problems in [latest](https://github.com/webcamoid/webcamoid/releases) and [development](https://github.com/webcamoid/webcamoid/) version. Reporting issues related to older versions will be rejected.
* Use [gist](https://gist.github.com/) to post logs longer than 1024 characters.
* Webcamoid as project, is not affiliated or endorsed to any distribution, report packaging problems in their respective issue tracker.
* Respect the templates, we need as much information as possible.
* Don't open an issue and disappear, we need you at least the first week to clear up missing information.
* Missing information makes useless and unsolvable an issue report.
* You are our debugger, eyes and hands, if an issue is not reproducible then you will be the only person able to solve it.
* Take a seat and wait for your turn, as many others that has their issues open, or much better try fixing it your self and collaborate with the solution :smile:

Choose one of the templates below that fit better your issue.

# Reporting a problem?

## Summary

Double clicking / starting the binary .exe doesn't open the program, but I can see it running in task manager. Same with the .bat file. 

It DOES open after about 15 minutes, but I'm not sure why the delay is there. 

## Current Behavior

I double click - wait, nothing happens, but program is visible in task manager details - software opens after about 15 minutes.

## Expected Behavior

It should open directly.

## Steps to Reproduce

1. Purchase newest portable binary build 9.1.1
2. Extract using 7zip
3. Double click .exe / .bat

## Suggestions and tips

How would you solve the problem?

## Your Environment

* Webcamoid information: 9.1.1
* Operating System information: Windows 11 LTSC IOT
* Any other useful information: 

# Want a new feature?

Describe your idea the best as you can, include sketches, mockups and diagrams if required.  
Be patient, take in mind that there may be other priorities. Your idea will be accepted if fit in project goals.

# Questions and other matters?

Please help!

Go ahead!

lukamrkonjic, 2024-09-16:
Been rebooting my system and doing nothing special, and now it seems to work! No clue why or what happened with my system in the meantime, but it launches now, though to a delay of about 5-10 seconds, which still is quite slow for a camera app.

hipersayanX, 2024-09-16:
>but it launches now, though to a delay of about 5-10 seconds, which still is quite slow for a camera app.

I can't do much about that for now, maybe a solution would be loading the plugins asynchronously but I have to refactor the code a lot.

hipersayanX, 2024-09-16:
I will send you to daily build for testing, maybe the bug was fixed there. Is the same mail from your paypal account?

>Is there any log file / way to error check

The daily build has that feature.

lukamrkonjic, 2024-09-16:
Please do! Yes, indeed :)

hipersayanX, 2024-09-16:
Sent. Also, just in case sent as hipersayan dot x at g mail, don't download from anyone else.

lukamrkonjic, 2024-09-16:
Sweet, thanks. I’ll give it a go tommorow morning, not by the pc now :) I’ll report back!

hipersayanX, 2024-09-29:
I think the problem happens after installing the virtual camera, with the virtual camera uninstalled it starts in < 1 min. I will release 9.2.0, then work on the virtual camera related issues.