webcamoid/webcamoid #714: 9.1.1 purchased portable distribution for ARM64 architecture hang during launch on Raspberry Pi 5 w/ Raspberry Pi OS 12

brlin-tw, 2024-09-18:
## Summary

9.1.1 purchased portable distribution for ARM64 architecture didn't run on Raspberry Pi w/ Raspberry Pi OS 12.

## Current Behavior

* Double click and select execute action against the webcamoid.sh shell script didn't launch the application window.
* Running the shellscript in a text terminal reveals the following error output and the program hangs:

    ```txt
    ./webcamoid.sh 
    [2024-09-18 13:53:09.305, Webcamoid, 0x7ffed752ac00,  (0)] warning: QQmlApplicationEngine failed to load component
    [2024-09-18 13:53:09.305, Webcamoid, 0x7ffed752ac00, main.qml (23)] warning: qrc:/Webcamoid/share/qml/main.qml:23:1: module "QtQuick.Layouts" is not installed
    [2024-09-18 13:53:09.305, Webcamoid, 0x7ffed752ac00, main.qml (21)] warning: qrc:/Webcamoid/share/qml/main.qml:21:1: module "QtQuick.Window" is not installed
    [2024-09-18 13:53:09.305, Webcamoid, 0x7ffed752ac00, main.qml (-1)] warning: qrc:/Webcamoid/share/qml/main.qml: module "QtQml.WorkerScript" is not installed
    [2024-09-18 13:53:09.305, Webcamoid, 0x7ffed752ac00, main.qml (23)] warning: qrc:/Webcamoid/share/qml/main.qml:23:1: module "QtQuick.Layouts" is not installed
    [2024-09-18 13:53:09.305, Webcamoid, 0x7ffed752ac00, main.qml (21)] warning: qrc:/Webcamoid/share/qml/main.qml:21:1: module "QtQuick.Window" is not installed
    [2024-09-18 13:53:09.305, Webcamoid, 0x7ffed752ac00, main.qml (-1)] warning: qrc:/Webcamoid/share/qml/main.qml: module "QtQml.WorkerScript" is not installed
    [2024-09-18 13:53:09.305, Webcamoid, 0x7ffed752ac00, main.qml (23)] warning: qrc:/Webcamoid/share/qml/main.qml:23:1: module "QtQuick.Layouts" is not installed
    [2024-09-18 13:53:09.305, Webcamoid, 0x7ffed752ac00, main.qml (21)] warning: qrc:/Webcamoid/share/qml/main.qml:21:1: module "QtQuick.Window" is not installed
    [2024-09-18 13:53:09.305, Webcamoid, 0x7ffed752ac00, main.qml (-1)] warning: qrc:/Webcamoid/share/qml/main.qml: module "QtQml.WorkerScript" is not installed
    ```

## Expected Behavior

The application should launch normally as all the dependencies should be available in the portable distribution.

## Steps to Reproduce

1. Purchase ARM64 portable distribution
2. Extract tarball
3. Launch the application

## Additional information

[QML_IMPORT_TRACE output](https://github.com/user-attachments/files/17039181/qml-import-trace.out.log)

brlin-tw, 2024-09-18:
The QtQuick.Layouts and the QtQuick.Window module do not exist in the qml search paths:

```txt
$ ls lib/qt/qml/QtQuick/
Controls/             Dialogs/              libqtquick2plugin.so  plugins.qmltypes      qmldir                
```

hipersayanX, 2024-09-19:
This issue was fixed in the daily build, I will send you it. Is the same mail from Paypal, right?

brlin-tw, 2024-09-19:
The daily build errored with the following output:

```txt
$ ./webcamoid.sh 
webcamoid: /lib/aarch64-linux-gnu/libstdc++.so.6: version `GLIBCXX_3.4.32' not found (required by webcamoid)
webcamoid: /lib/aarch64-linux-gnu/libm.so.6: version `GLIBC_2.38' not found (required by webcamoid-portable-linux-daily-master-arm64/bin/../lib/libQt6Quick.so.6)
webcamoid: /lib/aarch64-linux-gnu/libc.so.6: version `GLIBC_2.38' not found (required by webcamoid-portable-linux-daily-master-arm64/bin/../lib/libQt6Quick.so.6)
webcamoid: /lib/aarch64-linux-gnu/libm.so.6: version `GLIBC_2.38' not found (required by webcamoid-portable-linux-daily-master-arm64/bin/../lib/libQt6Qml.so.6)
webcamoid: /lib/aarch64-linux-gnu/libstdc++.so.6: version `GLIBCXX_3.4.32' not found (required by webcamoid-portable-linux-daily-master-arm64/bin/../lib/libQt6Qml.so.6)
webcamoid: /lib/aarch64-linux-gnu/libc.so.6: version `GLIBC_2.38' not found (required by webcamoid-portable-linux-daily-master-arm64/bin/../lib/libQt6Network.so.6)
webcamoid: /lib/aarch64-linux-gnu/libm.so.6: version `GLIBC_2.38' not found (required by webcamoid-portable-linux-daily-master-arm64/bin/../lib/libQt6Gui.so.6)
webcamoid: /lib/aarch64-linux-gnu/libc.so.6: version `GLIBC_2.38' not found (required by webcamoid-portable-linux-daily-master-arm64/bin/../lib/libQt6Gui.so.6)
webcamoid: /lib/aarch64-linux-gnu/libm.so.6: version `GLIBC_2.38' not found (required by webcamoid-portable-linux-daily-master-arm64/bin/../lib/libicui18n.so.74)
webcamoid: /lib/aarch64-linux-gnu/libc.so.6: version `GLIBC_2.38' not found (required by webcamoid-portable-linux-daily-master-arm64/bin/../lib/libicui18n.so.74)
webcamoid: /lib/aarch64-linux-gnu/libm.so.6: version `GLIBC_2.38' not found (required by webcamoid-portable-linux-daily-master-arm64/bin/../lib/libicuuc.so.74)
webcamoid: /lib/aarch64-linux-gnu/libc.so.6: version `GLIBC_2.38' not found (required by webcamoid-portable-linux-daily-master-arm64/bin/../lib/libicuuc.so.74)
webcamoid: /lib/aarch64-linux-gnu/libm.so.6: version `GLIBC_2.38' not found (required by webcamoid-portable-linux-daily-master-arm64/bin/../lib/./libduktape.so.207)
webcamoid: /lib/aarch64-linux-gnu/libc.so.6: version `GLIBC_2.38' not found (required by webcamoid-portable-linux-daily-master-arm64/bin/../lib/./libduktape.so.207)
webcamoid: /lib/aarch64-linux-gnu/libc.so.6: version `GLIBC_2.38' not found (required by webcamoid-portable-linux-daily-master-arm64/bin/../lib/./libgnutls.so.30)
webcamoid: /lib/aarch64-linux-gnu/libc.so.6: version `GLIBC_2.38' not found (required by webcamoid-portable-linux-daily-master-arm64/bin/../lib/./libldap.so.2)
webcamoid: /lib/aarch64-linux-gnu/libc.so.6: version `GLIBC_2.38' not found (required by webcamoid-portable-linux-daily-master-arm64/bin/../lib/./liblber.so.2)
webcamoid: /lib/aarch64-linux-gnu/libc.so.6: version `GLIBC_2.38' not found (required by webcamoid-portable-linux-daily-master-arm64/bin/../lib/./libcrypto.so.3)
webcamoid: /lib/aarch64-linux-gnu/libc.so.6: version `GLIBC_2.38' not found (required by webcamoid-portable-linux-daily-master-arm64/bin/../lib/./libsasl2.so.2)
```

The glibc package on the system is 2.36-9+rpt2+deb12u8, it seems that the build doesn't targeting old platforms.

hipersayanX, 2024-09-19:
I have switched to Qt6 and I use Ubuntu latest LTS version as build base, Its impossible for me to provide support for older distros.

brlin-tw, 2024-09-20:
@hipersayanX 

> Its impossible for me to provide support for older distros.

IMHO if that's the case it should be mentioned on [the product description of the Payhip store page](https://payhip.com/Webcamoid), currently, it indicates that it should be compatible with all distros that shipped glibc >=2.35:

[image]

hipersayanX, 2024-09-21:
Webcamoid 9.1.1 is compatible with old systems, but the daily build (that fixes the error above), needs much newer versions. The daily build should work fine in Debian testing for example.

**NOTE**

While writing this, I remembered that the build failed for older versions of Ubuntu in ARM, I'm using [uraimo/run-on-arch-action](https://github.com/uraimo/run-on-arch-action), I don't remember the exact error, but give some hours, I will try switching the build, maybe to ubuntu22.04 or bookworm, I will make you test it.
The build for x86_64 is build under Lunar and does not have this problem.

hipersayanX, 2024-09-21:
Ok, with ubuntu22.04 it failed because [it can't detect cmake](https://github.com/webcamoid/webcamoid/actions/runs/10969376827/job/30461954462), overthought [cmake was installed](https://github.com/webcamoid/webcamoid/blob/03a0cf27d571cd9d83d74c9f84cd7db110b55d35/ports/ci/linux-ubuntu/install_deps.sh#L155), maybe that Docker image is bugged? not sure :confused:
Will try with Debian.

brlin-tw, 2024-09-22:
@hipersayanX 

> with ubuntu22.04 it failed because [it can't detect cmake](https://github.com/webcamoid/webcamoid/actions/runs/10969376827/job/30461954462), overthought [cmake was [...]

I checked the logs and it seems [that appimagetool errored before that logic](https://github.com/webcamoid/webcamoid/actions/runs/10969376827/job/30461954462#step:4:2328):

```txt
  .local/appimagetool 100%[===================>]   5.83M  23.2MB/s    in 0.3s    
  
  2024-09-21 03:47:10 (23.2 MB/s) - '.local/appimagetool-aarch64.AppImage' saved [6115712/6115712]
  
  ./ports/ci/linux-ubuntu/install_deps.sh: line 145: ./appimagetool-aarch64.AppImage: cannot execute binary file: Exec format error
  /usr/bin/cp: cannot stat 'squashfs-root/usr/*': No such file or directory
```

and [the `apt-get install` call errored and become a no-op because of a missing package](https://github.com/webcamoid/webcamoid/actions/runs/10969376827/job/30461954462#step:4:2427)

```txt
  E: Unable to locate package qml6-module-qtquick3d-spatialaudio
```

You should add either the following commands to avoid bash scripts' continued execution after a command returns a non-zero exit status code to facilitation detection of such problems:

```bash
set -e
set -o errexit
```

hipersayanX, 2024-09-22:
I have updated the scripts, thanks!

brlin-tw, 2024-09-24:
@hipersayanX Using the last daily build you sent me I can verify the problem is no longer reproduced, thanks!

hipersayanX, 2024-09-25:
If everything go well, i will make a new release the 1 of October.

hipersayanX, 2024-10-01:
Fixed in 9.2.0.

closed 2024-10-01 by hipersayanX