webcamoid/webcamoid #715: Feature suggestion: Add text overlay

lbacherle, 2024-09-27:
Hey! Thanks for your work with this project :)

Is it possible to add a basic text overlay to my video stream? Just a text that says 'hello' on top of my webcam input video, for example.
I am not sure if this feature exists and is just broken for me, or if it is not implemented.

If it does not exist, is there a reason for it? Is it hard to implement or outside the scope of this project?

hipersayanX, 2024-09-28:
Sure, this feature isn't available yet but isn't that much hard to implement.

Boruch-Baum, 2026-02-02:
It would be so nice to be able to label or timestamp footage!

hipersayanX, 2026-11-01:
Something like this would be ok?

[image]

hipersayanX, 2026-11-01:
@hipersayanX 

>Sure, this feature isn't available yet but isn't that much hard to implement.

Sure, my ⚾⚾
It would have been impossible to implement this feature without AI.