webcamoid/webcamoid #718: Webcamoid cannot use the USB camera

zhang13359, 2024-10-20:
# Reporting a problem?

## Summary

After connecting a USB camera to the phone, the camera cannot be found in Webcamoid 9.2.3.

## Expected Behavior

The USB camera can be used normally.

## Steps to Reproduce

1.  Connect the USB camera to the phone using a Type-C to USB adapter.
2.  Open webcamoid.
3.  After clicking "Main menu" and then "Video," the interface does not display the USB camera; only the phone's built-in camera is shown.

## Suggestions and tips

After connecting the USB camera, when using an application like Webcamoid, a prompt appears asking, "Do you allow access to 'NIR Camera'?"
Android 9.0 and above require camera permissions to access external UVC devices.
I suspect that Webcamoid did not request permissions.

## Your Environment

* Webcamoid information: (9.2.3) (Android)
* Operating System information: (Galaxy S24 Ultra) (Android 14)

hipersayanX, 2024-10-21:
In __Preferences/General options/Video capture__, try selecting **Android NDK Camera**, and tell me if it works.

zhang13359, 2024-10-25:
> **在“首选项/常规选项/视频捕获”**中，尝试选择**“Android NDK 相机”**，然后告诉我它是否有效。

it didn’t work.

hipersayanX, 2024-10-25:
Will check!