webcamoid/webcamoid #720: linux/usb/video.h and linux/uvcvideo.h don't exist on BSDs

yurivict, 2024-11-08:
[These 2 lines](https://github.com/webcamoid/webcamoid/blob/master/libAvKys/Plugins/VideoCapture/src/capture/v4l2sys/src/uvcextendedcontrols.cpp#L32) fail on BSDs.

They should be conditional on C++ and cmake variables that are only set when the corresponding libraries are present.

hipersayanX, 2024-11-12:
Apparently clang is only marking ```linux/usb/video.h``` as [the only conflicting symbol](https://cirrus-ci.com/task/4848098533441536?logs=main#L950), are uvc_xu_control_query, UVCIOC_*, et al. defined somewhere else? because if FreeBSD implements that API I can just remove or substitute the conflicting headers instead of disabling the feature completely.

hipersayanX, 2025-07-29:
Issue fixed in Webcamoid 9.3.0, closing.

closed 2025-07-29 by hipersayanX