webcamoid/webcamoid #721: libsndio.so.7.0 dependency when installing rpm package on Fedora 41

theokargas, 2024-12-03:
I just bought the webcamoid-official-9.2.3-1.x86_64.rpm package and tried to install on my Fedora 41 but:

Problem: conflicting requests
  - nothing provides libsndio.so.7.0()(64bit) needed by webcamoid-official-9.2.3-1.x86_64 from @commandline

I can't seem to find the package providing that library. I'd appreciate your help. Thank you.

hipersayanX, 2024-12-03:
I have resent the ticket with the AppImage.
Apparently, rpm is still trying to find the libraries in the system, when actually the libraries are actually bundled, not sure how I can fix that (without creating one RPM for every distro and version).

fduarte42, 2024-12-03:
I have the same issue. Maybe it is the problem, that this rpm requires libsndio and provides it at the same time.

hipersayanX, 2024-12-03:
@fduarte42 Resent the ticket with the AppImage, check your mail.
Will disable RPM download for now (it still available in the private repository just in case).

theokargas, 2024-12-09:
I'm sorry, I was busy, just checked this. I received a second PayHip ticket on Dec 3rd, but it was for the RPM package again. Could you send me the AppImage one please? Thank you.