webcamoid/webcamoid #722: Compilation issue on fedora 41

Vcald, 2024-12-31:
I recently had problems with certain modules. When typing 'cmake .' I get the following:
-- Checking for module 'liblsmash'
--   Package 'liblsmash' not found
-- Checking for module 'mp4v2'
--   Package 'mp4v2' not found
-- Checking for module 'faac'
--   Package 'faac' not found
-- Checking for module 'lame'
--   Package 'lame' not found

But all the relevant *-devel files are regularly present and installed.
Any hint?
Thanks

VC

hipersayanX, 2024-12-31:
Take in mind that those libraries are not required at all, just enabling ffmpeg is more than enough. But if you still want to enable the related plugins, the required libraries are:

- [lame-devel](https://pkgs.org/search/?q=lame-devel), in some distributions this is still available in the non-free repositories, in others this is available in the main one since all patents has expired already.
- [faac-devel](https://pkgs.org/search/?q=faac-devel), in most distributions this is available in the  non-free repositories. Patents expires near ~2031.  
- [libmp4v2-devel](https://pkgs.org/search/?q=libmp4v2-devel), usually not available and being removed on many distros since the last release was 4 years ago, you can get the source code at [here](https://github.com/sergiomb2/libmp4v2).
- [l-smash-devel](https://pkgs.org/search/?q=l-smash-devel) same as above, you can get the source code at [here](https://github.com/l-smash/l-smash).

Most dependencies in Webcamoid are alternatives, never requirements.

Vcald, 2025-01-01:
Thank you very much for your prompt reply.
What I mean is that I have already installed lame-devel, faac-devel and libmp4v2-devel as you suggest in your reply, but apparently they are not detected by cmake. This is weird since I have installed several other devel rpms for other webcamoid plugins and they are correctly detected.
Also, I have downloaded l-smash from github, compiled and installed it but cmake does not detect it either.
Happy new year
VC

hipersayanX, 2025-01-01:
Are the libraries registered with pkgconf?

```
pkgconf --list-all | grep 'lame\|faac\|mp4v2\|liblsmash'
```

Vcald, 2025-01-01:
This is the output I get when issuing the command you suggested:
# pkgconf --list-all | grep 'lame\|faac\|mp4v2\|liblsmash'
twolame                        twolame - An optimized MPEG Audio Layer 2 encoder

I have installed lame-devel, faac-devel and libmp4v2-devel using dnf and non-free rpmfusion repositories...concerning l-smash I have downloaded it from github, compiled and installed it.

hipersayanX, 2025-01-01:
Apparently [Fedora packages](https://fedora.pkgs.org/rawhide/fedora-aarch64/lame-devel-3.100-18.fc41.aarch64.rpm.html) does not provides the .pc files while [Arch Linux](https://archlinux.pkgs.org/rolling/archlinux-extra-aarch64/lame-3.100-5-aarch64.pkg.tar.xz.html) does. Maybe a solution would be copying the pkgconf files to the ```/usr/lib64/pkgconfig/```, remember to change the library paths of the .pc to the  right paths of Fedora, ie. lib to lib64, etc..

Vcald, 2025-01-02:
Yes...that was the problem. It works now.
Thanks