webcamoid/webcamoid #724: New problem on fedora 41

Vcald, 2025-02-07:
I have tried to compile the last sources on fedora 41...I get this error:

[ 15%] Built target AudioDevice_pipewire
[ 15%] Running AUTOMOC file extraction for target AudioEncoder_lame
[ 15%] Running moc --collect-json for target AudioEncoder_lame
[ 16%] Building CXX object libAvKys/Plugins/Codecs/Audio/Encoders/lame/CMakeFiles/AudioEncoder_lame.dir/AudioEncoder_lame_autogen/mocs_compilation.cpp.o
[ 16%] Building CXX object libAvKys/Plugins/Codecs/Audio/Encoders/lame/CMakeFiles/AudioEncoder_lame.dir/src/audioencoderlame.cpp.o
[ 16%] Linking CXX shared module ../../../build/lib64/qt/plugins/avkys/libAudioGen.so
[ 16%] Built target AudioGen
[ 16%] Building CXX object libAvKys/Plugins/Codecs/Audio/Encoders/lame/CMakeFiles/AudioEncoder_lame.dir/src/audioencoderlameelement.cpp.o
[ 16%] Linking CXX shared module ../../../../../../build/lib64/qt/plugins/avkys/libAudioEncoder_faac.so
[ 17%] Linking CXX shared module ../../../../../../build/lib64/qt/plugins/avkys/libVideoMuxer_lsmash.so
[ 18%] Running AUTOMOC file extraction for target AudioEncoder_opus
[ 18%] Running moc --collect-json for target AudioEncoder_opus
[ 18%] Built target AudioEncoder_faac
[ 18%] Running AUTOMOC file extraction for target AudioEncoder_vorbis
[ 18%] Building CXX object libAvKys/Plugins/Codecs/Audio/Encoders/opus/CMakeFiles/AudioEncoder_opus.dir/AudioEncoder_opus_autogen/mocs_compilation.cpp.o
[ 18%] Running moc --collect-json for target AudioEncoder_vorbis
[ 18%] Building CXX object libAvKys/Plugins/Codecs/Audio/Encoders/opus/CMakeFiles/AudioEncoder_opus.dir/src/audioencoderopus.cpp.o
[ 18%] Building CXX object libAvKys/Plugins/Codecs/Audio/Encoders/vorbis/CMakeFiles/AudioEncoder_vorbis.dir/AudioEncoder_vorbis_autogen/mocs_compilation.cpp.o
/usr/bin/ld: /usr/lib/gcc/x86_64-redhat-linux/14/../../../liblsmash.a(description.o): relocation R_X86_64_32 against `.bss' can not be used when making a shared object; recompile with -fPIC
/usr/bin/ld: failed to set dynamic section sizes: bad value
collect2: error: ld returned 1 exit status
make[2]: *** [libAvKys/Plugins/Codecs/Format/Muxers/lsmash/CMakeFiles/VideoMuxer_lsmash.dir/build.make:161: build/lib64/qt/plugins/avkys/libVideoMuxer_lsmash.so] Error 1
make[1]: *** [CMakeFiles/Makefile2:4384: libAvKys/Plugins/Codecs/Format/Muxers/lsmash/CMakeFiles/VideoMuxer_lsmash.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 18%] Building CXX object libAvKys/Plugins/Codecs/Audio/Encoders/vorbis/CMakeFiles/AudioEncoder_vorbis.dir/src/audioencodervorbis.cpp.o
[ 18%] Building CXX object libAvKys/Plugins/Codecs/Audio/Encoders/opus/CMakeFiles/AudioEncoder_opus.dir/src/audioencoderopuselement.cpp.o
[ 18%] Linking CXX shared module ../../../../../../build/lib64/qt/plugins/avkys/libVideoMuxer_ffmpeg.so
[ 18%] Linking CXX shared module ../../../../../build/lib64/qt/plugins/avkys/libAudioDevice_sdl.so
[ 19%] Built target VideoMuxer_ffmpeg
[ 19%] Building CXX object libAvKys/Plugins/Codecs/Audio/Encoders/vorbis/CMakeFiles/AudioEncoder_vorbis.dir/src/audioencodervorbiselement.cpp.o
[ 19%] Linking CXX shared module ../../../../../../build/lib64/qt/plugins/avkys/libVideoMuxer_mp4v2.so
[ 19%] Linking CXX shared module ../../../../../build/lib64/qt/plugins/avkys/libAudioDevice_qtaudio.so
[ 19%] Built target AudioDevice_sdl
[ 19%] Built target VideoMuxer_mp4v2
[ 19%] Built target AudioDevice_qtaudio
[ 19%] Linking CXX shared module ../../../../../../build/lib64/qt/plugins/avkys/libAudioEncoder_fdkaac.so
[ 19%] Built target AudioEncoder_fdkaac
[ 19%] Linking CXX shared module ../../../../../../build/lib64/qt/plugins/avkys/libAudioEncoder_lame.so
[ 19%] Built target AudioEncoder_lame
[ 19%] Linking CXX shared module ../../../../../../build/lib64/qt/plugins/avkys/libAudioEncoder_ffmpeg.so
[ 19%] Built target AudioEncoder_ffmpeg
[ 19%] Linking CXX shared module ../../../../../../build/lib64/qt/plugins/avkys/libAudioEncoder_opus.so
[ 20%] Linking CXX shared module ../../../../../../build/lib64/qt/plugins/avkys/libAudioEncoder_vorbis.so
[ 20%] Built target AudioEncoder_opus
[ 20%] Built target AudioEncoder_vorbis
make: *** [Makefile:136: all] Error 2

I have also tried to exclude libsmash...in this case it compiles fine apparently...but then when I launch the executable I get a core dump.
Any hint?
Thanks

hipersayanX, 2025-02-07:
The Build in Fedora is working [fine](https://github.com/webcamoid/webcamoid/actions/runs/12935573496/job/36079217147), maybe try cleaning your build folder?

Vcald, 2025-02-07:
I think the problem with fPIC was that I have compiled l-smash library manually without disabling the static library compilation...so webcamoid apparently looked for the static library instead of the shared one.
So...I compiled l-smash with shared libs only...compilation of webcamoid went smoothly, but when I launch the executable I get a core dump.
So I thought it could be a problem with the smash library version...I then installed the opensuse rpms...compilation of webcamoid went again smooth...but again a core dump upon launching the executable.
Then I found on rpm pbone site a version of liblsmash and l-smash-devel rpms compiled for fedora 40. I installed them and compiled again webcamoid. Compilation was again fine...but again when launching the executable I get a core dump.

hipersayanX, 2025-02-08:
Run Webcamoid with gdb and look where the core dump is get thrown.

Vcald, 2025-02-08:
The core dump is thrown in /var/lib/systemd/coredump.
I have uncompressed it...then I ran 'gdb webcamoid uncompressed_coredump'
and I get an enornous amount of info...that I do not know how to interpret.
What shoud I look at?

Vcald, 2025-02-09:
Now...when I launch the executable...this is what I get. Could the problem be due to some graphic driver thing?
My graphic situation is:
00:02.0 VGA compatible controller: Intel Corporation UHD Graphics 620 (rev 07)
01:00.0 Display controller: Advanced Micro Devices, Inc. [AMD/ATI] Topaz XT [Radeon R7 M260/M265 / M340/M360 / M440/M445 / 530/535 / 620/625 Mobile] (rev c1)

I hope this gives you some clue.

localhost]# build/bin/webcamoid
[2025-02-09 07:54:04.736, Webcamoid, 0x14b5632b2a80,  (0)] info: Using Qt multimedia with FFmpeg version 7.0.2 GPL version 3 or later
[2025-02-09 07:54:04.737, Webcamoid, 0x14b5632b2a80,  (0)] info: Available HW decoding frameworks:
libva info: VA-API version 1.22.0
libva info: Trying to open /usr/lib64/dri-nonfree/iHD_drv_video.so
libva info: Found init function __vaDriverInit_1_22
libva info: va_openDriver() returns 0
libva info: VA-API version 1.22.0
libva info: Trying to open /usr/lib64/dri-nonfree/iHD_drv_video.so
libva info: Found init function __vaDriverInit_1_22
libva info: va_openDriver() returns 0
[2025-02-09 07:54:05.359, Webcamoid, 0x14b5632b2a80,  (0)] info:      vaapi
[2025-02-09 07:54:05.359, Webcamoid, 0x14b5632b2a80,  (0)] info:      vdpau
[2025-02-09 07:54:05.359, Webcamoid, 0x14b5632b2a80,  (0)] info:      qsv
[2025-02-09 07:54:05.359, Webcamoid, 0x14b5632b2a80,  (0)] info:      vulkan
[2025-02-09 07:54:05.359, Webcamoid, 0x14b5632b2a80,  (0)] info: Available HW encoding frameworks:
[2025-02-09 07:54:05.359, Webcamoid, 0x14b5632b2a80,  (0)] info:      vaapi
[2025-02-09 07:54:05.359, Webcamoid, 0x14b5632b2a80,  (0)] info:      vdpau
[2025-02-09 07:54:05.359, Webcamoid, 0x14b5632b2a80,  (0)] info:      qsv
[2025-02-09 07:54:05.359, Webcamoid, 0x14b5632b2a80,  (0)] info:      vulkan
Segmentation fault (core dumped)

hipersayanX, 2025-02-09:
Shown me the gdb log.

Vcald, 2025-02-10:
Please find attached the gdb log

[webcamoid_debug.log](https://github.com/user-attachments/files/18730728/webcamoid_debug.log)

hipersayanX, 2025-02-10:
Git pull, build, and try again, it was probably due to the unchecked pointers.

Vcald, 2025-02-10:
Yes...it works now
Thank you very much

closed 2025-02-10 by hipersayanX